morrownr/8821cu-20210118 #4: rtw_led_ctrl=0 does not work

smp79, 2022-02-17:
```
# lsusb
Bus 001 Device 003: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC

# ls /sys/module/8821cu/parameters/
if2name                          rtw_notch_filter
ifname                           rtw_pci_dynamic_aspm_linkctrl
rtw_FileMaskEfuse                rtw_phy_file_path
rtw_GLNA_type                    rtw_phydm_ability
rtw_OffEfuseMask                 rtw_pll_ref_clk_sel
rtw_RFE_type                     rtw_power_mgnt
rtw_TxBBSwing_2G                 rtw_powertracking_type
rtw_TxBBSwing_5G                 rtw_pwrtrim_enable
rtw_acs                          rtw_qos_opt_enable
rtw_acs_auto_scan                rtw_quick_addba_req
rtw_adaptivity_en                rtw_recvbuf_nr
rtw_adaptivity_mode              rtw_rfintfs
rtw_adaptivity_th_edcca_hl_diff  rtw_rrsr_value
rtw_adaptivity_th_l2h_ini        rtw_rx_ampdu_amsdu
rtw_ampdu_enable                 rtw_rx_ampdu_sz_limit_1ss
rtw_amplifier_type_2g            rtw_rx_ampdu_sz_limit_2ss
rtw_amplifier_type_5g            rtw_rx_ampdu_sz_limit_3ss
rtw_ant_num                      rtw_rx_ampdu_sz_limit_4ss
rtw_antdiv_cfg                   rtw_rx_nss
rtw_antdiv_type                  rtw_rx_path_lmt
rtw_antenna_gain                 rtw_rx_stbc
rtw_ap_fwd_b2u_flags             rtw_rxgain_offset_2g
rtw_ap_src_b2u_flags             rtw_rxgain_offset_5gh
rtw_btcoex_enable                rtw_rxgain_offset_5gl
rtw_busy_thresh                  rtw_rxgain_offset_5gm
rtw_bw_mode                      rtw_scan_interval_thr
rtw_ch_switch_offload            rtw_single_ant_path
rtw_channel                      rtw_smart_ps
rtw_channel_plan                 rtw_stbc_cap
rtw_check_hw_status              rtw_switch_usb_mode
rtw_chip_version                 rtw_target_tx_pwr_2g_a
rtw_country_code                 rtw_target_tx_pwr_2g_b
rtw_decrypt_phy_file             rtw_target_tx_pwr_2g_c
rtw_dfs_region_domain            rtw_target_tx_pwr_2g_d
rtw_drv_ant_band_switch          rtw_target_tx_pwr_5g_a
rtw_drv_log_level                rtw_target_tx_pwr_5g_b
rtw_dynamic_agg_enable           rtw_target_tx_pwr_5g_c
rtw_en_dyn_rrsr                  rtw_target_tx_pwr_5g_d
rtw_en_gro                       rtw_trx_path_bmp
rtw_en_napi                      rtw_tsf_update_pause_factor
rtw_enusbss                      rtw_tsf_update_restore_factor
rtw_excl_chs                     rtw_tx_aclt_conf_ap_m2u
rtw_full_ch_in_p2p_handshake     rtw_tx_aclt_conf_default
rtw_halrf_ability                rtw_tx_aclt_flags
rtw_hiq_filter                   rtw_tx_ampdu_amsdu
rtw_ht_enable                    rtw_tx_bw_mode
rtw_hw_wps_pbc                   rtw_tx_nss
rtw_hwpdn_mode                   rtw_tx_path_lmt
rtw_hwpwrp_detect                rtw_tx_pwr_by_rate
rtw_initmac                      rtw_tx_pwr_lmt_enable
rtw_ips_mode                     rtw_usb_rxagg_mode
rtw_iqk_fw_offload               rtw_vcs_type
rtw_lbkmode                      rtw_vht_24g_enable
rtw_load_phy_file                rtw_vht_enable
rtw_low_power                    rtw_vht_rx_mcs_map
rtw_lps_chk_by_tp                rtw_vrtl_carrier_sense
rtw_lps_level                    rtw_wifi_spec
rtw_max_bss_cnt                  rtw_wireless_mode
rtw_max_roaming_times            rtw_wmm_enable
rtw_mp_mode                      rtw_wowlan_sta_mix_mode
rtw_network_mode
```

morrownr, 2022-02-17:
Hi @smp79 

I haven't had time to add that capability yet. It is on my list so expect it to happen as I have time.

Can you give me a report on the driver so far?

Regards

smp79, 2022-02-17:
> Can you give me a report on the driver so far?

Works fine, no other issues to report :+1:

morrownr, 2022-02-18:
Hi @smp79 

Since you did a good job of following instructions when reporting the issue, I moved this issue to priority one.

Recommend:

```
sudo ./remove-driver.sh
git pull
sudo ./install-driver.sh
```
Would you be so kind as to test all 3 options and provide a report?

Regards

smp79, 2022-02-18:
LED control now works as expected. Thank you!