morrownr/8821cu-20210118 #7: just to inform, new version works!

ohquait, 2022-02-18:
Test's OK
messages:
`Re: dkms-rtl8821cu UPDATE to...
« Sent to: [onkelho](https://www.pclinuxos.com/forum/index.php?action=profile;u=47978) on: Yesterday at 11:16:38 AM »
« You have forwarded or responded to this message. »

    [Reply](https://www.pclinuxos.com/forum/index.php?action=pm;sa=send;f=inbox;pmsg=107828;u=10787)
    [Quote](https://www.pclinuxos.com/forum/index.php?action=pm;sa=send;f=inbox;pmsg=107828;quote;u=10787)
    [Delete](https://www.pclinuxos.com/forum/index.php?action=pm;sa=pmactions;pm_actions[107828]=delete;f=inbox;start=45;bd9e4d91de=902a9fb4f9ca6d44066906d5110c5102)

Checked and working, thumbs up

~]$ dkms status
rtl8821cu, 5.12.0-1pclos2022, 5.15.24-pclos1, x86_64: installed
rtl8821cu, 5.12.0-1pclos2022, 5.16.10-pclos1, x86_64: installed
nvidia340, 340.108-13pclos2022, 5.15.24-pclos1, x86_64: installed
nvidia340, 340.108-13pclos2022, 5.16.10-pclos1, x86_64: installed
rtl88x2bu, 5.13.1-6pclos2022, 5.15.24-pclos1, x86_64: installed
rtl88x2bu, 5.13.1-6pclos2022, 5.16.10-pclos1, x86_64: installed

[Quote from: onkelho on Yesterday at 02:21:53 AM](https://www.pclinuxos.com/forum/index.php?action=profile;u=47978)

    https://qua-it.org/XFCE/repo/apt/quait/2022/RPMS.x86_64/dkms-rtl8821cu-5.12.0-1pclos2022.x86_64.rpm[](https://www.pclinuxos.com/forum/index.php?action=pm;sa=report;l=-1;pmsg=107828)

    Hey Ed,
    if you have time, please test this new version.
    morrownr have switched to newer driver-source-code from Realtek for ...

    Greetings!`

spec:
`# imported from,
# git 
#old https://github.com/brektrou/rtl8821CU/tree/5.8.1
#old https://github.com/mbrysa/rtl8821CU
### thx to Ed ####
#new real version 5.12.0  https://github.com/morrownr/8821cu-20210118

%global	module	8821cu
%global pname rtl%{module}
#%global lname %{pname}-legacy

Name:		dkms-%{pname}
Version:	5.12.0
Release:	%mkrel 1
Source0:	%{pname}-%{version}.tar.xz
License:	GPLv2
Group:		System/Kernel and hardware
URL:		https://github.com/morrownr/8821cu
Summary:	dkms package for realtek RTL8811CU, RTL8821CU and RTL8731AU Chipsets

Patch0:		rtl8821cu-pclos-%{version}.patch
Patch1:		rtl8821cu-conffile-%{version}.patch

BuildRoot:	%{_tmppath}/%{name}-%{version}-buildroot

Provides:	kmod(rtl%{module}.ko) = %{version}
		
Requires(post):		dkms
Requires(preun):	dkms

Obsoletes:	dkms-%{pname} < %{version}-%{release}

%description
Linux Driver for USB WiFi Adapters that are based on the
RTL8811CU, RTL8821CU and RTL8731AU Chipsets
    v5.12.0 (Realtek) (20210118) plus updates from the Linux community
Features
    IEEE 802.11 b/g/n/ac WiFi compliant
    802.1x, WEP, WPA TKIP and WPA2 AES/Mixed mode for PSK and TLS (Radius)
[... 307 lines cut ...]
 	#$(MAKE) -C $(KSRC) M=$(shell pwd) clean
-	cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
-	cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
-	cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
-	cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
-	cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
-	cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
-	cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
-	cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
-	cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
+	cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko*
+	cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko*
+	cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko*
+	cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko*
+	cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko*
+	cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko*
+	cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko*
+	cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko*
+	cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko*
 	rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
-	rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
+	rm -fr *.mod.c *.mod *.o .*.cmd *.ko* *~
 	rm -fr .tmp_versions
 endif
`

morrownr, 2022-02-18:
Hi @ohquait 

Thanks for the report. So far this appears to be a good driver, certainly better than the last. Unfortunately, out-of-kernel drivers will be problematic in the long run and these Realtek drivers are missing some features that will require users to move to Mediatek based adapters. We really need Realtek to add usb support to the 8821cu code that is in the kernel. It is not clear to me why this is a problem for Realtek.

Regards

ohquait, 2022-02-18:
Hey Nick,
the based problem is not by Realtek or Linux, this is based in manufactur's
they will use Realtek without open sourced drivers,....
not clear for why, i think it's questionable licenses with and
money-dependenies
...
can you find on Larry Fingers repo?
or on debian-sources?
for today i can Not doing more, next days will try search,...
Excu my bad Language, mobil device and to biggest hands

morrownr, 2022-02-22:
I've seen Larry's repo but I really think Realtek needs to be behind the effort to do it right.