morrownr/8821cu-20210118 #8: Bluetooth scan / connection failed if WiFi is ON

ffongmdsp, 2022-02-28:
Hello, 

I updated to the latest 8821cu driver (14798a1f917383acef2d8b3f7b40acaa64b5f5ef).  Now it is great to see the latest dongle works. Wi-Fi and Bluetooth works but only _separately_:

- Bluetooth scan fails to find any devices unless I turned off the Wi-Fi interface. 
- Bluetooth connection will be dropped as soon as I turned on the Wi-Fi interface. 

The system is Linux 5.4.75. The hardware is EDUP USB Dongle. I could provide more details is needed.

_Note: an older version of the dongle does not have this problem. (It is a version where it appears as USB Mass-Storage when inserted, requires usb-modeswitch to turn it into NIC)._

morrownr, 2022-02-28:
Hi @ffongmdsp 

I'm going to mark this issue as `needs help` so if anyone else that may have a solution comes by, they may see it better.

I'm at a disadvantage in the my 8811cu adapter does not have BT capability so it is impossible for me to attempt to duplicate the problem. Also, my to-do list is long right now. I will take a look at I can.

Separate subject: How did you handle step 9 of the installation steps. That Rockchip is ARM based, correct?

Regards

ffongmdsp, 2022-03-01:
Hello @morrownr,

Thanks for your feedbacks.

1. Regarding how I configure to compile for the Rockchip Platform (Step-9) - instead of DKMS, I simply run the Makefile passing in some variable values:
`make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- KSRC=<Path to my Rockhip kernel source>`

3. I could send you a EDUP Dongle (WiFi + BT) for reference if that helps.

4. Feel free to let me know if you would like to experiment changes of the code. I could do this here for you.

Thanks
Frankie

morrownr, 2022-03-14:
Hi @ffongmdsp 

> I could send you a EDUP Dongle (WiFi + BT) for reference if that helps.

I may have a 8822bu adapter headed my way from another source so I may be able to sort this out with 88x2bu driver. Thanks for the offer but let me take a rain check for now as I am so far behind on my to-do list that this could be something I don't get to for a while.

Regards

ffongmdsp, 2022-03-14:
Thanks for the update. I investigated a little further. I built the Realtek driver and run on Raspberry Pi 3B. There is no such problem. That RPi image is using 5.4.79+ Linux kernel with BlueZ 5.50. I am trying to follow this configuration on my Nanopi. I tried BlueZ 5.50 on it but problem persists. I am currently seeking support to have the same kernel version running on the Nanopi. Will see how it goes.

ffongmdsp, 2022-03-16:
Here is an update. I have upgraded my nanopi system to 5.4.79+ Linux Kernel and BlueZ 5.50 - exactly matching those version to my Raspberry Pi. But same problem remains. I compared the 8821 logs (captured with rtw_drv_log_level=6). I don't see any obvious difference so far. And I did look for outputs related to Bluetooth Coexistence. And this line appears in nanopi log:
`EEPROM Disable BT-coex by hal_spec`
However, the same line appears in Raspberry Pi also.

morrownr, 2022-03-16:
Explain to me what BlueZ 5.50 is.

ffongmdsp, 2022-03-17:
> Explain to me what BlueZ 5.50 is.

BlueZ is a user-space Bluetooth stack. 5.5.0 is the version number. I think it is the most widely used Bluetooth interface in Linux application. It provides various utilities for user to manage Bluetooth adapter, to discover, pair, connect to various Bluetooth devices.  
[http://www.bluez.org/](http://www.bluez.org/)

I mentioned this because this is the main dependency for my application in terms of using the Bluetooth function of the 8821cu-based USB dongle. My nanopi was using v5.23 before my attempt to match the Raspberry Pi which hosts v5.50.

Hope this helps.