morrownr/8821cu-20210118 #9: cc1: error: incompatible gcc/plugin versions [Arch Linux 32]

tjoepies, 2022-03-14:
I have a compiling error on Arch Linux 32 after a kernel update.

```
$ make
make ARCH=i386 CROSS_COMPILE= -C /lib/modules/5.16.12-arch1-1.0/build M=/home/theunis/Downloads/8821cu/8821cu-20210118  modules
make[1]: Entering directory '/usr/lib/modules/5.16.12-arch1-1.0/build'
  CC [M]  /home/theunis/Downloads/8821cu/8821cu-20210118/core/rtw_cmd.o
cc1: error: incompatible gcc/plugin versions
cc1: error: failed to initialize plugin ./scripts/gcc-plugins/structleak_plugin.so
make[2]: *** [scripts/Makefile.build:287: /home/theunis/Downloads/8821cu/8821cu-20210118/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1846: /home/theunis/Downloads/8821cu/8821cu-20210118] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.16.12-arch1-1.0/build'
make: *** [Makefile:2498: modules] Error 2

```

```
$ uname -a
Linux GateWay 5.16.12-arch1-1.0 #1 SMP PREEMPT Fri, 04 Mar 2022 13:50:58 +0000 i686 GNU/Linux

```

```
$ cat /proc/version
Linux version 5.16.12-arch1-1.0 (linux@archlinux) (gcc (GCC) 11.2.0, GNU ld (GNU Binutils) 2.38) #1 SMP PREEMPT Fri, 04 Mar 2022 13:50:58 +0000

```

```
$ gcc --version | head -n1
gcc (GCC) 11.2.0

```

```
$ pacman -Q binutils
binutils 2.38-3.9

```

the `gcc` version seems to match the kernel's compiled version. I'm stumped, any ideas?

morrownr, 2022-03-16:
Hi @tjoepies 

I don't see anything that I can offer to help at this point. It is not everyday that I see kernel 5.16 and gcc 11.2 in use on a i386 system. I test the drivers here on a couple of i386 based systems but kernel and gcc are usually much older... like 4.19 or thereabouts.

morrownr, 2022-03-17:
Right now my to-do list is long so my ability to work on this issue is limited so I will throw out what I think is going on and maybe it will help you find a solution.

What I think is happening is when you install the second kernel, you are causing a mismatch with the kernel headers that are installed. How to fix that is something that is best asked over on an Arch forum. What you might consider doing is uninstalling that updated kernel. Ask yourself this: Is there anything the new kernel is doing for me on this hardware?

morrownr, 2022-05-19:
Status?

closed 2022-07-08 by morrownr