morrownr/8821cu-20210118 #10: Module not found

ktistes, 2022-03-23:
I tried to install the driver on various linux distros: void, garuda, mint and the driver is installed (with dkms or with make) but I always get this message: "FATAL: Module rtl8821cu not found in directory /lib/modules/5.4.0-104-generic" . No wifi interface, whereas the bluetooth works
[rtl8821cu_log.txt](https://github.com/morrownr/8821cu-20210118/files/8331449/rtl8821cu_log.txt)
.

morrownr, 2022-03-25:
Hi @ktistes 

I have been pondering this issue. My first diagnosis would be that we are seeing a problem with secure mode but it is not clear to me if your system supports secure mode. This driver should support secure mode installation, at least on Linux Mint, 

If your system does support secure mode, turn it off, at least temporarily, for testing. You will need to run the `remove-driver.sh` script to clean everything out before attempting to reinstall.

Let me know what happens.

ktistes, 2022-03-28:
Thanks Nick, I disabled safe mode in / etc / default / grub in Mint and wifi connection still doesn't work. If I unplug the cable, the wireless icon appears, but it only points to the potential connection shared with the phone's bluetooth

morrownr, 2022-03-28:
> 34.219018] kernel: 8821cu: module verification failed: signature and/or required key missing

It appears to me that there is an issue with secure mode. If secure mode is properly setup when you install the operating system, then dkms (using the installation script) will be able to set the driver up to run with secure mode. This is known to work on Debian and Debian based systems such as Mint and it is well tested. There can be issues with secure mode on openSuse and Fedora but I have been unable to get help from users of those distros to track the problem down.

If secure mode is on in your bios, you probably want to consider turning it off while troubleshooting.

ktistes, 2022-04-04:
﻿I have tried the installation into 3 devices, all multiboot. Now I'm focusing on a desktop with windows 10 on the hard disk and 2 linuxes (mint and void) on an usb disk. Trying with mint, I disabled the safe mode in grub (and there is no option for secure mode in this old bios), but no success (yes bluetooth, no wifi). I don't know if there are other ways to disable safe mode in linux, but I wonder how other owners of these devices did with this driver.

morrownr, 2022-04-04:
> there is no option for secure mode in this old bios

Then the computer does not support secure mode so this cannot be the issue. Maybe we need to ensure the chipset in the adapter. Can you post the output of the following command?

```
lsusb
```

ktistes, 2022-04-05:
Ok.
lsusb
Bus 002 Device 003: ID 174c:1153 ASMedia Technology Inc. ASM1153 SATA 3Gb/s bridge
Bus 002 Device 002: ID 0bda:c820 Realtek Semiconductor Corp. 802.11ac NIC
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 003: ID 0000:0538   USB OPTICAL MOUSE
Bus 008 Device 002: ID 046d:c312 Logitech, Inc. DeLuxe 250 Keyboard
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

morrownr, 2022-04-05:
> Bus 002 Device 002: ID 0bda:c820 Realtek Semiconductor Corp. 802.11ac NIC

That confirms that your adapter is based on the 8811cu chipset so the driver in this repository is the correct driver.

Let's try this:

Go you your driver directory and run the following:

```
$ make clean
$ make
after taking a look...
$ make clean
```

Do you see any error messages? I realize it would be a lot to post the entire output but if something looks wrong, post the part that looks wrong.

Also, post the results of the following...

```
$ sudo dkms status
$ lsmod
```

FYI: I took a look at the repo usage info. I'm seeing around 500 downloads over the last 2 weeks but, as you can see, there have been very few issues reported. I'm aware that some distros, such as openSUSE and Arch/Manjaro can have issues if secure mode is turned on. I'd like to fix the issues but I don't have the time and would need a knowledgeable user of those distros to work with me for a fix.  Secure mode works fine with Debian and Debian based distros. Hopefully info from a manual compile will give us a hint as to what the problem is.

morrownr, 2022-04-22:
I am closing the issue due to inactivity. You can reopen the issue if necessary.

closed 2022-04-22 by morrownr