morrownr/8821cu-20210118 #11: Not compatible with hcxdumptool?

Liniya, 2022-03-24:
I'm trying to use [hcxdumptool](https://github.com/ZerBea/hcxdumptool/) as described [here](https://github.com/ZerBea/hcxdumptool/blob/master/docs/first_steps.txt). `sudo hcxdumptool -i <wlan_interface> --check_driver` throws an error like this:

`failed to set monitor mode, ioctl(SIOCSIWMODE) not supported by driver`

Is it possible to fix?

```
Linux username 5.16.14-1-MANJARO #1 SMP PREEMPT Fri Mar 11 14:12:18 UTC 2022 x86_64 GNU/Linux
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 003: ID 1532:006e Razer USA, Ltd DeathAdder Essential
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 0b05:18f3 ASUSTek Computer, Inc. AURA LED Controller
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
9: phy9: Wireless LAN
        Soft blocked: yes
        Hard blocked: no
openrazer-driver/3.2.0, 5.16.14-1-MANJARO, x86_64: installed
rtl8821cu/5.8.1.7.7.4bdd7c8, 5.16.14-1-MANJARO, x86_64: installed
phy#9
        Interface wlp5s0f1u2
                ifindex 46
                wdev 0x900000001
                addr <edited>
                type managed
                txpower -100.00 dBm
```

Liniya, 2022-03-27:
If I use `start-mon.sh` that comes with the driver to enable the monitor mode, `hcxdumptool` 'sort of' works. However, you can't set the channel ('No available frequencies'), and it doesn't see 5G AP's at all.

Liniya, 2022-03-28:
The version compiled from the latest sources works without a problem. Closing.

closed 2022-03-28 by Liniya

morrownr, 2022-03-28:
> The version compiled from the latest sources works without a problem.

I was reviewing this and saw that you were using an old version of the driver when you closed it.

This new version is much better.

Regards