morrownr/8821cu-20210118 #12: Cannot set channel/width in monitor mode using iw (8811cu)

Liniya, 2022-03-28:
I can't seem to be able to set channel and width in monitor mode using `iw`. I'm using an 8811cu adapter (Tenda U9).

It's possible to set the channel using another command `iwconfig`, but it doesn't allow you to set the channel width. Without this, the captured data becomes incomplete in case of 2.4G 40MHz AP's and similar.

```
$ sudo iw dev wlp5s0f1u2 set type monitor
$ iw dev
phy#0
        Interface wlp5s0f1u2
                ifindex 20
                wdev 0x1
                addr <...>
                type monitor
                channel 1 (2412 MHz), width: 20 MHz (no HT), center1: 2412 MHz
                txpower 19.00 dBm

$ sudo iw --debug dev wlp5s0f1u2 set channel 8 HT40+
-- Debug: Sent Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [NETLINK HEADER] 16 octets
    .nlmsg_len = 60
    .type = 36 <0x24>
    .flags = 5 <REQUEST,ACK>
    .seq = 1648489760
    .port = 1367399177
  [GENERIC NETLINK HEADER] 4 octets
    .cmd = 2
    .version = 0
    .unused = 0
  [PAYLOAD] 40 octets
    08 00 03 00 14 00 00 00 08 00 26 00 8f 09 00 00 ..........&.....
    08 00 9f 00 02 00 00 00 08 00 27 00 03 00 00 00 ..........'.....
    08 00 a0 00 99 09 00 00                         ........
---------------------------  END NETLINK MESSAGE   ---------------------------
-- Debug: Received Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [NETLINK HEADER] 16 octets
    .nlmsg_len = 36
    .type = 2 <ERROR>
    .flags = 256 <ROOT>
    .seq = 1648489760
    .port = 1367399177
  [ERRORMSG] 20 octets
    .error = 0 "Success"
  [ORIGINAL MESSAGE] 16 octets
    .nlmsg_len = 16
    .type = 36 <0x24>
    .flags = 5 <REQUEST,ACK>
    .seq = 1648489760
    .port = 1367399177
---------------------------  END NETLINK MESSAGE   ---------------------------

$ iw dev
phy#0
        Interface wlp5s0f1u2
                ifindex 20
                wdev 0x1
                addr 12:40:52:63:2c:e0
                type monitor
                channel 1 (2412 MHz), width: 20 MHz (no HT), center1: 2412 MHz
                txpower 19.00 dBm
```

In managed mode, it sets everything correctly when I connect to an AP.

```
$ iw dev
phy#0
        Interface wlp5s0f1u2
                ifindex 20
                wdev 0x1
                addr <...>
                ssid <...>
                type managed
                channel 8 (2447 MHz), width: 40 MHz, center1: 2437 MHz
                txpower 19.00 dBm
```

morrownr, 2022-03-28:
Try the below attached script. It is basically a replacement for airmon-ng and is nice neat code that should be easy to read and modify.

```
sudo ./start-mon.sh wlan0
```

[start-mon.sh.tar.gz](https://github.com/morrownr/8821cu-20210118/files/8365968/start-mon.sh.tar.gz)

It should change your channel fine. I haven't had time to add width capability yet.

The reason for this script is primary because Realtek, in its infinite wisdom, does not bother to make Linux Standards compliant drivers. This script helps ensure a nice transition to monitor mode. Like I said, I have not added width capability yet but if you figure it out before I have time, please share so I can get it in there.

Nick

Liniya, 2022-03-29:
Hello. `start-mon.sh` is the same script that comes with the driver, isn't it? I see that it also uses the `iw` command to set the channel:

`iw dev $iface0mon set channel $chan`

Point is, setting the channel with `iw` simply **does not work**, at least with my adapter (system?). When I run `iw dev` afterwards, the channel remains at 1. Only `iwconfig` works:

`sudo iwconfig wlan0mon channel 8`

>Like I said, I have not added width capability yet but if you figure it out before I have time, please share so I can get it in there.

:) This stuff is probably way beyond me. No worries, I can wait. Thanks for your work!

morrownr, 2022-03-29:
> Hello. `start-mon.sh` is the same script that comes with the driver, isn't it? 

The one I attached is somewhat cleaned up as I make improvements as I have time.
 
> `iw dev $iface0mon set channel $chan`
> 
> Point is, setting the channel with `iw` simply **does not work**,

Actually it does work. What does not work right is this driver providing the correct information to `iw dev`.

Here is the latest `start-mon.sh` with better documentation in the channel set routine.

[start-mon.sh.tar.gz](https://github.com/morrownr/8821cu-20210118/files/8371433/start-mon.sh.tar.gz)

Here is that updated documentation:

```
#		Documentation:
#		iw dev <devname> set channel <channel> [NOHT|HT20|HT40+|HT40-|5MHz|10MHz|80MHz]
#		iw dev <devname> set freq <freq> [NOHT|HT20|HT40+|HT40-|5MHz|10MHz|80MHz]
#		iw dev <devname> set freq <control freq> [5|10|20|40|80|80+80|160] [<center1_freq> [<center2_freq>]]
#		Select one or modify as required:
		iw dev $iface0mon set channel $chan
#		iw dev $iface0mon set channel $chan HT40-		
#		iw dev $iface0mon set channel $chan 80MHz
#		To test if channel was set correctly:
#		aireplay-ng --test <wlan0>
```

Since I do not know how familiar you are with BASH, let me explain some things. Any line that starts with `#` is a comment. It does nothing. It is there for you to read.

Right now, in the above section there is only one active line:

```
		iw dev $iface0mon set channel $chan
```

If you comment this line and uncomment one of the two lines below it, you change the active line.

Keep in mind that the script is started with `sudo` so `sudo` is active while running the script so if you run these lines outside of the script, you need to use `sudo`.

Run the above script, set the channel you want, and then test it:

```
sudo aireplay-ng --test <wlan0>
```

You will need to replace `<wlan0>` with the name of your interface which will likely be `wlan0mon`.

Tell me if it uses the channel you selected. If it did, then setting the channel is working. 

> :) This stuff is probably way beyond me. No worries, I can wait. Thanks for your work!

You can get there. I understand that this can be confusing at the beginning so I will give you some advice:

Experienced security experts/pen testers generally find that using usb wifi adapters that use in-kernel drivers provide much better results with much less aggravation. The following site has over 75 links to adapters that use in-kernel drivers:

https://github.com/morrownr/USB-WiFi

In-kernel drivers are Linux Standards compliant. Realtek drivers are not standards compliant and they are buggy... as you can see. I don't maintain this driver to advocate that Linux users buy Realtek based adapters. On the contrary, I advocate that Linux users buy usb wifi adapters that use in-kernel drivers. I only maintain this driver to help those that already have Realtek based adapters.

Regards

Liniya, 2022-03-29:
Ah, now I see what you mean. Setting the channel _and_ width with `iw dev` in monitor mode does in fact work, it's just reported incorrectly by `iw dev` because the driver provides wrong information to it.

It's a bit of a problem with some tools as far as I can see, e.g. `aireplay-ng` gives the 'AP is on channel N but the monitor uses channel M' error if you try to deauth some clients. But nothing too drastic, and definitely is good news for me, as previously I assumed _the driver itself_ doesn't yet implement this setting in monitor mode. Thanks again - I can work with this. :)