morrownr/8821cu-20210118 #13: My dongle stopped

JuniorPolegato, 2022-04-15:
Hi!
Can you help me to solve this problem, please?
I tried many rtw_RFE_type to resolve this but no life...
There is a way to get wifi_efuse_8821cu.map?
Thanks.

Bus 001 Device 005: ID 0bda:b820 Realtek Semiconductor Corp. 802.11ac WLAN Adapter

Kernel log:

[   42.809417] usb 1-1: new high-speed USB device number 5 using xhci_hcd
[   42.962866] usb 1-1: New USB device found, idVendor=0bda, idProduct=b820, bcdDevice= 2.00
[   42.962873] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   42.962876] usb 1-1: Product: 802.11ac WLAN Adapter
[   42.962878] usb 1-1: Manufacturer: Realtek
[   42.969089] Bluetooth: hci1: RTL: examining hci_ver=08 hci_rev=000c lmp_ver=08 lmp_subver=8821
[   42.969913] Bluetooth: hci1: RTL: rom_version status=0 version=1
[   42.969918] Bluetooth: hci1: RTL: loading rtl_bt/rtl8821c_fw.bin
[   42.970683] bluetooth hci1: firmware: direct-loading firmware rtl_bt/rtl8821c_fw.bin
[   42.970712] Bluetooth: hci1: RTL: loading rtl_bt/rtl8821c_config.bin
[   42.970885] bluetooth hci1: firmware: direct-loading firmware rtl_bt/rtl8821c_config.bin
[   42.970910] Bluetooth: hci1: RTL: cfg_sz 10, total sz 31990
[   43.116893] 8821cu: loading out-of-tree module taints kernel.
[   43.143008] 8821cu: unknown parameter 'rtw_beamform_cap' ignored
[   43.143274] RTW: module init start
[   43.143275] RTW: rtl8821cu v5.12.0-16-g5229e7f95.20210118_COEX20200730-5151
[   43.143277] RTW: build time: Apr 15 2022 12:52:48
[   43.143278] RTW: rtl8821cu BT-Coex version = COEX20200730-5151
[   43.143292] RTW: rtw_inetaddr_notifier_register
[   43.143313] RTW: 
               usb_endpoint_descriptor(0):
[   43.143315] RTW: bLength=7
[   43.143316] RTW: bDescriptorType=5
[   43.143317] RTW: bEndpointAddress=84
[   43.143318] RTW: wMaxPacketSize=512
[   43.143319] RTW: bInterval=0
[   43.143320] RTW: RT_usb_endpoint_is_bulk_in = 4
[   43.143320] RTW: 
               usb_endpoint_descriptor(1):
[   43.143321] RTW: bLength=7
[   43.143322] RTW: bDescriptorType=5
[   43.143323] RTW: bEndpointAddress=5
[   43.143324] RTW: wMaxPacketSize=512
[   43.143325] RTW: bInterval=0
[   43.143325] RTW: RT_usb_endpoint_is_bulk_out = 5
[   43.143326] RTW: 
               usb_endpoint_descriptor(2):
[   43.143327] RTW: bLength=7
[   43.143328] RTW: bDescriptorType=5
[   43.143329] RTW: bEndpointAddress=6
[   43.143330] RTW: wMaxPacketSize=512
[   43.143330] RTW: bInterval=0
[   43.143331] RTW: RT_usb_endpoint_is_bulk_out = 6
[   43.143332] RTW: 
               usb_endpoint_descriptor(3):
[   43.143333] RTW: bLength=7
[   43.143333] RTW: bDescriptorType=5
[   43.143334] RTW: bEndpointAddress=87
[   43.143335] RTW: wMaxPacketSize=64
[   43.143336] RTW: bInterval=3
[   43.143337] RTW: RT_usb_endpoint_is_int_in = 7, Interval = 3
[   43.143338] RTW: 
               usb_endpoint_descriptor(4):
[   43.143339] RTW: bLength=7
[   43.143339] RTW: bDescriptorType=5
[   43.143340] RTW: bEndpointAddress=8
[   43.143341] RTW: wMaxPacketSize=512
[   43.143342] RTW: bInterval=0
[   43.143343] RTW: RT_usb_endpoint_is_bulk_out = 8
[   43.143343] RTW: nr_endpoint=5, in_num=2, out_num=3

[   43.143345] RTW: USB_SPEED_HIGH
[   43.143346] RTW: CHIP TYPE: RTL8821C
[   43.143367] RTW: [HALMAC]55772M
               HALMAC_MAJOR_VER = 1
               HALMAC_PROTOTYPE_VER = 6
               HALMAC_MINOR_VER = 6
               HALMAC_PATCH_VER = 22
[   43.143754] RTW: Chip Version Info: CHIP_8821C_U2_1T1R_RomVer(1)
[   43.143756] RTW: config_chip_out_EP OutEpQueueSel(0x07), OutEpNumber(3)
[   43.143839] RTW: HW EFUSE
[   43.143842] RTW: 0x000: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.143854] RTW: 0x010: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.143866] RTW: 0x020: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.143878] RTW: 0x030: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.143889] RTW: 0x040: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.143901] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.143913] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.143924] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.143936] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.143948] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.143959] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.143971] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.143983] RTW: 0x0C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.143994] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144006] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144018] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144029] RTW: 0x100: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144041] RTW: 0x110: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144053] RTW: 0x120: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144064] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144076] RTW: 0x140: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144088] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144099] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144111] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144123] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144134] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144146] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144158] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144170] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144181] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144193] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144204] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   43.144226] RTW: rtw_read_efuse_from_file /system/etc/wifi/wifi_efuse_8821cu.map is not readable
[   43.144228] RTW: rtl8821c_read_efuse: <WARN> invalid phy efuse and read from file fail, will use driver default!!
[   43.144229] RTW: EEPROM ID = 0xffff
[   43.144230] RTW: <WARN> EEPROM ID is invalid!!
[   43.144230] RTW: EEPROM Version = 1
[   43.144242] RTW: rtw_read_macaddr_from_file /data/wifimac.txt is not readable
[   43.144243] RTW: EEPROM Regulatory=0x00
[   43.144244] RTW: EEPROM Board Type=0x00
[   43.144245] RTW: EEPROM Disable BT-coex, ant_num=1
[   43.144247] RTW: hal_com_config_channel_plan country code:"BR" with chplan:0x62
[   43.144248] RTW: EEPROM crystal_cap=0x00
[   43.144249] RTW: EEPROM ThermalMeter=0x12
[   43.144250] RTW: EEPROM Customer ID=0x00
[   43.144251] RTW: EEPROM SupportRemoteWakeup=0
[   43.144252] RTW: ERROR 
               
               Empty EFUSE with unknown REF type!!

[   43.144253] RTW: ERROR please program efuse or specify correct RFE type.
[   43.144253] RTW: ERROR cmd: insmod rtl8821cx.ko rtw_RFE_type=<rfe_type>

[   43.144254] RTW: rtw_hal_read_chip_info in 0 ms
[   43.144261] RTW: rtw_usb_primary_adapter_init Failed!
[   43.144303] usbcore: registered new interface driver rtl8821cu
[   43.144305] RTW: module init ret=0
[   43.452919] Bluetooth: hci1: RTL: fw version 0x829a7644

morrownr, 2022-04-15:
Hi

I need you to read the pinned issue titled: Problem Reporting Guidelines - Please Read.

Can you elaborate on "My dongle stopped?" Does that indicate that it worked previous to this problem?

JuniorPolegato, 2022-04-15:
Hi!
It was works fine, but about 4 weeks ago its stopped, I tried with Windows also, but nothing, without life.
So I think it lost the firmware (efuse), then I can try reactive it if I get /system/etc/wifi/wifi_efuse_8821cu.map.
About secure boot, I tried with this disabled, and enabled now signed for me.
Thanks.

root@debian:~# sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev
Linux debian 5.16.0-6-amd64 #1 SMP PREEMPT Debian 5.16.18-1 (2022-03-29) x86_64 GNU/Linux
SecureBoot enabled
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04f2:b469 Chicony Electronics Co., Ltd HD WebCam
Bus 001 Device 004: ID 04ca:300b Lite-On Technology Corp. Atheros AR3012 Bluetooth
Bus 001 Device 005: ID 0bda:b820 Realtek Semiconductor Corp. 802.11ac WLAN Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: acer-wireless: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: acer-bluetooth: Bluetooth
	Soft blocked: no
	Hard blocked: no
2: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
3: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
4: hci1: Bluetooth
	Soft blocked: no
	Hard blocked: no
nvidia-legacy-390xx/390.147, 5.10.0-9-amd64, x86_64: installed (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!)
nvidia-legacy-390xx/390.147, 5.15.0-3-amd64, x86_64: installed (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!)
nvidia-legacy-390xx/390.147, 5.16.0-3-amd64, x86_64: installed (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!)
nvidia-legacy-390xx/390.147, 5.16.0-4-amd64, x86_64: installed (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!)
nvidia-legacy-390xx/390.147, 5.16.0-5-amd64, x86_64: installed (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!)
nvidia-legacy-390xx/390.147, 5.16.0-6-amd64, x86_64: installed
rtl8821cu/5.12.0, 5.16.0-6-amd64, x86_64: installed (WARNING! Diff between built and installed module!)
phy#0
	Interface wlp3s0
		ifindex 3
		wdev 0x1
		addr 5c:c9:d3:54:90:51
		ssid Junior 2.4
		type managed
		channel 11 (2462 MHz), width: 20 MHz, center1: 2462 MHz
		txpower 16.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
root@debian:~#

morrownr, 2022-04-15:
Hello again

I've ran a few things through my mind and here is all I can come up with:

If your system has more than one usb port, change the adapter to the other port and reboot. Any better?

If that doesn't work, you may simply have an adapter that is at its end of life. The firmware for this driver is included in the driver so I don't think chasing this option down will help.

If you are able to buy a new adapter maybe my repo listed below can help:

https://github.com/morrownr/USB-WiFi

If you have questions, you can post in the above repo.

Regard

JuniorPolegato, 2022-04-16:
Yeah, it's over! I tried everything in these 4 weeks, many versions of ports and computers, many drivers versions for Linux and Windows, the last chance would be this wifi_efuse_8821cu.map file. At least, I'm going to use as bluetooth dongle. Thanks so much.

morrownr, 2022-04-22:
Good luck. Closing issue.

closed 2022-04-22 by morrownr