morrownr/8821cu-20210118 #17: Had to re-install the driver to get the module working

Shravansingh, 2022-05-18:
Hello,

So I had installed the driver back in December of 2021. And it was working fine on Ubuntu 20.04. But then somewhere around May 2022. It stopped working for the new batch of EDUP wifi modules. As soon as I installed the new driver it started working.

I suspect there is something with the new batch of EDUP modules that is the cause of this issue. Or a combination of both the driver and new batch. 
Now EDUP maintains that they have not changed dongle in any way. Is there a way to get log file that might point out what was the issue and why it is working with the new driver.

I don't understand dmesg that much. But if there are some log files that I could get, that would shed some light into this issue that might help me a lot.

```
blue@blue:~$ lsusb 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 14e1:3508 Dialogue Technology Corp. 
Bus 001 Device 005: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 006: ID 0bda:c820 Realtek Semiconductor Corp. USB Receiver
Bus 001 Device 004: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
blue@blue:~$ mokutil --sb-state 
This system doesn't support Secure Boot
blue@blue:~$ rfkill list all
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
blue@blue:~$ dkms status
rtl8821cu, 5.12.0, 5.13.0-41-generic, x86_64: installed
rtl8821cu, 5.8.1.7, 5.13.0-40-generic, x86_64: installed
rtl8821cu, 5.8.1.7, 5.13.0-41-generic, x86_64: built
blue@blue:~$ iw dev
phy#0
	Interface wlxe84e0696e702
		ifindex 3
		wdev 0x1
		addr e8:4e:06:96:e7:02
		ssid Blue
		type managed
		channel 157 (5785 MHz), width: 40 MHz, center1: 5795 MHz
		txpower 17.00 dBm
blue@blue:~$ sudo uname -a
Linux blue 5.13.0-41-generic #46~20.04.1-Ubuntu SMP Wed Apr 20 13:16:21 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
blue@blue:~
```

Shravansingh, 2022-05-18:
Some more information. So I re-imaged the system. And the information for my earlier image is shown below

```
blue@blue:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 14e1:3508 Dialogue Technology Corp. 
Bus 001 Device 005: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 007: ID 0bda:c820 Realtek Semiconductor Corp. USB2.0 Hub
Bus 001 Device 004: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
blue@blue:~$ dkms status
rtl8821cu, 5.8.1.7, 5.13.0-28-generic, x86_64: installed
blue@blue:~$ iw dev
phy#0
	Interface wlxe84e0696e702
		ifindex 3
		wdev 0x1
		addr e8:4e:06:96:e7:02
		ssid Blue
		type managed
		txpower 12.00 dBm
blue@blue:~$ mokutil --sb-state
This system doesn't support Secure Boot
blue@blue:~$ sudo una
unalias              uname                unattended-upgrade   unattended-upgrades  
blue@blue:~$ sudo uname -a
[sudo] password for blue: 
Linux blue 5.13.0-28-generic #31~20.04.1-Ubuntu SMP Wed Jan 19 14:08:10 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
blue@blue:~$ 
``` 

In this version too only one of the EDUP usb dongle works. Rest of them don't work after a soft reboot.
What I find interesting is why would only one dongle work and not all of them? and why is only that one working?

morrownr, 2022-05-18:
Hi @Shravansingh 

> blue@blue:~$ dkms status
rtl8821cu, 5.12.0, 5.13.0-41-generic, x86_64: installed
rtl8821cu, 5.8.1.7, 5.13.0-40-generic, x86_64: installed
rtl8821cu, 5.8.1.7, 5.13.0-41-generic, x86_64: built

That indicates to me that the old driver was not removed before installing this new version of the driver. Please run the following:

$ ./remove-driver.sh

Now run and post the results of:

$ dkms status

Having more than one version of the driver installed usually does not end well.

Regards

Shravansingh, 2022-05-18:
I did that. My question this version was working fine for some time `rtl8821cu, 5.8.1.7, 5.13.0-40-generic, x86_64: installed`
And then stopped working for all but one. 
As soon as I install `rtl8821cu, 5.12.0, 5.13.0-41-generic, x86_64: installed` all the dongle work again.
What logs should I get to understand what could be the reason behind it?

Shravansingh, 2022-05-19:
> I want to compare the device IDs such as the numbers after ID below"

```
## Not Working with 5.8.1.7 
## Working with 5.12.0

blue@blue:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 14e1:3508 Dialogue Technology Corp. 
Bus 001 Device 005: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 006: ID 0bda:c820 Realtek Semiconductor Corp. USB2.0 Hub
Bus 001 Device 004: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
blue@blue:~$

[    3.769322] usb 1-1.1.1: new high-speed USB device number 6 using xhci_hcd
[    3.922817] logitech-djreceiver 0003:046D:C52B.0003: hiddev0,hidraw0: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-2/input2
[    3.969726] usb 1-1.1.1: New USB device found, idVendor=0bda, idProduct=c820, bcdDevice= 2.00
[    3.969737] usb 1-1.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.969745] usb 1-1.1.1: Product: 802.11ac NIC
[    3.969750] usb 1-1.1.1: Manufacturer: Realtek
[    3.969756] usb 1-1.1.1: SerialNumber: FF
[    6.667238] usbcore: registered new interface driver btusb
[    7.012572] usbcore: registered new interface driver rtl8821cu
[    8.208904] audit: type=1400 audit(1652970167.920:5): apparmor="STATUS" operation="profile_load" profile="unconfined" name="ippusbxd" pid=536 comm="apparmor_parser"

## Not Working with 5.8.1.7 
## Working with 5.12.0

blue@blue:~$ dmesg | grep usb
[  264.508083] usb 1-1.1.1: USB disconnect, device number 6
[  264.911070] usb 1-1.1.1: new high-speed USB device number 8 using xhci_hcd
[  274.815063] usb 1-1.1.1: new high-speed USB device number 9 using xhci_hcd
[  275.015480] usb 1-1.1.1: New USB device found, idVendor=0bda, idProduct=c820, bcdDevice= 2.00
[  275.015493] usb 1-1.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  275.015501] usb 1-1.1.1: Product: 802.11ac NIC
[  275.015506] usb 1-1.1.1: Manufacturer: Realtek
[  275.015512] usb 1-1.1.1: SerialNumber: FF
blue@blue:~$ lsusb 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 14e1:3508 Dialogue Technology Corp. 
Bus 001 Device 005: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 009: ID 0bda:c820 Realtek Semiconductor Corp. USB2.0 Hub
Bus 001 Device 004: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
blue@blue:~$
 
## Not Working with 5.8.1.7 
## Working with 5.12.0

[  428.860129] usb 1-1.1.1: USB disconnect, device number 9
[  429.283067] usb 1-1.1.1: new full-speed USB device number 10 using xhci_hcd
[  429.463072] usb 1-1.1.1: device descriptor read/64, error -32
[  440.703060] usb 1-1.1.1: new high-speed USB device number 11 using xhci_hcd
[  441.063058] usb 1-1.1.1: new full-speed USB device number 12 using xhci_hcd
[  441.263273] usb 1-1.1.1: not running at top speed; connect to a high speed hub
[  441.264268] usb 1-1.1.1: New USB device found, idVendor=0bda, idProduct=c820, bcdDevice= 2.00
[  441.264280] usb 1-1.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  441.264287] usb 1-1.1.1: Product: 802.11ac NIC
[  441.264293] usb 1-1.1.1: Manufacturer: Realtek
[  441.264298] usb 1-1.1.1: SerialNumber: FF
blue@blue:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 14e1:3508 Dialogue Technology Corp. 
Bus 001 Device 005: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver
[... 109 lines cut ...]
Version: 5.12.0
Kernel:  5.13.0-28-generic (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

8821cu.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.13.0-28-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod......

DKMS: uninstall completed.

------------------------------
Deleting module version: 5.12.0
completely from the DKMS tree.
------------------------------
Done.
blue@blue:~$ dkms status
blue@blue:~$ 
```

Shravansingh, 2022-05-19:
> Once installed and rebooted. See if the adapters are working.
Yes, I did do that and all of them are working now.

morrownr, 2022-05-19:
Okay. That is good to hear. That tells me that you have adapters with at least 2 different types of circuitry for the antenna isolation and the BT coexistence mechanism. EDUP may not even know that. It also tells me that I need to pull that option from the default configuration and simply document the issue as users with problems that require this option will simply have to play with it. Thank you Realtek.

I appreciate you taking the time help get this worked out.

Regards

morrownr, 2022-05-19:
Yes, I like to work with bash.

A good place to get me is: morrownr at gmail.com

Note: After posting this comment, I will then turn around and remove my email address so as to limit public exposure but it should come to you in email.

Regards

Shravansingh, 2022-05-19:
yes, Done Thank you

closed 2022-05-19 by Shravansingh