morrownr/8821cu-20210118 #19: adapter not recognized - 0bda:1a2b Realtek - raspberry pi 4 b, ubuntu server lts 20.04 64-bit

blacksteel1288, 2022-05-29:
Hi,

After following the install steps, my wifi adapter still is not recognized.  This is a known good adapter that was previously working on a Raspberry Pi 4 B running Raspberry Pi OS 32-bit (buster) using the drivers from http://downloads.fars-robotics.net/wifi-drivers/8821cu-drivers/

I am trying to get it working on the same device using the Raspberry Pi version of Ubuntu 20.04.4 LTS 64-bit

Here's the diagnostic info including the Realtek ID:

```
ubuntu@ubuntu:~$ sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev
Linux ubuntu 5.4.0-1060-raspi #68-Ubuntu SMP PREEMPT Tue Apr 26 09:31:19 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
EFI variables are not supported on this system
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0bda:1a2b Realtek Semiconductor Corp. USB2.0 Hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
rtl8821cu, 5.12.0, 5.4.0-1060-raspi, aarch64: installed
```

Here's a link to the adapter where it was purchased:
https://www.amazon.com/gp/product/B07LFC7VDC/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

blacksteel1288, 2022-05-29:
Ah, cancel that.  Solved my own problem.  This was a usb_modeswitch problem, solved by:

```
sudo nano /lib/udev/rules.d/40-usb_modeswitch.rules
```

Then add these lines:

```
# Realtek 8211CU Wifi AC USB
ATTR{idVendor}=="0bda", ATTR{idProduct}=="1a2b", RUN+="/usr/sbin/usb_modeswitch -K -v 0bda -p 1a2b"
```

After that the real wifi device ID is shown in ```lsusb``` which turned out to be ```Bus 001 Device 004: ID 0bda:c811 Realtek Semiconductor Corp. USB2.0 Hub``` which is recognized by the driver.

morrownr, 2022-05-29:
That old version of Ubuntu may not have an up to date version of the usb_modeswitch data file.

I just looked in the current version of `/lib/udev/rules.d/40-usb_modeswitch.rules` as shown in RasPiOS 2022-04-04 and I found:

```
# D-Link DWA-171 Wifi Dongle
ATTR{idVendor}=="0bda", ATTR{idProduct}=="1a2b", RUN+="usb_modeswitch '/%k'"
```

That should trigger the switch unless something else is wrong.

Regards

blacksteel1288, 2022-05-30:
For some reason that usb_modeswitch line didn't work on this version of Ubuntu, but my original edit does.