morrownr/8821cu-20210118 #20: Fails install in Kali

RChadwick7, 2022-06-11:
I'm trying to install the driver on Kali in a VMWare VM. It's a direct download, with updates installed.
----------------------

Linux kali 5.17.0-kali3-amd64 #1 SMP PREEMPT Debian 5.17.11-1kali1 (2022-05-30) x86_64 GNU/Linux
EFI variables are not supported on this system
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 004: ID 0e0f:0008 VMware, Inc. Virtual Bluetooth Adapter
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
0: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
rtl8821cu/5.12.0: added

--------------------
When I try to install the driver, the script throws an error: 
--------------------

Running install-driver.sh version 20220108
Operating System: Kali GNU/Linux Rolling
5.17.0-kali3-amd64
x86_64
Starting installation...
Copying source files to: /usr/src/rtl8821cu-5.12.0
Copying 8821cu.conf to: /etc/modprobe.d
Creating symlink /var/lib/dkms/rtl8821cu/5.12.0/source -> /usr/src/rtl8821cu-5.12.0

Building module:
cleaning build area...
'make' -j4 KVER=5.17.0-kali3-amd64 KSRC=/lib/modules/5.17.0-kali3-amd64/build.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.17.0-kali3-amd64 (x86_64)
Consult /var/lib/dkms/rtl8821cu/5.12.0/build/make.log for more information.
An error occurred. dkms build error = 10
Please report this error.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh

----------------------------------------------
Here is the make.log:

DKMS make.log for rtl8821cu-5.12.0 for kernel 5.17.0-kali3-amd64 (x86_64)
Sat Jun 11 01:41:00 PM EDT 2022
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.17.0-kali3-amd64/build M=/var/lib/dkms/rtl8821cu/5.12.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.17.0-kali3-amd64'
/bin/sh: 1: bc: not found
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_io.o
/var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_debug.c: In function ‘dump_drv_version’:
/var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_debug.c:45:69: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
   45 |         RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                                                     ^
In file included from /usr/src/linux-headers-5.17.0-kali3-common/include/linux/kernel.h:29,
                 from /usr/src/linux-headers-5.17.0-kali3-common/arch/x86/include/asm/percpu.h:27,
                 from /usr/src/linux-headers-5.17.0-kali3-common/arch/x86/include/asm/current.h:6,
                 from /usr/src/linux-headers-5.17.0-kali3-common/arch/x86/include/asm/processor.h:17,
                 from /usr/src/linux-headers-5.17.0-kali3-common/arch/x86/include/asm/timex.h:5,
                 from /usr/src/linux-headers-5.17.0-kali3-common/include/linux/timex.h:65,
                 from /usr/src/linux-headers-5.17.0-kali3-common/include/linux/time32.h:13,
                 from /usr/src/linux-headers-5.17.0-kali3-common/include/linux/time.h:60,
                 from /usr/src/linux-headers-5.17.0-kali3-common/include/linux/stat.h:19,
                 from /usr/src/linux-headers-5.17.0-kali3-common/include/linux/module.h:13,
                 from /var/lib/dkms/rtl8821cu/5.12.0/build/include/basic_types.h:76,
                 from /var/lib/dkms/rtl8821cu/5.12.0/build/include/drv_types.h:26,
                 from /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_debug.c:17:
/var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_debug.c:45:51: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
   45 |         RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                                   ^~~~~~~~
/usr/src/linux-headers-5.17.0-kali3-common/include/linux/printk.h:418:33: note: in definition of macro ‘printk_index_wrap’
  418 |                 _p_func(_fmt, ##__VA_ARGS__);                           \
      |                                 ^~~~~~~~~~~
/var/lib/dkms/rtl8821cu/5.12.0/build/include/rtw_debug.h:85:26: note: in expansion of macro ‘printk’
   85 |         #define _dbgdump printk
      |                          ^~~~~~
/var/lib/dkms/rtl8821cu/5.12.0/build/include/rtw_debug.h:252:25: note: in expansion of macro ‘RTW_PRINT’
  252 |                         RTW_PRINT(fmt, ##arg); \
      |                         ^~~~~~~~~
/var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_debug.c:45:9: note: in expansion of macro ‘RTW_PRINT_SEL’
   45 |         RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |         ^~~~~~~~~~~~~
/var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_debug.c:45:61: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
   45 |         RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                                             ^~~~~~~~
/usr/src/linux-headers-5.17.0-kali3-common/include/linux/printk.h:418:33: note: in definition of macro ‘printk_index_wrap’
  418 |                 _p_func(_fmt, ##__VA_ARGS__);                           \
      |                                 ^~~~~~~~~~~
/var/lib/dkms/rtl8821cu/5.12.0/build/include/rtw_debug.h:85:26: note: in expansion of macro ‘printk’
   85 |         #define _dbgdump printk
      |                          ^~~~~~
/var/lib/dkms/rtl8821cu/5.12.0/build/include/rtw_debug.h:252:25: note: in expansion of macro ‘RTW_PRINT’
  252 |                         RTW_PRINT(fmt, ##arg); \
      |                         ^~~~~~~~~
/var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_debug.c:45:9: note: in expansion of macro ‘RTW_PRINT_SEL’
   45 |         RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |         ^~~~~~~~~~~~~
In file included from /var/lib/dkms/rtl8821cu/5.12.0/build/include/drv_types.h:60,
                 from /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_debug.c:17:
/var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_debug.c:45:51: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
   45 |         RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
      |                                                   ^~~~~~~~
/var/lib/dkms/rtl8821cu/5.12.0/build/include/rtw_debug.h:254:46: note: in definition of macro ‘RTW_PRINT_SEL’
  254 |                         _seqdump(sel, fmt, ##arg) /*rtw_warn_on(1)*/; \
      |                                              ^~~
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_mlme.o
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-5.17.0-kali3-common/scripts/Makefile.build:293: /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_debug.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-5.17.0-kali3-common/Makefile:1855: /var/lib/dkms/rtl8821cu/5.12.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.17.0-kali3-amd64'
make: *** [Makefile:2498: modules] Error 2

-----------------
Any ideas?
Thanks!

morrownr, 2022-06-11:
A quick and dirty look tells me the package `bc` is not installed.

> /bin/sh: 1: bc: not found

I may need to investigate why that all of the sudden that came up.

Nevertheless, you can install it via:

$ sudo apt install bc

You will want to run the following to clean things up before trying again:

$ sudo ./remove-driver.sh

I am interested in what version of gcc you are using and what iso of Kali you installed if you would be so kind as to provide that information.

Let me know.

morrownr, 2022-07-05:
status?

KoryMac, 2022-07-08:
I manually installed bc and the install finished properly (was having this same issue).

morrownr, 2022-07-08:
I thought I had this sorted out long ago but stuff happens and it does not cause problems if we tell apt to install something that is already installed anyway. I have revised the installation steps in README to add `bc` for Kali.

Thanks for the help fixing this.