morrownr/8821cu-20210118 #22: Pi 2B v1.1 - Volumio 3.301 (Buster)

ArturNadolski, 2022-06-15:
Running install-driver.sh version 20220108
Raspbian GNU/Linux 10 (buster)
5.10.92-v7+
armv7l
Starting installation...
Copying source files to: /usr/src/rtl8821cu-5.12.0
Copying 8821cu.conf to: /etc/modprobe.d

Creating symlink /var/lib/dkms/rtl8821cu/5.12.0/source ->
                 /usr/src/rtl8821cu-5.12.0

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.10.92-v7+ cannot be found at
/lib/modules/5.10.92-v7+/build or /lib/modules/5.10.92-v7+/source.
An error occurred. dkms build error = 1
Please report this error.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh

Linux volumio 5.10.92-v7+ #1514 SMP Mon Jan 17 17:36:39 GMT 2022 armv7l GNU/Linux
-bash: mokutil: command not found
Bus 001 Device 004: ID 0bda:c811 Realtek Semiconductor Corp.
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
-bash: rfkill: command not found
-bash: dkms: command not found
-bash: iw: command not found

morrownr, 2022-06-16:
Hi @ArturNadolski 

Your dev environment is not setup correctly. This may be hard to sort out. Any possibility of me convincing you to download and burn a more modern version of the OS? Raspberry Pi OS (2022-04-04) 32 bit.

For the installation steps, make sure you are very careful on steps 3 and 9:

For Step 3:

$ sudo apt install -y raspberrypi-kernel-headers bc build-essential dkms git

For Step 9:

./ARM_RPI.sh

Let me know how it goes.

Regards

morrownr, 2022-07-05:
status?

closed 2022-07-08 by morrownr