morrownr/8821cu-20210118 #23: How rtl8811cu works on openwrt?

liuhong785738008, 2022-07-04:
I am running OpenWRT-21.02.3 on MT7628, using RTL8811cu to connect to WIFI, but I have a problem, every time I configure WIFI, I have to restart the OpenWrt system to take effect, restart /etc/init.d/network does not take effect. Why does this happen?

morrownr, 2022-07-04:
Hi @liuhong785738008 

Let me see if I understand:

You have a usb wifi adapter based on a rtl8811cu chipset. This adapter is plugged into a usb port on the mt7626 based router? Or is the adapter plugged into something else? What driver are you using for the usb wifi adapter?

FYI: I am a user of OpenWRT but do not compile it or work on it so, for now, my knowledge is limited to user knowledge.

Regards

liuhong785738008, 2022-07-05:
Hi @morrownr 
Yes, I have a router for MT7628 and I connect an RTL8811cu chipset to the MT7628 using USB. I used the driver you provided, and I can use WIFI through RTL8811cu, but I am not satisfied with the fact that I have to restart the system after configuring WIFI each time to take effect. Could you help me?
Thanks!

morrownr, 2022-07-05:
> Could you help me?

I hope you are not in a hurry. OpenWRT is designed to use limited space so it is tailored in many areas in ways that are different than Ubuntu or RasPiOS or most regular distros. 

I'd like to know how you got this driver working on OpenWRT. I'm impressed. If you can help me with that, maybe I can help figure out the answer to your question.

Regards

liuhong785738008, 2022-07-06:
@morrownr 

I made changes based on this projectI made changes based on this project [https://github.com/XinRoom/Openwrt_rtl8811cu.git](https://github.com/XinRoom/Openwrt_rtl8811cu.git)

liuhong785738008, 2022-07-06:
I added a patch as follows：

--- a/Makefile
+++ b/Makefile
@@ -55,7 +55,7 @@ CONFIG_GSPI_HCI = n
 CONFIG_AP_MODE = y 
 CONFIG_P2P = y 
 CONFIG_MP_INCLUDED = y 
-CONFIG_CONCURRENT_MODE = n
+CONFIG_CONCURRENT_MODE = y
 CONFIG_POWER_SAVING = y 
 CONFIG_IPS_MODE = default
 CONFIG_LPS_MODE = default
@@ -140,7 +140,7 @@ CONFIG_LAYER2_ROAMING = y
 #bit0: ROAM_ON_EXPIRED, #bit1: ROAM_ON_RESUME, #bit2: ROAM_ACTIVE
 CONFIG_ROAMING_FLAG = 0x3 
 ###################### Platform Related #######################
-CONFIG_PLATFORM_I386_PC = y
+CONFIG_PLATFORM_I386_PC = n
 CONFIG_PLATFORM_ARM_RPI = n 
 CONFIG_PLATFORM_ARM64_RPI = n 
 CONFIG_PLATFORM_ANDROID_X86 = n 
@@ -152,6 +152,7 @@ CONFIG_PLATFORM_ARM_S3C6K4 = n
 CONFIG_PLATFORM_MIPS_RMI = n 
 CONFIG_PLATFORM_RTD2880B = n 
 CONFIG_PLATFORM_MIPS_AR9132 = n 
+CONFIG_PLATFORM_MIPS_MT7628 = y
 CONFIG_PLATFORM_RTK_DMP = n 
 CONFIG_PLATFORM_MIPS_PLM = n 
 CONFIG_PLATFORM_MSTAR389 = n 
@@ -1357,6 +1358,18 @@ INSTALL_PREFIX :=
 STAGINGMODDIR := /lib/modules/$(KVER)/kernel/drivers/staging
 endif
 
+ifeq ($(CONFIG_PLATFORM_MIPS_MT7628), y)
+ifeq ($(CONFIG_CONCURRENT_MODE), y)
+EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE
+endif
+EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
+EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT
+ARCH := mips
+#CROSS_COMPILE = mipsel-openwrt-linux-musl-
+CROSS_COMPILE ?=
+KSRC := $(TOPDIR)/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/linux-5.4.188
+endif
+
 ### START RASPBERRY PI OS SUPPORT
 
 ifeq ($(CONFIG_PLATFORM_ARM_RPI), y)

closed 2022-10-05 by morrownr