morrownr/8821cu-20210118 #24: D-Link DWA-171 AC600

fzoric8, 2022-07-04:
Hi everyone, 

I recently bought this USB dongle quite naively believing that everything will go just fine. As i plugged it in into my linux PC, 
I figured out I'll need to put in some work before it actually works. 

I've tried following instructions from this [page](https://ubuntuforums.org/archive/index.php/t-2462286.html) however didn't manage to get it working. 

Commands and their respective output is following: 
- cmd: `uname -r` 
- out: `Linux goran-GL553VD 5.4.0-120-generic #136~18.04.1-Ubuntu SMP Fri Jun 10 18:00:44 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`
- cmd:  `mokutil --sb-state` 
- out: `SecureBoot disabled`
- cmd: `lsusb`
- out: 
```
 Bus 002 Device 002: ID 0bda:0423 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 8087:0a2a Intel Corp. 
Bus 001 Device 004: ID 0bda:57f5 Realtek Semiconductor Corp. 
Bus 001 Device 003: ID 046d:0825 Logitech, Inc. Webcam C270
Bus 001 Device 007: ID 0b05:1854 ASUSTek Computer, Inc. 
Bus 001 Device 008: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 006: ID 0bda:5423 Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 0bda:1a2b Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub 
```
- cmd: `rfkill list all`
- out:
```
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
```
- cmd: `dkms status: 
- out: 
```evdi, 1.11.0, 5.4.0-113-generic, x86_64: installed
evdi, 1.11.0, 5.4.0-120-generic, x86_64: installed
evdi, 5.2.14: added
nvidia, 470.129.06, 5.4.0-113-generic, x86_64: installed
nvidia, 470.129.06, 5.4.0-120-generic, x86_64: installed
rtl8821cu, 5.12.0, 5.4.0-120-generic, x86_64: installed
```
- cmd: `iw dev`: 
- out:
```
phy#0
	Unnamed/non-netdev interface
		wdev 0x2
		addr e4:42:a6:6b:76:06
		type P2P-device
		txpower 0.00 dBm
	Interface wlp2s0
		ifindex 3
		wdev 0x1
		addr e4:42:a6:6b:76:05
		ssid dlink-65C0
		type managed
		channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
		txpower 22.00 dBm
```
- cmd : `dmesg | grep usb` 
- out: 
```
[  368.310547] usb 1-10.3: USB disconnect, device number 10
[ 1390.240303] usb 1-1: USB disconnect, device number 2
[ 1392.340366] usb 1-1: new high-speed USB device number 11 using xhci_hcd
[ 1392.488976] usb 1-1: New USB device found, idVendor=0bda, idProduct=1a2b, bcdDevice= 2.00
[ 1392.488978] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1392.488979] usb 1-1: Product: DISK
[ 1392.488980] usb 1-1: Manufacturer: D-Link Corporation
[ 1392.490272] usb-storage 1-1:1.0: USB Mass Storage device detected
[ 1392.490593] scsi host3: usb-storage 1-1:1.0
```

I know that i have one WiFi interface already running, but I would really need to have 2 different WiFi interfaces and that's why I'm trying to get it up and running. 

Thank you for your help.

fzoric8, 2022-07-04:
I've managed to resolve the issue. 

After writing the issue I've rebooted and entered the Windows to start dongle. 

After rebooting from Windows, it managed it seems to have booted up correctly on Linux and now I have 2 WiFi interfaces running which was exactly what I wanted.

morrownr, 2022-07-04:
Hi @fzoric8 

I don't use Windows but I have heard that if you turn off Fastboot in Windows and then do a cold boot, this might fix this issue. Your mileage may vary.

USB WiFi adapters are handy and flexible. Getting the best one for your use with Linux is getting easier but there are things to be aware of. If you go to the Main Menu of this site and read menu items 1 and 2, you will have a lot of good information.

https://github.com/morrownr/USB-WiFi

If you are still within the return window for your adapter, you might consider returning it so as to purchase a good adapter that uses in-kernel drivers.

Regards

fzoric8, 2022-07-04:
Thank you @morrownr, I have unfortunately encountered this repo a little bit too late. 

I've managed to get it working, and it will be fine for occasional usage in my case.  

Thanks you for your effort, it is highly appreciated. 

If anyone, anytime reads this and finds out what makes dongle work after rebooting from windows, let me know. Cheers :))

closed 2022-07-04 by fzoric8

morrownr, 2022-07-05:
If you are interested, we can pursue some others issues.

You are running a pretty old kernel with a distro that I do not recognize so I can't be too specific.

When you boot up into Linux in the situation where you do not get a wifi interface, do you show an extra storage device such as a cd or flash driver or harddrive? The reason I ask is because this could be a multi-state adapter.

Does the adapter have bluetooth capability?

It is a USB2 device. Are you running it in a USB2 port? Have you tried different ports?

Are you using `$ iw dev` to check if a wifi interface is being created or are you assuming nothing happened when you can't connect?

It would help me help you if you were running a more modern Linux distro. Would you be willing to upgrade?

Regards

fzoric8, 2022-07-05:
Yes we can in case someone else has same issues as me. 

I'm aware that I'm running old Linux distro, however due to certain circumstances I don't want/can't upgrade it at the moment.
 
On another PC I'm running Linux Ubuntu 20.04 so I could basically try to see what happens there after driver installation. 

Regarding other questions, I'm answering them as follows: 
* When you boot up into Linux in the situation where you do not get a wifi interface, do you show an extra storage device such as a cd or flash driver or harddrive? --> Yes, I'm seeing flash device named DWA-171 
* Does the adapter have bluetooth capability --> Aside from mention of Bluetooth on page 25 of this [documentation](https://www.dlink.com.br/wp-content/uploads/2018/11/DWA-171_C1_Manual_v3.00_WW.pdf) I'm guessing **it doesn't have bluetooth capability**
* When I use `iw dev` there's only one wifi interface (integrated network card) so I'm assuming nothing happened.

morrownr, 2022-07-07:
> On another PC I'm running Linux Ubuntu 20.04 so I could basically try to see what happens there after driver installation.

Good idea.

> Yes, I'm seeing flash device named DWA-171

It is a multi-state adapter. Need to keep that in mind.