morrownr/8821cu-20210118 #25: Not sure how to resolve this?

Jibun-no-Kage, 2022-07-06:
Running install-driver.sh version 20220705
5.15.52-v7+
armv7l
Starting installation...
Copying source files to: /usr/src/rtl8821cu-5.12.0
Copying 8821cu.conf to: /etc/modprobe.d

Creating symlink /var/lib/dkms/rtl8821cu/5.12.0/source ->
                 /usr/src/rtl8821cu-5.12.0

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.15.52-v7+ cannot be found at
/lib/modules/5.15.52-v7+/build or /lib/modules/5.15.52-v7+/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
An error occurred. dkms build error = 1
Please report this error.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh

DoraemonAikouka, 2022-07-06:
I have the same problem.

DoraemonAikouka, 2022-07-06:
Run rpi-source(https://github.com/RPi-Distro/rpi-source) and then 
`sudo ./install-driver-no-dkms.sh` 
and solved the problem.

Jibun-no-Kage, 2022-07-07:
Sure... 3B Pi, Pi OS 11...
# uname -a
Linux eccentric 5.15.52-v7+ #1570 SMP Mon Jul 4 13:29:14 BST 2022 armv7l GNU/Linux

# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

# /opt/vc/bin/vcgencmd version
Jul  4 2022 14:41:54
Copyright (c) 2012 Broadcom
version 01db5c05be1b3594c8f040ed7c1ac2748890c2c8 (clean) (release) (start)

# cat /proc/cpuinfo [redacted]
Hardware        : BCM2835
Revision        : a52082
Serial          : 00000000f31936bf
Model           : Raspberry Pi 3 Model B Rev 1.2

I already had the various packages install, git, etc.  Including the current kernel headers.

I just downloaded the latest git url, did a git clone, moved into the downloaded directory, and executed dkms-install.sh script.  Done this before with no issues, so the failure to find the kernel headers was a surprise.

Oh, used 32bit variant.

Jibun-no-Kage, 2022-07-07:
I thought rpi-source was considered retired?  Or not officiail per se?

morrownr, 2022-07-07:
> I just downloaded the latest git url, did a git clone, moved into the downloaded directory, and executed dkms-install.sh script. Done this before with no issues, so the failure to find the kernel headers was a surprise.

It is a surprise to me also. It appears you are running RasPiOS 2022-04-04 32 bit on a RasPi3B. I have a RasPi3B and a RasPi4B. I test these drivers regularly with the 32 bit and 64 bit versions of RasPiOS on both Pis. My most recent of this driver was on the RasPi4B 64 bit about 2 weeks ago. No issues. That is not to say that I haven't broke something because that happens and I do work on these drivers regularly.

I can burn a fresh sd with 32 bit RasPiOS tomorrow and give it a test if you want.

The problem is with the headers. Now, what the issue is... is not clear.

Regards

DoraemonAikouka, 2022-07-07:
> I thought rpi-source was considered retired? Or not officiail per se? It still tries to use python2 which is long gone, so not sure that is a viable solution at all.

Yeah, I don't think it's a good solution either.
I wish there was another solution: ......

morrownr, 2022-07-07:
Test complete:

I burned a clean 32 GB sd card with rpi-imager.
OS: RasPiOS 32 bit 2022-04-04
Hardware: RasPi3B
Booted with newly burned sd, set appropriate settings and updated the system, then rebooted.

Open a terminal and proceeded to execute the following:

sudo apt update && sudo apt upgrade
sudo apt install -y raspberrypi-kernel-headers bc build-essential dkms git
mkdir -p ~/src
cd ~/src
git clone https://github.com/morrownr/8821cu-20210118.git
cd ~/src/8821cu-20210118
./ARM_RPI.sh
sudo ./install-driver.sh

Compile was clean.
After reboot, adapter (rtl8811cu chipset) worked as expected.

The instructions above are pulled directly from the `Installation Steps` section of the README.

Conclusion: I was unable to duplicate the issue. There is something different about your setup or there is an issue following the installation instructions. I will continue to help but I'm not seeing the problem here.

Regards

morrownr, 2022-07-07:
One last thing for now:

> I just downloaded the latest git url, did a git clone, moved into the downloaded directory, and executed dkms-install.sh script. Done this before with no issues, so the failure to find the kernel headers was a surprise.

I noticed this comment. 

There is no `dkms-install.sh` with this driver. There are 2 scripts:

install-driver.sh
install-driver-no-dkms.sh

>  I just downloaded the latest git url

I'll recommend against this. Please run the appropriate line as shown in the installation steps:

sudo apt install -y raspberrypi-kernel-headers bc build-essential dkms git

It does no harm to to attempt to install again. apt will simply tell you it is already installed and go on. On the other hand, you may think you have the right things installed but you do not.

The main problems I see with the Realtek out-of-kernel drivers:

- People using non-mainstream distros that have changes that blow things up.
- People not following the exact installation steps carefully.

Jibun-no-Kage, 2022-07-07:
Oh, I just, mistyped the script name.  Interesting that the clean build worked... which is really a good thing.  But clearly for more than just myself,, something is stepping on something.  I have some time this weekend, and will do a clean build as well.  Thanks for the validation and response to the issue I encountered.  I use a bash script which is hash table driven, to do all image builds, so I can step by step watch as each installation/customization is done, (not as complex as ansible, but similar idea, I used the same idea for deploy 10,000s of ESX (yes the older ESX servers years ago, before VMware moved to ESXi) so I should be able to spot the issue, or where the delta that impacts the install.  Truly, interesting, because other than standard packages and Node Red installation, nothing fancy on this build, I happen to have a NR flow that drives the wifi dongle to track changes in my local area, since most of my neighbors are constantly screwing with their wifi configurations, every day I capture over 100 2.4 and 5 GHz networks just with about 200 yard radius around my antenna location, so I have to routinely change my wifi as well.  If I could write a basic AI to do it... I would! LOL.  I may try at some point.

Jibun-no-Kage, 2022-07-07:
Oh, yes, I switch between 32 and 64 bit, I just happen to have 32 bit on this specific image.  I will make sure to test on both in due course.  Again, thanks for active support of your github release... greatly appreciated!

closed 2022-07-07 by Jibun-no-Kage

morrownr, 2022-07-08:
If you run onto the reason for the problem, please share and if you see mods I can make to the scripts that would help, please share. I use the scripts on 5 drivers here. This driver is actually the lowest volume of the 5 drivers. It currently shows about 400 clones per week. The highest volume driver is the one for the 8812au. If you add in the USB-WiFi repo which is the Main Menu for the site then we are talking about 10,000 + views per week. My point being that if you find something that makes things better, it will help a lot of people.

The Main Menu is located at:

 https://github.com/morrownr/USB-WiFi

Jibun-no-Kage, 2022-07-09:
Sure, if I find anything that I think can help, definitely will mention it.

Jibun-no-Kage, 2022-07-09:
Initial test with clean 64 bit image and driver installation was good as expected.  I am now working through the various OS customizations I do, to see if I break something in the process.

TobiaszCudnik, 2022-08-29:
`--kernelsourcedir` isnt accepted by the installation script, which is the main problem here I think.