morrownr/8821cu-20210118 #27: usb-wifi recognized as a CD-ROM

anhngzv, 2022-07-11:
My 8821CU doesn''t work. 
After I had tried `lsusb` it showed a message `Bus 002 Device 003: ID 0bda:1a2b Realtek Semiconductor Corp. RTL8188GU 802.11n WLAN Adapter (Driver CDROM Mode)`, then I unplugged and plugged again and it was worked. When I tried "sudo usb_modeswitch -R -v 0bda -p 1a2b" to change mode, it had an error `USB description data (for identification)
-------------------------
Manufacturer: Realtek
     Product: DISK
  Serial No.: not provided
-------------------------
Sending standard EJECT sequence
Looking for active drivers ...
Set up interface 0
Use endpoint 0x0b for message sending ...
Trying to send message 1 to endpoint 0x0b ...
 Sending the message returned error -7. Try to continue
Read the response to message 1 (CSW) ...
 Response reading failed (error -8)
 Device is gone, skip any further commands
-> Run lsusb to note any changes. Bye!
`Please help me fix it, thank you.

morrownr, 2022-07-11:
Hi @anhngzv 

We need to look at this issue one step at a time. It certainly looks like the adapter is a multi-state adapter.

It may help if I know what distro and hardware you are running.

Please shut the system down, place the adapter in a usb2 port and boot.
Run `$ lsusb` and post the entire output, not just the line you think is the adapter.
Tell me if you are seeing the cd that indicates the adapter is in storage mode.

morrownr, 2022-07-12:
> Thank you for replying to me. 

You are welcome.

> I'm sure that it's my adapter because it has the same name.

Famous last words. Not that I don't believe you but things don't always work like we think they do.

> Many people have the same problem as me. I've searched a lot on google.

Yes, multi-state adapters cause problems. There are good single-state adapters.

> They used sudo usb_modeswitch -R -v 0bda -p 1a2b and it worked. 

Try:

$ sudo usb_modeswitch -K -W -v 0bda -p 1a2b

> But I think because I use Ubuntu 22.04 then it didn't work for me.

We can find the problem but I but to do so, I really need you to provide the answers I ask for. Let me repeat:

Please shut the system down, place the adapter in a usb2 port and boot.
Run $ lsusb and post the entire output, not just the line you think is the adapter.

anhngzv, 2022-07-12:
Thank you, here is my output of `$lsusb`:
```
Bus 002 Device 003: ID 0bda:1a2b Realtek Semiconductor Corp. RTL8188GU 802.11n WLAN Adapter (Driver CDROM Mode)
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 05ac:024f Apple, Inc. Aluminium Keyboard (ANSI)
Bus 001 Device 003: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
and after I unplugged and replugged it:
```
Bus 002 Device 005: ID 0bda:2006 Realtek Semiconductor Corp. 802.11ac NIC
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 05ac:024f Apple, Inc. Aluminium Keyboard (ANSI)
Bus 001 Device 003: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

morrownr, 2022-07-12:
Thanks for the info.

Can you run the following and post the results:

$ usb_modeswitch --version

ID 0bda:1a2b is a pretty standard Realtek CD-ROM mode ID. I actually have an adapter that uses that ID but it is not based on the same chipset as your adapter.

FYI: Does your adapter support Bluetooth? If not, it is the 8811cu chipset. I'm just gathering details.

My main dev box run a distro that is a derivative og Ubuntu 22.04 so internally it should be mostly the same.

I will investigate here some and get back to you.

anhngzv, 2022-07-13:
My usb_modeswitch run on version 2.6.0, the newest version. My adapter doesn't support Bluetooth, it's name is TOTOLINK a650ua.

morrownr, 2022-07-13:
Thanks for that information. I am gathering data and will present it to you soon.

morrownr, 2022-07-13:
FYI: I am not finished investigating and will need you to do some testing, most likely, but it appears their are certain 8811cu based adapters that ship with faulty internal firmware. You may have one.

The problem you are seeing is not caused by this driver and it may not be caused by usb_modeswitch or Ubuntu.

morrownr, 2022-07-15:
Report:

I have one adapter that uses ID 0bda:1a2b. It does not use the 8811cu chipset but it should be good to see if we are getting the switching we are looking for based on the ID. I tested multiple times on a system with Ubuntu 22.04 and another system with Linux Lite 6 which is derived from Ubuntu 22.04. I am trying to get as close as possible to your setup in an effort to duplicate the problem. In all cases, usb_modeswitch worked as it should and switched the adapter to wifi mode very rapidly. I turned usb_modeswitch off per the instructions at the end of the following document and it switched manually as well.

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Modeswitch.md

Another similar issue was posted and fixed yesterday... #29

As I maintain several drivers and various bits of information, I get a view that most don't see. My estimate is that there are more problems with multi-state adapters based on 8811cu/21cu chipsets than all others combined. Part of the reason is that this chipset seems to be a favorite for multi-state adapters and most of the time, if there is a problem, it can be resolved without too much trouble. Then we see the rare case where it is not easy to fix.

Here is a related thread from the usb_modeswitch forum:

https://www.draisberghof.de/usb_modeswitch/bb/viewtopic.php?f=4&t=3010&p=19905&hilit=0bda%3A1a2b#p19905

If it is an internal firmware issue, it possibly could be fixed with a workaround but could take a lot of effort. it appears the issue is somewhere in the timing of the initial boot, before this driver is involved.

What would I do if it were me? I would look to find another use for the adapter or sell/donate it to a windows user and then take a look at the following site (Menu Item 2) for an adapter that is single state and uses an in-kernel adapter:

https://github.com/morrownr/USB-WiFi