morrownr/8821cu-20210118 #29: Cannot create NIC though driver installed successfully

zunoffon, 2022-07-14:
- Linux Distribution: 
Raspberry Pi OS (Legacy) with desktop
Release date: April 4th 2022
System: 32-bit
Kernel version: 5.10
Debian version: 10 (buster)

- USB WiFi:
Chipset RealTek RTL 8811CU
USB 2.0
600Mbps

-  Diagnostic info:
(onboard wlan0 already running)
```bash
pi@debianbuster:~ $ sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev
Linux debianbuster 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
EFI variables are not supported on this system
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 009: ID 0bda:1a2b Realtek Semiconductor Corp.
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
1: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
rtl8821cu, 5.12.0, 5.10.103-v7l+, armv7l: installed
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr e6:5f:01:2c:f3:17
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr e4:5f:01:2c:f3:17
                ssid z
                type managed
                channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
                txpower 31.00 dBm

```

The USB WiFi works on my 20.04.4 LTS Ubuntu, but not for Raspberry Pi.
When comparing kern log. There is a little bit different

- Ubuntu:
```bash
[ 2194.214563] usb 3-6: new high-speed USB device number 11 using xhci_hcd
[ 2194.362942] usb 3-6: New USB device found, idVendor=0bda, idProduct=1a2b, bcdDevice= 2.00
[ 2194.362954] usb 3-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2194.362958] usb 3-6: Product: DISK
[ 2194.362962] usb 3-6: Manufacturer: Realtek
[ 2194.366117] usb-storage 3-6:1.0: USB Mass Storage device detected
[ 2194.366372] scsi host5: usb-storage 3-6:1.0
[ 2195.393215] scsi 5:0:0:0: CD-ROM            Realtek  Driver Storage   1.00 PQ: 0 ANSI: 0 CCS
[ 2195.398522] sr 5:0:0:0: [sr0] scsi3-mmc drive: 0x/0x caddy
[ 2195.446512] sr 5:0:0:0: Attached scsi CD-ROM sr0
[ 2195.446757] sr 5:0:0:0: Attached scsi generic sg2 type 5
[ 2195.503635] usb 3-6: USB disconnect, device number 11
[ 2195.886353] usb 3-6: new high-speed USB device number 12 using xhci_hcd
[ 2196.034871] usb 3-6: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[ 2196.034882] usb 3-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2196.034887] usb 3-6: Product: 802.11ac NIC
[ 2196.034890] usb 3-6: Manufacturer: Realtek
[ 2196.034893] usb 3-6: SerialNumber: 123456
[ 2196.196197] rtl8821cu 3-6:1.0 wlx90de80311b0b: renamed from wlan0
[ 2196.909223] IPv6: ADDRCONF(NETDEV_CHANGE): wlx90de80311b0b: link becomes ready
[ 2202.104692] IPv6: ADDRCONF(NETDEV_CHANGE): wlx90de80311b0b: link becomes ready
```
- Raspberry Pi:
```bash
[ 1051.905120] usb 1-1.3: new high-speed USB device number 8 using xhci_hcd
[ 1052.035827] usb 1-1.3: New USB device found, idVendor=0bda, idProduct=1a2b, bcdDevice= 2.00
[ 1052.035850] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1052.035868] usb 1-1.3: Product: DISK
[ 1052.035885] usb 1-1.3: Manufacturer: Realtek
[ 1052.041524] usb-storage 1-1.3:1.0: USB Mass Storage device detected
[ 1052.045019] scsi host0: usb-storage 1-1.3:1.0
[ 1053.117982] scsi 0:0:0:0: CD-ROM            Realtek  Driver Storage   1.00 PQ: 0 ANSI: 0 CCS
[ 1053.130169] sr 0:0:0:0: [sr0] scsi3-mmc drive: 0x/0x caddy
[ 1053.176469] sr 0:0:0:0: Attached scsi CD-ROM sr0
[ 1053.176926] sr 0:0:0:0: Attached scsi generic sg0 type 5
[ 1053.375668] usb 1-1.3: reset high-speed USB device number 8 using xhci_hcd
```

Is it possible to fix?

zunoffon, 2022-07-15:
hi Nick,

> Option 1: ... What RasPi hardware are you using?

I'm using Raspi 4
> Option 2:

I go with this option and the issue is fixed :clap::clap::clap:
> Option 3:

I'll come back with the wiki when I get mad

Thanks a lot for all the helpful options.

morrownr, 2022-07-15:
Reopening as others may benefit from seeing this fix.