morrownr/8821cu-20210118 #30: Packet Injection not working.

ajkelsey, 2022-07-15:
I have a BrosTrend AC5L using the rtl8821cu. I was able to get into monitor mode, but deauth did not work. I ran a package injection test and it said it worked on the broadcast probe request and failed on direct probe requests. Any insight on why this is happening would be appreciated.

Linux kali 5.18.0-kali5-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.18.5-1kali5 (2022-07-04) x86_64 GNU/Linux
EFI variables are not supported on this system
Bus 002 Device 002: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 0e0f:0008 VMware, Inc. Virtual Bluetooth Adapter
Bus 001 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 001 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
0: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
1: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
rtl8821cu/5.12.0, 5.18.0-kali5-amd64, x86_64: installed
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 0c:cf:89:6f:c4:2a
                type monitor
                channel 40 (5200 MHz), width: 20 MHz (no HT), center1: 5200 MHz
                txpower 327.00 dBm

morrownr, 2022-07-15:
Hi @ajkelsey 

Before I set up to test, can I get you to go to:

https://github.com/morrownr/USB-WiFi

And select Main Menu item 10. Monitor Mode

You can click on `Monitor_Mode.md` to make it available for reading. I wrote it to help users of Realtek adapters work with monitor mode. The Realtek drivers, including this one, are not standards compliant so many guides don't work well with them. 

Also you can download the script and try it. It helps quickly take an interface into monitor mode and it handles interfering processes better, in my opinion, than does the alternative.

Let me know how that goes and if necessary, I can set things up here with my 8811cu based adapter to do some testing.

Regards

morrownr, 2022-07-15:
Hey, take your time and if you see things that need to be improved with the document or script, let me know. You could even clone the repo, edit on your end and submit a PR.

FYI: The adapter that is, in my opinion, the best you can buy right now  for wireless security analysis is the ALFA AWUS036ACHM. You can see it and other adapters that use in-kernel driver at the following location:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

Of note: Adapters that use the new mt7921au (wifi 6) chipset are now starting to ship. The driver is already in the mainline kernel. Monitor mode generally works better with the standards compliant in-kernel driver.

I'm going to mark this as solved for now but you are welcome to keep going as you have questions.

Cheers

ajkelsey, 2022-07-15:
Thanks. I'm new to this and didn't want to invest too heavily before I really had some decent knowledge. I'll check out that ALFA if it isn't outdated by the time I'm ready to upgrade. 😄