morrownr/8821cu-20210118 #31: DHCP Diagnosis?

BusinessKid, 2022-07-16:
First of all, very nice tidy neat work. The driver compiled on the console method without a warning. For me, it's a new driver for a new dongle in a new box, and something isn't right. I usually connect using 'wpa_supplicant <options> && dhcpcd <wlanX>'. I'm not new to linux, or wifi. Wpa_supplicant stays going. The 'dhcpcd' part failed. Here's the (verbose) output: `bash-5.1$ sudo dhcpcd -d wlan3
Password: 
dhcpcd-9.4.1 starting
chrooting as dhcpcd to /var/lib/dhcpcd
sandbox: seccomp
spawned manager process on PID 4980
spawned privileged proxy on PID 4981
spawned network proxy on PID 4982
spawned controller proxy on PID 4983
DUID 00:04:ef:ef:16:89:a4:a4:96:12:af:70:d8:bb:c1:16:e2:1e
wlan3: executing: /lib/dhcpcd/dhcpcd-run-hooks PREINIT
wlan3: executing: /lib/dhcpcd/dhcpcd-run-hooks NOCARRIER
wlan3: waiting for carrier
wlan3: carrier lost - roaming
wlan3: executing: /lib/dhcpcd/dhcpcd-run-hooks NOCARRIER_ROAMING
received SIGTERM, stopping
wlan3: removing interface
wlan3: executing: /lib/dhcpcd/dhcpcd-run-hooks STOPPED 

##### AND ALSO DHCLIENT #####

bash-5.1$ sudo dhclient -v  wlan3
Internet Systems Consortium DHCP Client 4.4.2-P1
Copyright 2004-2021 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan3/1c:bf:ce:e3:a6:38
Sending on   LPF/wlan3/1c:bf:ce:e3:a6:38
Sending on   Socket/fallback
DHCPDISCOVER on wlan3 to 255.255.255.255 port 67 interval 8
DHCPDISCOVER on wlan3 to 255.255.255.255 port 67 interval 11
DHCPDISCOVER on wlan3 to 255.255.255.255 port 67 interval 11
DHCPDISCOVER on wlan3 to 255.255.255.255 port 67 interval 17
Terminated`
Fool that I am, I tried it in windows, but the dongle won't see my router, although it does see others. :-1: ([https://www.linuxquestions.org/questions/showthread.php?p=6367838#post6367838](url) In this LQ post someone with slackware64-15.0 (my exact OS) reported having a couple of wifi dongles using this driver. Iwlist reports local stations including my own.  Any ideas or diagnosis welcome.

morrownr, 2022-07-16:
Hi @BusinessKid 

Glad to see you. I hope you don't mind if I start off with some off-topic stuff.

I've been using Linux since around '94. I have great respect for what I call the original 5 distros. I include Debian, Slackware, RedHat, SuSE and Arch in that list and would like to have good installation instructions here for each. I've basically done that with the exception of Slackware and I could use help if you have time to help add Slackware installation instructions.

Continuing off-topic: If I see a user that has just bought a new Realtek based adapter and they are having problems within the time allowed to return the adapter, I feel I would be doing a disservice not to bring the Main Menu to your attention:

https://github.com/morrownr/USB-WiFi

If you read menu items 1 and 2, you will see that Linux users have options that work with in-kernel drivers... which does away with a lot of problems. It might be different if the Realtek out-of-kernel drivers were standards compliant but they are not, they are a dead end.

To the topic: Can you point me to the location to download and install Slackware. I think I am going to have to see this for myself. 

Regards

BusinessKid, 2022-07-18:
I have this going in my RazPi 4B using the ARM64_RPI.sh script. It's up and working and I've decided to leave it there. So we can mark this solved. I have exactly the same issue on the RTL 8812 device. But I'll post it there, as it doesn't belong here.

closed 2022-07-20 by BusinessKid