morrownr/8821cu-20210118 #35: Swapping sd cards between Raspi 3 and 4

ashah976, 2022-08-11:
After successful install on Raspi 4 when sd card is moved to a Raspi 3 the usb adapter is unrecognised,
I have to then reinstall on the Raspi 3 to get it working  
The same happens when moving from Raspi 3 back to Raspi 4

USB adapter is RTL8811CUS

5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

morrownr, 2022-08-22:
@ashah976 

I seem to have missed this post. Have you figured this out?

ABelliqueux, 2022-09-26:
I'm encountering the same issue but with a RPI 1 B+ and a RPI 3 B with Raspi OS Buster and a rtl8821CU ( 0bda:c811).
`Linux 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux`

I installed the driver on the Rpi 1 B+, all is working great.
After switching to Rpi 3, drivers is not loading because it's not looking in the right folder, as it's using a different kernel according to the Rpi used ( **5.10.103+** for Rpi 1 and **5.10.103-v7+** for Rpi3 ).

`find /lib/modules -type f -name "*8821*"` gives :
`/lib/modules/5.10.103+/kernel/drivers/net/wireless/8821cu.ko`

and `sudo modprobe 8821cu` : 
`modprobe: FATAL: Module 8821cu not found in directory /lib/modules/5.10.103-v7+`

I also have a `5.10.103-v8+` folder in `/lib/modules/` which I suspect is the kernel for the RPI 4 ?

# Solution
So the solution was to rebuild the module from the rpi 3  kernel  with : 

```
sudo dkms build -m rtl8821cu -v 5.12.0
sudo dkms install -m rtl8821cu -v 5.12.0
# The module should loaded automatically but just if needed...
sudo modprobe 8821cu
```

I have no idea if this step could be done from the first install ? I.E build module for all kernels during first install ? or installing a hook to trigger building/install when a new processor arch is detected ?

morrownr, 2022-09-26:
> So maybe the install script should check for the module in the current arch instead of only relying on the /usr/src/rtl8821cu-5.12.0 folder to be present ?

Let me take a look at this. Are you willing to test?

Nick

morrownr, 2022-09-26:
@ABelliqueux 

I'm working on this now. Once I think I am finished, I'll need to test here as I need to get this right. Each repo has 2 installation scripts and I have 5 repos. These repos get over 1000 hits per day so you can imagine what would happen if there is a mistake.

Hopefully this will fix the problem. I had no considered the issue you brought up.

Nick

morrownr, 2022-09-26:
> /kernel/drivers/net/wireless/8821cu.ko

I ran into the reason I did it the way i did it a long time ago. dkms decides where to put the module and it varies depending on the distro. Here is a link to the current dkms man page at Ubuntu:

https://manpages.ubuntu.com/manpages/jammy/en/man8/dkms.8.html

I am sure there is a good way to detect previous installation so that we can abort and get users to run the removal script. I am going to have to ponder this. Do you see a good way to do it?

morrownr, 2022-09-27:
I've taken a fresh look. The `dkms add` command does support recognizing dkms already installed conditions and it kicks out an error code 3 for that. I also need to account for manual/depmod installs and I can do that with separate logic.

While looking at things, I decided to consolidate the removal scripts into one to keep things simple.

Let me do a little more work and testing and I get something up for you to test. I have RasPis, I just never thought to use the sd from one in another.

morrownr, 2022-09-29:
@ashah976  @ABelliqueux 

I just merged a large upgrade to the installation and removal scripts. I think I came up with a solution for this specific problem and I eliminted the no-dkms version of the scripts which makes things more simple for everyone.

I would appreciate both you testing.

Nick

ABelliqueux, 2022-09-29:
So I tested the new script and was able to run `install-driver.sh` successfully on a rpi3 without it asking to run `remove-driver.sh` first. The rpi 1's kernel module was left untouched. 

Is the new script supposed to build the driver for all kernel or only the current one ?

morrownr, 2022-09-29:
> So I tested the new script and was able to run install-driver.sh successfully on a rpi3 without it asking to run remove-driver.sh first. The rpi 1's kernel module was left untouched.

Okay. So it appears we may be headed the right direction. I really need you to run the remove-driver.sh script for testing as well.

> Is the new script supposed to build the driver for all kernels or only the current one ?

The installation script builds for the current kernel. The removal script removes all. This has worked well but I can see the logic of the removal script only removing for the current kernel. Take a look and tell me what you think?

Keep in mind that I have to be careful not to break things as a lot of folks use this repo. I will be testing these new scripts here and if I don't see any problems, I will propagate the changes to the other 4 Realtek repos.

ABelliqueux, 2022-09-29:
Ok,I just tried the install/remove scripts and they seem to behave as planned ; build module for current kernel, remove module for every kernel.

Maybe @ashah976 or someone else could confirm this all works fine ?

What I do after installing on current kernel is running : 
```
ls /lib/modules | \
    sudo xargs -n1 /usr/lib/dkms/dkms_autoinstaller start
```
to build for other kernels.

morrownr, 2022-09-30:
Okay. So far, so good. I can see a couple more things that should be done but are not critical. This can get messy.

I think I have a way to handle secure mode in the scripts but that is going to take some testing since dkms already trys to work secure... though it does not work on all distros.

Thanks for the help.

morrownr, 2022-10-04:
I'm going to mark this solved for now. Let me know if anything comes up.

ashah7, 2022-11-23:
apologies everyone I haven't checked this in a while but the solution for me was to build from source and not use the installation script