morrownr/8821cu-20210118 #36: Fails to build on Arch Linux kernel 5.19.2

MiKl-LS, 2022-08-18:
Building the module on 5.19.2 errors with `/var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/ioctl_cfg80211.c:10669:17: error: ‘struct wireless_dev’ has no member named ‘current_bss’`

Still works on 5.19.1 so I'll just downgrade for the moment
```
DKMS make.log for rtl8821cu-5.12.0 for kernel 5.19.2-arch1-1 (x86_64)
Thursday, 18 August, 2022 07:38:35 PM PST
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.19.2-arch1-1/build M=/var/lib/dkms/rtl8821cu/5.12.0/build  modules
make[1]: Entering directory '/usr/lib/modules/5.19.2-arch1-1/build'
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/monitor/rtw_radiotap.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/wds/rtw_wds.o
/var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_sta_mgt.c: In function ‘rtw_mfree_stainfo’:
/var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_sta_mgt.c:391:25: warning: the comparison will always evaluate as ‘true’ for the address of ‘lock’ will never be NULL [-Waddress]
  391 |         if (&psta->lock != NULL)
      |                         ^~
In file included from /var/lib/dkms/rtl8821cu/5.12.0/build/include/drv_types.h:112,
                 from /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_sta_mgt.c:17:
/var/lib/dkms/rtl8821cu/5.12.0/build/include/sta_info.h:267:17: note: ‘lock’ declared here
  267 |         _lock   lock;
      |                 ^~~~
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_ft.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_wnm.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_mbo.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_rm_util.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_roch.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/crypto/aes-internal.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/crypto/aes-internal-enc.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/crypto/aes-gcm.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/crypto/aes-ccm.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/crypto/aes-omac1.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/crypto/ccmp.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/crypto/gcmp.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/crypto/aes-siv.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/crypto/aes-ctr.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/crypto/sha256-internal.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/crypto/sha256.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/crypto/sha256-prf.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/crypto/rtw_crypto_wrap.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_swcrypto.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/wifi_regd.o
/var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/ioctl_cfg80211.c:478:9: error: too few arguments to function ‘cfg80211_ch_switch_notify’
  478 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/rtl8821cu/5.12.0/build/include/osdep_service_linux.h:98,
                 from /var/lib/dkms/rtl8821cu/5.12.0/build/include/osdep_service.h:54,
                 from /var/lib/dkms/rtl8821cu/5.12.0/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:7967:6: note: declared here
 7967 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/ioctl_cfg80211.c: At top level:
/var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/ioctl_cfg80211.c:10389:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, unsigned int)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *)’ [-Werror=incompatible-pointer-types]
10389 |         .stop_ap = cfg80211_rtw_stop_ap,
      |                    ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/ioctl_cfg80211.c:10389:20: note: (near initialization for ‘rtw_cfg80211_ops.stop_ap’)
/var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/ioctl_cfg80211.c:10433:24: error: initialization of ‘int (*)(struct wiphy *, struct wireless_dev *, unsigned int,  struct cfg80211_chan_def *)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct wireless_dev *, struct cfg80211_chan_def *)’ [-Werror=incompatible-pointer-types]
10433 |         .get_channel = cfg80211_rtw_get_channel,
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/ioctl_cfg80211.c:10433:24: note: (near initialization for ‘rtw_cfg80211_ops.get_channel’)
/var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_wdev_unregister’:
/var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/ioctl_cfg80211.c:10669:17: error: ‘struct wireless_dev’ has no member named ‘current_bss’
10669 |         if (wdev->current_bss) {
      |                 ^~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:249: /var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1849: /var/lib/dkms/rtl8821cu/5.12.0/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.19.2-arch1-1/build'
make: *** [Makefile:2498: modules] Error 2
```

morrownr, 2022-08-18:
I am aware of this and am working on a patch.

morrownr, 2022-08-18:
Patch applied. Can I get you to test and report?

fishears, 2022-08-18:
I can confirm that the patch has fixed the issue for Arch Linux 5.19-2
Thank you

morrownr, 2022-08-18:
I appreciate the report.

MiKl-LS, 2022-08-19:
The patch worked for my Arch Linux 5.19.2 as well
Thanks! Closing this issue

closed 2022-08-19 by MiKl-LS

CueXXIII, 2022-08-22:
I can confirm it also works on vanilla linux 6.0-rc1 and 6.0-rc2 (only tested managed mode), too.