morrownr/8821cu-20210118 #37: Low speed

ilgiznurgaliev, 2022-08-22:
Hello.

I have rtl8821cu based usb wi-fi adapter (without bluetooth). This driver works fine but download/upload speed is low, up to 60Mbit/s.
Tested using iperf3 with home server that connected to the same router directly to ethernet port.

The adapter plugged in to rear usb port, pc is 2 meters away from wi-fi router. Other wi-fi clients shows up to 200-250Mbit/s, so router performance seems good.

Probably some driver options needs to be tuned to get better performance. Already tried to turn off power saving option, but it didn't help.

Could anyone please suggest what else can be checked/tuned?

```
$ uname -a
Linux casper-pc 5.19.2-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 17 Aug 2022 13:48:51 +0000 x86_64 GNU/Linux
```
```
$ lsusb
Bus 002 Device 003: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
```
```
$ grep -v ^# /etc/modprobe.d/8821cu.conf
options 8821cu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_dfs_region_domain=0 rtw_RFE_type=7
```
```
$ iwconfig
wlp0s29u1u2  IEEE 802.11an  ESSID:"CasperAP_5G"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:5.24 GHz  Access Point: ***   
          Bit Rate:72.2 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=78/100  Signal level=58/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```
```
$ iw dev
phy#0
        Interface wlp0s29u1u2
                ifindex 3
                wdev 0x1
                addr ***
                ssid CasperAP_5G
                type managed
                channel 48 (5240 MHz), width: 20 MHz, center1: 5240 MHz
                txpower 17.00 dBm
```
```
$ sudo iw phy
Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 9
        max scan IEs length: 2304 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CMAC (00-0f-ac:6)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CCMP-256 (00-0f-ac:10)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
                * CMAC-256 (00-0f-ac:13)
        Available Antennas: TX 0 RX 0
[... 162 lines cut ...]
                 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
        Supported extended features:
                * [ RRM ]: RRM
```
```
$ iperf3 -s
-----------------------------------------------------------
Server listening on 5201 (test #1)
-----------------------------------------------------------
Accepted connection from 10.16.0.4, port 56488
[  5] local 10.16.0.10 port 5201 connected to 10.16.0.4 port 56500
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  3.65 MBytes  30.6 Mbits/sec                  
[  5]   1.00-2.00   sec  3.58 MBytes  30.1 Mbits/sec                  
[  5]   2.00-3.00   sec  3.94 MBytes  33.0 Mbits/sec                  
[  5]   3.00-4.00   sec  6.16 MBytes  51.7 Mbits/sec                  
[  5]   4.00-5.00   sec  6.59 MBytes  55.3 Mbits/sec                  
[  5]   4.00-5.00   sec  6.59 MBytes  55.3 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-5.00   sec  25.5 MBytes  42.8 Mbits/sec                  receiver
iperf3: the client has terminated
```

morrownr, 2022-08-22:
Good morning @ilgiznurgaliev 

Can I get you to install `wavemon` so as to take a better look at some of the connection information:

$ sudo apt install wavemon
$ wavemon

I'm interested in the `link quality`, `signal level` and `channel width` that you see.

The information about seemed to indicate you have a channel width of 20 and a Bit Rate:72.2 Mb/s. Those two numbers go together and getting around 30 Mb/s with those numbers is about right but the question is why are we seeing this?

An easy way to edit 8821cu.conf is:

sudo ./edit-options.sh

Here is what you posted:

options 8821cu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_dfs_region_domain=0 rtw_RFE_type=7

Here are some changes to try:

rtw_power_mgnt=1
rtw_dfs_region_domain=0 (read the docs and set this appropriately)
rtw_RFE_type=7 (delete this parameter as it may not be necessary in your case)

Are you using an extension cable or plugged directly into a port on the back of the system?

Regards

ilgiznurgaliev, 2022-08-22:
Thanks for response @morrownr.

Ok, I have just changed parameters as you suggested and reloaded 8821cu module. Speed is the same, about 40Mbit/s.

`wavemon` output:
```
┌─Interface──────────────────────────────────────────────────────────────────────┐
│wlp0s29u1u2 - wdev 1, phy 1, reg: n/a, SSID: CasperAP_5G                        │
├─Levels─────────────────────────────────────────────────────────────────────────┤
│                                                                                │
│link quality: 100%  (70/70)                                                     │
│================================================================================│
│                                                                                │
│                                                                                │
│signal level: -31 dBm (0,79 uW)                                                 │
│=============================================================                   │
│                                                                                │
├─Packet Counts──────────────────────────────────────────────────────────────────┤
│RX: 88k (0 B)                                                                   │
│TX: 102k (0 B)                                                                  │
├─Info───────────────────────────────────────────────────────────────────────────┤
│mode: Managed, connected to: ***, time: 0 sec, inactive: 57ms     │
│freq: 5240 MHz, channel: 48 (width: 20 MHz), bands: 2                           │
│station flags: (none), preamble: short, slot: long                              │
│rx rate: n/a                                                                    │
│tx rate: 72.2 MBit/s                                                            │
│tx power: 17 dBm (50,12 mW), power save: on                                     │
│retry short/long: 7/4, rts/cts: off, frag: off                                  │
├─Network────────────────────────────────────────────────────────────────────────┤
│wlp0s29u1u2 #4 (UP RUNNING BROADCAST MULTICAST)                                 │
│mode: dormant, qdisc: mq, txq: 4, qlen: 1000                                    │
│mac: ***                                                         │
│ip4: 10.16.0.10/24, valid: 23:49h                                               │
│ip6: n/a                                                                        │
└────────────────────────────────────────────────────────────────────────────────┘
```

I don't use any extension cable, it plugged dierctly into a port on the back of the system. Other usb ports give the same result.

morrownr, 2022-08-22:
> wlp0s29u1u2

Hmmm...

What is the output of:

$ iw dev

morrownr, 2022-08-22:
Well, the mystery continues:

Can you post the results of the following:

$ lspci
$ lsusb

What Linux distro are you using?

In case you are interested in where I am headed with this:

That interface you posted does not look like what I would expect from this driver. Does your system have an internal wifi card?

ilgiznurgaliev, 2022-08-22:
Below is as requested:

```
$ lspci
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v2/Ivy Bridge DRAM Controller (rev 09)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v2/3rd Gen Core processor PCI Express Root Port (rev 09)
00:16.0 Communication controller: Intel Corporation 6 Series/C200 Series Chipset Family MEI Controller #1 (rev 04)
00:1a.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #2 (rev 05)
00:1b.0 Audio device: Intel Corporation 6 Series/C200 Series Chipset Family High Definition Audio Controller (rev 05)
00:1c.0 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 1 (rev b5)
00:1c.4 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 5 (rev b5)
00:1c.5 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 6 (rev b5)
00:1d.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #1 (rev 05)
00:1f.0 ISA bridge: Intel Corporation H61 Express Chipset LPC Controller (rev 05)
00:1f.2 IDE interface: Intel Corporation 6 Series/C200 Series Chipset Family Desktop SATA Controller (IDE mode, ports 0-3) (rev 05)
00:1f.3 SMBus: Intel Corporation 6 Series/C200 Series Chipset Family SMBus Controller (rev 05)
00:1f.5 IDE interface: Intel Corporation 6 Series/C200 Series Chipset Family Desktop SATA Controller (IDE mode, ports 4-5) (rev 05)
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Tonga PRO [Radeon R9 285/380] (rev f1)
01:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Tonga HDMI Audio [Radeon R9 285/380]
02:00.0 SATA controller: Marvell Technology Group Ltd. 88SE9125 PCIe SATA 6.0 Gb/s controller (rev 11)
03:00.0 USB controller: ASMedia Technology Inc. ASM1042A USB 3.0 Host Controller
04:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 06)
```
```
$ lsusb
Bus 002 Device 005: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 002 Device 004: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 002 Device 007: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 05e3:0626 Genesys Logic, Inc. Hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Distro is ArchLinux (with KDE) with latest updates installed.
My PC doesn't have any other wifi cards.

morrownr, 2022-08-22:
> Bus 002 Device 007: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC

There is the adapter. It is based on the 8811cu chipset. This is the correct driver.

Have you installed any other drivers for this adapter?

$ dkms status

ilgiznurgaliev, 2022-08-22:
only this one:
```
$ dkms status
rtl8821cu/5.12.0, 5.19.2-arch1-1, x86_64: installed
```

Previously I used another driver from [this](https://github.com/brektrou/rtl8821CU) repo, but after latest OS update I'm unable to compile it anymore. That's why I swithed to your driver today, but faced the issue with low speed. With that driver from **brektrou** repo average speed was 110-120Mbit/s.

morrownr, 2022-08-22:
Two more questions:

Post the results of:

$ lsmod

When did you download this driver? Was it prior to yesterday?

ilgiznurgaliev, 2022-08-22:
```
$ lsmod
Module                  Size  Used by
8821cu               3756032  0
uinput                 20480  1
snd_seq_dummy          16384  0
snd_hrtimer            16384  1
snd_seq                94208  7 snd_seq_dummy
snd_seq_device         16384  1 snd_seq
rfcomm                 90112  16
qrtr                   57344  4
intel_rapl_msr         20480  0
intel_rapl_common      32768  1 intel_rapl_msr
cmac                   16384  3
algif_hash             16384  1
algif_skcipher         16384  1
x86_pkg_temp_thermal    20480  0
af_alg                 36864  6 algif_hash,algif_skcipher
intel_powerclamp       20480  0
nct6775                28672  0
bnep                   32768  2
nct6775_core           77824  1 nct6775
snd_hda_codec_via      24576  1
hwmon_vid              16384  1 nct6775
snd_hda_codec_generic    98304  1 snd_hda_codec_via
kvm_intel             385024  0
coretemp               20480  0
ledtrig_audio          16384  1 snd_hda_codec_generic
snd_hda_codec_hdmi     86016  1
btusb                  65536  0
btrtl                  28672  1 btusb
kvm                  1130496  1 kvm_intel
snd_hda_intel          61440  2
btbcm                  24576  1 btusb
irqbypass              16384  1 kvm
snd_intel_dspcfg       36864  1 snd_hda_intel
crct10dif_pclmul       16384  1
snd_intel_sdw_acpi     20480  1 snd_intel_dspcfg
btintel                45056  1 btusb
crc32_pclmul           16384  0
snd_hda_codec         188416  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_via
eeepc_wmi              16384  0
nls_iso8859_1          16384  1
ghash_clmulni_intel    16384  0
btmtk                  16384  1 btusb
asus_wmi               69632  1 eeepc_wmi
snd_hda_core          118784  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_via
aesni_intel           380928  4
vfat                   24576  1
iTCO_wdt               16384  0
crypto_simd            16384  1 aesni_intel
fat                    98304  1 vfat
mei_pxp                20480  0
intel_pmc_bxt          16384  1 iTCO_wdt
mei_hdcp               24576  0
snd_hwdep              16384  1 snd_hda_codec
sparse_keymap          16384  1 asus_wmi
at24                   28672  0
iTCO_vendor_support    16384  1 iTCO_wdt
wmi_bmof               16384  0
platform_profile       16384  1 asus_wmi
r8169                 102400  0
cfg80211             1089536  1 8821cu
bluetooth             876544  54 btrtl,btmtk,btintel,btbcm,bnep,btusb,rfcomm
cryptd                 28672  3 crypto_simd,ghash_clmulni_intel
snd_pcm               172032  4 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_core
rapl                   16384  0
realtek                36864  1
intel_cstate           20480  0
snd_timer              49152  3 snd_seq,snd_hrtimer,snd_pcm
mdio_devres            16384  1 r8169
ecdh_generic           16384  2 bluetooth
joydev                 28672  0
mousedev               24576  0
i2c_i801               45056  0
rfkill                 32768  9 asus_wmi,bluetooth,cfg80211
intel_uncore          217088  0
pcspkr                 16384  0
libphy                172032  3 r8169,mdio_devres,realtek
lpc_ich                28672  0
i2c_smbus              20480  1 i2c_i801
snd                   126976  16 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_timer,snd_hda_codec_via,snd_pcm
mei_me                 53248  2
mac_hid                16384  0
soundcore              16384  1 snd
video                  61440  1 asus_wmi
mei                   172032  5 mei_hdcp,mei_pxp,mei_me
wmi                    45056  3 nct6775,asus_wmi,wmi_bmof
vboxnetflt             32768  0
vboxnetadp             28672  0
vboxdrv               544768  2 vboxnetadp,vboxnetflt
crypto_user            24576  0
fuse                  176128  3
bpf_preload            24576  0
ip_tables              36864  0
x_tables               57344  1 ip_tables
ext4                 1015808  2
crc32c_generic         16384  0
crc16                  16384  2 bluetooth,ext4
mbcache                16384  1 ext4
jbd2                  188416  1 ext4
ata_generic            16384  0
pata_acpi              16384  0
dm_mod                184320  6
crc32c_intel           24576  4
xhci_pci               20480  0
ata_piix               45056  0
xhci_pci_renesas       24576  1 xhci_pci
radeon               1867776  0
hid_logitech_hidpp     69632  0
hid_logitech_dj        36864  0
usbhid                 73728  1 hid_logitech_dj
amdgpu               9379840  45
gpu_sched              49152  1 amdgpu
drm_ttm_helper         16384  2 amdgpu,radeon
ttm                    94208  3 amdgpu,radeon,drm_ttm_helper
drm_display_helper    180224  2 amdgpu,radeon
cec                    81920  1 drm_display_helper
```

I downloaded it tomorrow morning, at about 10 AM GMT.

morrownr, 2022-08-22:
I was looking to if the kernel you are using has the patched in in-kernel driver that has not been mainlined. That would cause a conflict. I added code to the installers yesterday afternoon to blacklist the in-kernel driver if this one is installed. However, I am not seeing evidence of the in-kernel driver.

Have you tried a different USB port?

Have you tried removing the adapter and replacing it while booted?

I'm going to get my 8811cu adapter and do some testing. This is a heavily used driver so a lot of folks are using it without problem.

This chipset has shown some interesting results at times. It is a favorite of adapters maker to use multi-state technology, which is not a good thing, and some adapters (this is rare) with this chipset seem to have firmware errata. I'm not sure what dirrection to go with this but will test on my main dev box to see what I see. I have kernel 5.19.2.

ilgiznurgaliev, 2022-08-22:
I have already tried all usb ports and I think all posible options.
Anyway thank you very much for your time.

Next I'm gonna to try this adapter in Ubuntu 22.04 live cd with your driver.

morrownr, 2022-08-22:
Test complete:

Fresh download and compile on Linux Mint (based on debian and ubuntu).

Adapter: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC

Kernel: 5.19.2-051902-generic

iperf3 results:

```
[  5]   0.00-1.00   sec  30.2 MBytes   254 Mbits/sec    0    966 KBytes       
[  5]   1.00-2.00   sec  30.0 MBytes   252 Mbits/sec    0   1.20 MBytes       
[  5]   2.00-3.00   sec  31.2 MBytes   262 Mbits/sec    0   1.42 MBytes       
[  5]   3.00-4.00   sec  32.5 MBytes   273 Mbits/sec    0   1.42 MBytes       
[  5]   4.00-5.00   sec  30.0 MBytes   252 Mbits/sec    0   1.42 MBytes       
[  5]   5.00-6.00   sec  32.5 MBytes   273 Mbits/sec    0   1.42 MBytes       
[  5]   6.00-7.00   sec  32.5 MBytes   273 Mbits/sec    0   1.42 MBytes       
[  5]   7.00-8.00   sec  31.2 MBytes   262 Mbits/sec    0   1.42 MBytes       
[  5]   8.00-9.00   sec  32.5 MBytes   273 Mbits/sec    0   1.58 MBytes       
[  5]   9.00-10.00  sec  32.5 MBytes   273 Mbits/sec    0   1.58 MBytes  
```

morrownr, 2022-08-22:
> I have already tried all usb ports and I think all posible options. Anyway thank you very much for your time.

You are welcome. Sorry we can get it going like you want. Stop by the Main Menu of the site when you have time:

https://github.com/morrownr/USB-WiFi

> Next I'm gonna to try this adapter in Ubuntu 22.04 live cd with your driver.

I'm not sure that will work given everything that has to be installed but I could be wrong.

If you figure it out, let me know.

Regards

ilgiznurgaliev, 2022-08-22:
The same issue on Ubuntu 22.04 with kernel 5.15 and the same on Windows 10. So I assume something wrong with my adapter.

Thanks again for your help. The issue can be closed I think.

closed 2022-08-22 by ilgiznurgaliev

morrownr, 2022-08-22:
>  So I assume something wrong with my adapter.

I have seen adapters fail before. I ran out of ideas in this case.

I guess it is a good thing I sent the link earlier. On Menu Menu item 2, you will find a long list of adapters that use Linux in-kernel drivers. Maybe you can find something you like. There are even WiFi 6e adapters available and the driver is in the 5.19.2 kernel you are using. You may have to update the firmware depending on what you do.

Nick