morrownr/8821cu-20210118 #38: Documentation for recommended settings

peterhoeg, 2022-08-23:
First of all, thank you for all your hard work!

Do you have any documentation/reasoning for the recommendations you make: https://github.com/morrownr/8821cu-20210118#recommended-wifi-router-access-point-settings

morrownr, 2022-08-23:
Hi @peterhoeg 

I can't say that I have documentation but there are probably thousands of links if one of the search engines is your friend. Things do change over time and with the situation at hand so you have to be aware that over time documents suffer bit rot. This document was written by me over time and I try to keep it current as time passes. Right now I need to look at addressing WPA3 and 6 GHz. It is very common that less than optimal router settings are the cause of the problem and not the adapter or driver.

My source: Me. I've been using WiFi since 80211b (WiFi 1) was the hot ticket. I've been using Linux since 1994. I learned to program on an IBM mainframe using FORTRAN. Now you can guess my age.  He he.

I can elaborate on anything you want. Just ask.

Nick

peterhoeg, 2022-08-23:
Hi Nick,

some of the stuff you mention in there goes against Apple’s recommendations as an example: <https://support.apple.com/en-us/HT202068>

Additional context would make it very helpful (don’t get me wrong - your resources are already *very* helpful) to determine when to deviate from the recommendation either because it doesn’t make sense in a particular scenario or because other considerations simply overrule them.

And don’t worry - I’m not going to hold your FORTRAN experience against you. I’m sure you have many redeeming qualities! ;-)

morrownr, 2022-08-23:
> some of the stuff you mention in there goes against Apple’s recommendations... 

From the document:

> Security
> Set to WPA3 Personal for better security
> Set to WPA2/WPA3 Transitional for compatibility with older devices

Apple's recommendations are for Apple products connecting to a router. My advice is if you are using this driver. You are welcome to try Apple's advice about Security. If it works great, if it does not, see my advice. For Apple's advice about security to work well on Linux, you really need to use adapters based on Mediatek chipsets. The simple way to put it is these Realtek out-of-kernel drivers are not that good is certain aspects and security is one of those.

The Main Menu for this site has information and links to adapters based on Mediatek chipsets. See menu item 2:

https://github.com/morrownr/USB-WiFi

> Additional context would make it very helpful

Yes, it would. The main context to keep in mind is that the advice here is for the driver that is here. If you happened to be using one of new Mediatek mt7921au WiFi 6e based adapters, my advice would match Apple's advice...but then if you had one those adapters you would not be here. Right?

In reading the rest of the Apple document, some advice is the same and some is different because it needs to be different. Example: SSID: ...and don't give your 2.4GHz and 5GHz bands different names.

I will argue this point all day long. I can't remember how many users I have helped fix problems by simply recommending that they give the bands different names.

I can't see any advice I want to change after reading Apple's document. Could additional edits help? Sure. If you want to use other documents to help make this document better, please edit and submit a PR. I would welcome it.

I could have gone into much more detail on some topics and I will if you have specific questions.

Nick

peterhoeg, 2022-08-29:
> Apple's recommendations are for Apple products connecting to a router. My advice is if you are using this driver. 

I get that. And it's not that I am asking you counter each of Apple's recommendations.

The point is that there are many other reputable sources of wifi recommendations (one of which is Apple). And if one of your recommendation goes along the line of: "I suggest you set the Foo toggle to Bar because it solves problems AA, BB and CC in the following scenarios. Not doing so (and here are some reasons why you might not want to), could possible give you problems with ZZ and YY."

Then it becomes much easier to make an informed decision.

Otherwise we easily end up with cargo-culting.

Again, I understand if you don't have the time to take this on - but I think it would be very helpful.