morrownr/8821cu-20210118 #40: Driver not installed?

andreasloe, 2022-09-02:
First, a big thank you that you are putting so much effort into this for us (free of charge). Unfortunately, my driver seems not to be installed:
`EFI variables are not supported on this system
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 1241:1122 Belkin Typhoon Stream Optical Mouse USB+PS/2
Bus 001 Device 004: ID 1a0e:2003  
Bus 001 Device 003: ID 03eb:3301 Atmel Corp. at43301 4-Port Hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 2001:3314 D-Link Corp. 802.11n WLAN Adapter
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
rtl8812au, 4.3.14: added
rtl8812au, 4.3.8.12175.20140902+dfsg, 5.15.0-46-generic, x86_64: installed
rtl8821cu, 5.12.0, 5.15.0-46-generic, x86_64: installed
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
`
iwconfig says "no wireless extensions"

morrownr, 2022-09-02:
Hi @andreasloe 

> ID 2001:3314 D-Link Corp. 802.11n WLAN Adapter

This tells me your chipset is the rtl8811au which uses the rtl8821au driver. We need to clean house and then I will put you on the correct driver.

The following are showing as installed:

```
rtl8812au, 4.3.14: added
rtl8812au, 4.3.8.12175.20140902+dfsg, 5.15.0-46-generic, x86_64: installed
rtl8821cu, 5.12.0, 5.15.0-46-generic, x86_64: installed
```

The last one is easy to remove as it is the driver from here. Using a terminal, go to the driver directory and run:

$ sudo ./remove-driver.sh

...and delete the directory the driver is in. This completes removal of the last driver on the list.

Now, the first 2 drivers on the list much be removed. Run both of these commands:

$ sudo dkms remove -m rtl8812au -v 4.3.14 --all
$ sudo dkms remove -m rtl8812au -v 4.3.8.12175.20140902+dfsg --all

Now run `dkms status` and you should get nothing. If so, hopefully things are cleaned up enough to proceed.

The driver you need is:

https://github.com/morrownr/8821au-20210708

Let me know how it goes.

Lastly, the Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

I generally recommend Linux USB WiFi users read the first 2 menu items and then there is menu item 5 that goes over the Realtek drivers maintained here.

Cheers,

Nick

andreasloe, 2022-09-02:
Wonderful, thank you. Everything worked out as outlined! Case can be closed.