morrownr/8821cu-20210118 #41: Fails to build on Ubuntu 22.04.1 LTS (GNU/Linux 5.19.5-02955-g068aa9070f39 aarch64)

TodWulff, 2022-09-03:
Target is an Amlogic AML-S905X-CC based SBC ( [Libre Computer's Le Potato](https://libre.computer/products/s905x/) ).
Image thereon is Ubuntu Server 22.04.1 LTS with Kernel 5.19.5

Install script console session:
`Running install-driver.sh version 20220821
5.19.5-02955-g068aa9070f39
aarch64
Starting installation...
Copying source files to: /usr/src/rtl8821cu-5.12.0
Copying 8821cu.conf to: /etc/modprobe.d
Copying rtw88_8821cu.conf to: /etc/modprobe.d
Creating symlink /var/lib/dkms/rtl8821cu/5.12.0/source -> /usr/src/rtl8821cu-5.12.0

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.19.5-02955-g068aa9070f39 KSRC=/lib/modules/5.19.5-02955-g068aa9070f39/build...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8821cu: 5.12.0 not found
Error! Bad return status for module build on kernel: 5.19.5-02955-g068aa9070f39 (aarch64)
Consult /var/lib/dkms/rtl8821cu/5.12.0/build/make.log for more information.
An error occurred. dkms build error = 10
Please report this error.
Please copy all screen output and paste it into the report.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh
pi@aml-s905x-cc:~/8821cu-20210118$`

make log:
'
DKMS make.log for rtl8821cu-5.12.0 for kernel 5.19.5-02955-g068aa9070f39 (aarch64)
Sat Sep  3 05:58:34 EDT 2022
make ARCH=aarch64 CROSS_COMPILE= -C /lib/modules/5.19.5-02955-g068aa9070f39/build M=/var/lib/dkms/rtl8821cu/5.12.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.19.5-02955-g068aa9070f39'
Makefile:701: arch/aarch64/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/aarch64/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.19.5-02955-g068aa9070f39'
make: *** [Makefile:2498: modules] Error 2'

morrownr, 2022-09-03:
Hi @TodWulff 

Can you tell me how you handled installation Step 9?

TodWulff, 2022-09-04:
I didn't.  (lol?)
I am not *nix savvy.  I was walking the minefield of on-line tutorials and ended up running this repo's code, as a step in getting the wifi/bt dongle working.
I will endeavor to do so now, following the readme's guidance.  Sorry for being a less-than-informed user.  I'll revert herein if I have an issue, or close same, if it ends up working.

TodWulff, 2022-09-04:
https://i.imgur.com/tAfraFT.png

Issue resolved.  Step 9 was the missing piece that enabled this tool to successfully build.

Thank you.

closed 2022-09-04 by TodWulff

morrownr, 2022-09-05:
You are welcome.

> Sorry for being a less-than-informed user. 

Let me see if I can help you become more informed. The Main Menu for this site is located...

https://github.com/morrownr/USB-WiFi