morrownr/8821cu-20210118 #43: WIFI Connecting  Issue

Nabghpok, 2022-09-14:
hello sir, I successfully installed a driver without any error but I am facing a **wifi connecting issue**. When I try to connect to available wifi networks it shows connecting and asks wifi password again and again even though my wifi password is 12345678 which is the correct password.
Sir, can you help me out?
This problem I am facing in Kali Linux.
****

morrownr, 2022-09-14:
Hi @Nabghpok 

There are two things that come to mind:

- If your wifi ap/router is dual band, are both bands using the same essid (name)? If so, recommend you change the name of the 5 GHz network to something new. Maybe adding a "-5" to the existing name. Then reboot the router.

- Your country code is not set:

> country 00: DFS-UNSET

The US below needs to reflect what your country code is. If not sure, tell me what country you are in.
$ sudo iw reg set US 
$ sudo reboot
$ sudo iw reg get

And post the results.

Let's see what happens.

Nick

Nabghpok, 2022-09-15:
5.18.0-kali7-amd64 x86_64
EFI variables are not supported on this system
Bus 001 Device 003: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
rtl8821cu/5.12.0, 5.18.0-kali7-amd64, x86_64: installed
rtl8821CU/5.4.1: added
phy#0
        Interface wlan1
                ifindex 4
                wdev 0x2
                addr d6:6f:46:6f:84:59
                type managed
                txpower -100.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 90:61:0c:a3:12:a2
                type managed
                txpower -100.00 dBm
global
country ES: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
        (5725 - 5875 @ 80), (N/A, 13), (N/A)
        (5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)
sir,
I did change but the problem is still the same
I am using 4 Ghz router 
I am also not able to connect to open  wifi

morrownr, 2022-09-15:
Okay. Let's try some more things.

> rtl8821CU/5.4.1: added

The about needs to be removed. Do you know how to do that with dkms?

Does this adapter have bluetooth capability?

> Interface wlan0

You have a wifi interface. Have you tried to connect to a channel in the 2.4 GHz band?

Nabghpok, 2022-09-16:
yes I tried to connect the 2.4 GHz band it also doesn't work 
It does not support  bluetooth.
No, i don't know how to do with that with dkms can you guide me sir ?

morrownr, 2022-09-16:
> No, i don't know how to do with that with dkms can you guide me sir ?

$ sudo dkms remove -m rtl8821CU -v 5.4.1 --all

Then...

$ dkms status

And it should be gone.

Have you tried forgetting existing connections?

I'm trying to remember how to do that on Kali and can't think of it right now, Maybe you can search.

Nabghpok, 2022-09-16:
yes, sir, I tried to connect by forgetting existing connections but it didn't work.

dkms status:
 rtl8821cu/5.12.0, 5.18.0-kali7-amd64, x86_64: installed

still, the problem is same.
I installed the same driver in Linux mint but there I saw a popup message "activation of network connection fail"

morrownr, 2022-09-16:
Are Kali and Mint running inside Virtualbox?

Which versions of Kali and Mint are you using?

The problem you describe is consistent with using the wrong password or security that is not supported on one end or the other of the connection. What is the security setting on the wifi ap? WPA3 Mixed?

FYI: This repo averages of 1,000 hits per week. My point being that this repo is heavily used but generates very few problem reports so we are likely looking for something that is different than what most users are doing.

Nabghpok, 2022-09-17:
yes, sir kali and mint are running inside the virtual box.
At first, I also think I put the wrong password but when I change my wifi password to "12345678" I face the same problem " Activation of network connection failed " 
kali version :
PRETTY_NAME="Kali GNU/Linux Rolling"
NAME="Kali GNU/Linux"
ID=kali
VERSION="2022.3"
VERSION_ID="2022.3"
VERSION_CODENAME="kali-rolling"
ID_LIKE=debian
ANSI_COLOR="1;31"

NAME="Linux Mint"
VERSION="21 (Vanessa)"
ID=linuxmint
ID_LIKE="ubuntu debian"
PRETTY_NAME="Linux Mint 21"
VERSION_ID="21"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.linuxmint.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=vanessa
UBUNTU_CODENAME=jammy

morrownr, 2022-09-17:
> kali and mint are running inside the virtual box.

https://null-byte.wonderhowto.com/forum/wifi-hacking-attach-usb-wireless-adapter-with-virtual-box-0324433/

Let me know if the above link helps. FYI: The link was provided in the FAQ.

Also, Unless there is a specific need, I generally recommend people go with dual boot instead of VM's. You might consider that.

Nick

Nabghpok, 2022-09-18:
sir my wireless adaptor USB is 2.0 so I am trying Virtualbox USB 2.0.
so I am not able to connect wifi but when I try In 3.0 USB port  it is connected now
my Wireless usb adaptor is USB 2.0
I got it . 
Thank you, sir!

morrownr, 2022-09-18:
Hey @Nabghpok 

You lost me with the USB3 vs. USB2 stuff. The adapter is USB2 and will work fine in either type of port but it will be in USB2 mode.

> I got it .

I'll take that to mean your adapter is able to do what you want it to do now.

Great,

Nick

prazedotid, 2022-10-24:
Can confirm @Nabghpok's solution works. I have a Ubuntu 22.04 system with the TOTOLINK A650UA adapter, the adapter keeps failing to connect and asking for the password even though I have entered it correctly. Plugging the adapter into a USB 3.0 port fixes this

Nabghpok, 2022-10-30:
@prazedotid  yes I did test on both kali Linux and Ubuntu it works when I change it fix my problem and some problem fixed with the above commands provided sir @morrownr Thanks!

morrownr, 2022-10-30:
Gents, let me get this straight: Both of you using Virtualbox and if you use a USB3 port and the proper settings, the adapters works fine but not if the adapter is plugged into a USB2 port?

prazedotid, 2022-10-31:
In my case I wasn't using Virtualbox though, it's an Ubuntu 22.04 system on bare metal. But yes the Wi-Fi connection only works if I plug the adapter into a USB3 port. As for settings I actually only followed the guides in the README without configuring any other thing.

morrownr, 2022-10-31:
Hi @prazedotid 

Well, I'm scratching my head. I've seen a lot of strange things with USB ports so let's not close this so others can read it.

Thanks.