morrownr/8821cu-20210118 #44: Exports duplicate symbol __cfg80211_alloc_event_net_skb

gorghino, 2022-09-20:
Hi!
I'm trying to use a Cudy WU650 with my Nvidia NX (w/ HYAI-310UAV aarch64 carrier board).
I succesfully used your drivers for an older carrier board who supported Ubuntu 16.04 only but I cannot use the Wifi Adapter with Ubuntu 20.04.

DKMS building worked fine but I got this error when I plug my Wifi adapter:

```
[ 2652.850457] usb 1-2.1: new high-speed USB device number 8 using tegra-xusb
[ 2652.951087] usb 1-2.1: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[ 2652.951109] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2652.951116] usb 1-2.1: Product: 802.11ac NIC
[ 2652.951123] usb 1-2.1: Manufacturer: Realtek
[ 2652.951130] usb 1-2.1: SerialNumber: 123456
[ 2653.083281] cfg80211: exports duplicate symbol __cfg80211_alloc_event_net_skb (owned by kernel)
```

This is the debug commands you requested:

```
nvidia@tegra-ubuntu:~/Downloads/rtl8821cu_linux$ sudo uname -mr; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev; iw reg get
5.10.104-tegra aarch64
-bash: mokutil: command not found
Bus 002 Device 002: ID 0451:8240 Texas Instruments, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 046d:c53f Logitech, Inc. 
Bus 001 Device 008: ID 0bda:c811 Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 0451:8242 Texas Instruments, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
rtl8821cu, 5.4.1, 5.10.104-tegra, aarch64: installed
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), NO-IR
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, NO-IR
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, NO-IR
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, NO-IR
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, NO-IR
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, NO-IR
	(5735 - 5835 @ 80), (N/A, 20), (N/A), NO-IR
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

```

morrownr, 2022-09-20:
Hi @gorghino 

Let me see...

> rtl8821cu, 5.4.1, 5.10.104-tegra, aarch64: installed

That line tells me you have an old version of a driver for the 8821cu installed. It will conflict with this driver. Do you know how to uninstall it?

Which script did you use to install this driver?  I ask because I don't see it installed with dkms.

> country 00: DFS-UNSET

Your country code is not set. This is usually not a problem with the 2.4 GHz band but it will be  problem with the 5 GHz band. Do you know how to set it?

Which ARM script did you run in Step 9? I think it is step 9.

Can you provide links to the hardware and distro you are using?

Thanks,

Nick

gorghino, 2022-09-21:
Hi @morrownr ,
The version mismatch was probably an attempt I did with another repository but I had the same error I got with your drivers. Sorry for the error.
I'm retrying with your code and a clean install now:

```
nvidia@tegra-ubuntu:~/8821cu-20210118$ git pull
Already up to date.
nvidia@tegra-ubuntu:~/8821cu-20210118$ sudo uname -mr; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev; iw reg get
5.10.104-tegra aarch64
-bash: mokutil: command not found
Bus 002 Device 002: ID 0451:8240 Texas Instruments, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 17ef:60d0 Lenovo 
Bus 001 Device 003: ID 046d:c53f Logitech, Inc. 
Bus 001 Device 002: ID 0451:8242 Texas Instruments, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), NO-IR
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, NO-IR
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, NO-IR
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, NO-IR
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, NO-IR
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, NO-IR
	(5735 - 5835 @ 80), (N/A, 20), (N/A), NO-IR
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
```

I run `./ARM64_RPI.sh` and I installed the dkms driver:

```
Running install-driver.sh version 20220821
5.10.104-tegra
aarch64
Starting installation...
Copying source files to: /usr/src/rtl8821cu-5.12.0
Copying 8821cu.conf to: /etc/modprobe.d
Copying rtw88_8821cu.conf to: /etc/modprobe.d

Creating symlink /var/lib/dkms/rtl8821cu/5.12.0/source ->
                 /usr/src/rtl8821cu-5.12.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j6 KVER=5.10.104-tegra KSRC=/lib/modules/5.10.104-tegra/build....................................................................................................................
cleaning build area...

DKMS: build completed.

8821cu.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.104-tegra/updates/dkms/

depmod........

DKMS: install completed.
The driver was installed successfully.
Do you want to edit the driver options file now? [y/N] n
Do you want to reboot now? (recommended) [y/N] N

nvidia@tegra-ubuntu:~/8821cu-20210118$ sudo uname -mr; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev; iw reg get
5.10.104-tegra aarch64
-bash: mokutil: command not found
Bus 002 Device 002: ID 0451:8240 Texas Instruments, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 17ef:60d0 Lenovo 
Bus 001 Device 003: ID 046d:c53f Logitech, Inc. 
Bus 001 Device 002: ID 0451:8242 Texas Instruments, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
rtl8821cu, 5.12.0, 5.10.104-tegra, aarch64: installed
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), NO-IR
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, NO-IR
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, NO-IR
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, NO-IR
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, NO-IR
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, NO-IR
	(5735 - 5835 @ 80), (N/A, 20), (N/A), NO-IR
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
```

I plug Cudy WU650:

```
[ 1002.229584] usb 1-2.1: new high-speed USB device number 5 using tegra-xusb
[ 1002.334300] usb 1-2.1: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[ 1002.334313] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1002.334319] usb 1-2.1: Product: 802.11ac NIC
[ 1002.334326] usb 1-2.1: Manufacturer: Realtek
[ 1002.334332] usb 1-2.1: SerialNumber: 123456
[ 1002.490065] cfg80211: exports duplicate symbol __cfg80211_alloc_event_net_skb (owned by kernel)
```

About the country flag, I haven't seen how to set it yet. 

Thank you!

morrownr, 2022-09-21:
Hi @gorghino 

> About the country flag, I haven't seen how to set it yet.

Replace US with your country code. If you do not know your country code, tell me your country:

$ sudo iw reg set US

To check:

$ sudo iw reg get

Now to the problem of why the adapter is not working: I don't know. That it compiles and installs is a good sign. However, my ability to work this issue will be hampered by the fact that I do not have the hardware. I do have Raspberry Pi hardware (3B and 4B) and a Rock 4SE. They are ARM64 boards. The best bet is probably burning an sd card with Ubuntu Desktop 22.04.1 LTS for the Raspberry Pi 4. Will this duplicate the problem so I can work it? I don't know. Are you interested in me pursuing this option?

Let me throw something out:

When I see Linux users exploring the various hardware options that are available, if the hardware is a lesser used variety, I recommend users go with usb wifi adapters that use in-kernel drivers. This is usually a far more reliable option as patching these out-of-kernel Realtek drivers is problematic. The following link is to the Main Menu of this site:

https://github.com/morrownr/USB-WiFi

Recommend reading menu items 1 and 2. Item 2 contains links to around 70 adapters that use in-kernel drivers. Something that uses a mt7612u chipset might be good for your use case.

Cheers,

Nick

closed 2022-10-22 by morrownr