morrownr/8821cu-20210118 #45: Totolink AC650UA can't connect on Pop OS

wahyupriadi95, 2022-09-23:
**A brief description of the problem**
Followed the steps until install the driver then reboot. Adapter detected, attempt to connect but it keep asking password. Please I need a help.

**The name and version of the Linux distro** 
PopOS 22.04

**The name and model of the computer** 
Ryzen 5 2600, Asrock B450 Pro

**The information obtained** 
```
5.19.0-76051900-generic x86_64
SecureBoot disabled
Platform is in Setup Mode
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 003 Device 002: ID 0bda:8771 Realtek Semiconductor Corp. Bluetooth Radio
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 2717:ff88 Xiaomi Inc. Mi/Redmi series (RNDIS + ADB)
Bus 001 Device 003: ID 0bda:2006 Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
rtl8821cu/5.12.0, 5.19.0-76051900-generic, x86_64: installed
system76/1.0.14~1643391291~22.04~78ede46, 5.19.0-76051900-generic, x86_64: installed
system76_acpi/1.0.2~1659568831~22.04~a88dcce, 5.19.0-76051900-generic, x86_64: installed (original_module exists)
system76-io/1.0.2~1655490480~22.04~0217576, 5.19.0-76051900-generic, x86_64: installed
phy#0
	Interface wlx5c925ed335a2
		ifindex 3
		wdev 0x1
		addr 5c:92:5e:d3:35:a2
		type managed
		txpower -100.00 dBm
global
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (6, 20), (N/A)
	(2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
```

morrownr, 2022-09-23:
Hi @wahyupriadi95 

> country 00: DFS-UNSET

Your country code is not set,

$ sudo iw reg SET US
replace the US with your country code

$ sudo iw reg get
to check if it is set now

That will have a bad effect on 5 GHz channels.

Tell me about your AP/Router:

Is it dual band?
If so, are both bands using the same name?
What security setting is in use? WPA2, mixed, WPA3?

wahyupriadi95, 2022-09-23:
> Is it dual band? If so, are both bands using the same name? What security setting is in use? WPA2, mixed, WPA3?

Thanks for the reply.
The router indeed has dual band. I set both bands in different names. Security WPA/WPA2-Personal (Checked via android).

morrownr, 2022-09-23:
Were you able to set the country code?

The information you provided rules out many problems. You even have a wifi interface ready to go.

The specific chipset can be a little problematic at times so we may need to dig in: The key document to focus on now is...

8821cu.conf

That file sets driver options (module paramters). An easy way to access and edit it is:

$ sudo ./edit-options.sh

There is only one active line in the file. It starts with the word "options". There is a lot of documentation in this file. Please copy the options line and send it to me. It will help a lot of I know what country you are located in.

wahyupriadi95, 2022-09-24:
> Were you able to set the country code?

Yes and here's the updated info

```
5.19.0-76051900-generic x86_64
SecureBoot disabled
Platform is in Setup Mode
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 003 Device 002: ID 0bda:8771 Realtek Semiconductor Corp. Bluetooth Radio
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 2717:ff88 Xiaomi Inc. Mi/Redmi series (RNDIS + ADB)
Bus 001 Device 003: ID 0bda:2006 Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
rtl8821cu/5.12.0, 5.19.0-76051900-generic, x86_64: installed
system76/1.0.14~1643391291~22.04~78ede46, 5.19.0-76051900-generic, x86_64: installed
system76_acpi/1.0.2~1659568831~22.04~a88dcce, 5.19.0-76051900-generic, x86_64: installed (original_module exists)
system76-io/1.0.2~1655490480~22.04~0217576, 5.19.0-76051900-generic, x86_64: installed
phy#0
	Interface wlx5c925ed335a2
		ifindex 3
		wdev 0x1
		addr 5c:92:5e:d3:35:a2
		type managed
		txpower -100.00 dBm
global
country ID: DFS-JP
	(2400 - 2483 @ 40), (N/A, 26), (N/A), NO-OUTDOOR
	(5150 - 5350 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
	(5725 - 5825 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
```

> There is only one active line in the file. It starts with the word "options". There is a lot of documentation in this file. Please copy the options line and send it to me. It will help a lot of I know what country you are located in.

`options 8821cu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_dfs_region_domain=0`

For your information, this driver works well on fedora 36 kernel 5.19. I don't know why it has some issue on Pop OS 22.04.

morrownr, 2022-09-24:
Try: (remember that you need to reboot after making changes)

options 8821cu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_country_code-JP 

If it does not work then try:

options 8821cu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_country_code-JP rtw_RFE_type=7

> For your information, this driver works well on fedora 36 kernel 5.19. I don't know why it has some issue on Pop OS 22.04.

Good info. This probably means it has to do with proper passing of the country code.

wahyupriadi95, 2022-09-24:
> options 8821cu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_country_code-JP
> 
> options 8821cu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_country_code-JP rtw_RFE_type=7
 
Unfortunately, both of these don't work.
I got my old pcie adapter Tp-link RTL8192EE works without driver. But well, it doesn't have 5 Ghz capability. Does pcie adapter always compatible with linux at start?

morrownr, 2022-09-24:
I'm getting short on ideas. This chipset is one where there have been a few individuals where we never could get things going. It seems to happen at times on adapters that also have BT.

> Does pcie adapter always compatible with linux at start?

Most modern PCIe adapters are supported with in-kernel drivers. I'd recommend cards with Mediatek, Intel or Atheros chipsets in that order. There are also many usb wifi adapters that work with in-kernel adapters. The Main Menu of this site is:

https://github.com/morrownr/USB-WiFi

Menu items 1 and 2 should get you off to a good start. Ask questions in `Issues`.

wahyupriadi95, 2022-09-24:
> I'm getting short on ideas. This chipset is one where there have been a few individuals where we never could get things going. It seems to happen at times on adapters that also have BT.

It's okay. Probably, USB Wifi Adapter not going well if the brand didn't provide official driver for linux.

> Most modern PCIe adapters are supported with in-kernel drivers. I'd recommend cards with Mediatek, Intel or Atheros chipsets in that order. There are also many usb wifi adapters that work with in-kernel adapters.

I see, the listed products mostly not available yet in my country, but thanks for the recommendation.

morrownr, 2022-10-22:
Closing the issue for now but welcome you back if you have more questions.

closed 2022-10-22 by morrownr