morrownr/8821cu-20210118 #47: Cannot connect to WiFi (Debian 11)

IntoTheVoid-900, 2022-10-01:
Hi!
I am trying to make a server out of my old laptop with Debian. I am able to build the drivers and install it with make fine, but after rebooting it doesnt seem like any changes have been made. I still cannot connect to WiFi. I have redone the tutorial about 7 times now. Help?

```
5.10.0-18-amd64 x86_64
SecureBoot disabled
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 30c9:0035 Luxvisions Innotech Limited HP TrueVision HD Camera
Bus 001 Device 004: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 003: ID 0bda:b00a Realtek Semiconductor Corp. Realtek Bluetooth 4.2 Adapter
Bus 001 Device 002: ID 0781:55a1 SanDisk Corp. Cruzer Spark
Bus 001 Device 012: ID 0b05:7774 ASUSTek Computer, Inc. Zenfone GO (ZB500KL) (RNDIS mode)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
-bash: rfkill: command not found
-bash: dkms: command not found
-bash: iw: command not found
-bash: iw: command not found
```

morrownr, 2022-10-01:
Hi @IntoTheVoid-900 

Based on what you posted, I don't think you have a usb wifi adapter. That means this usb wifi adapter driver is not the right driver. However, let me help if I can:

> Bus 001 Device 003: ID 0bda:b00a Realtek Semiconductor Corp. Realtek Bluetooth 4.2 Adapter

That is bluetooth, not wifi. I don't see a usb wifi adapter.

Run `$ lspci` and post the results.

IntoTheVoid-900, 2022-10-03:
Sorry, this is not a USB wifi dongle. I was hoping the driver would still work.

closed 2022-10-03 by IntoTheVoid-900

morrownr, 2022-10-03:
> Sorry, this is not a USB wifi dongle. I was hoping the driver would still work.

This driver won't work but that doesn't mean I can't help. Show me the results of...

$ lspci

If the wifi is based on a 8821ce chipset, the there is an in kernel driver. It may not be working due to lack of firmware. Debian has this thing about firmware so it may not be a driver you are looking for.

What you could do to test is download and burn the latest release of Xubuntu, 22.04.1, burn it to a flash drive and boot. If the wifi works, you have your answer. Xubuntu will have the firmware.

Nick

IntoTheVoid-900, 2022-10-05:
Thank you for offering support, but after I closed this issue I got [This driver](https://github.com/tomaspinho/rtl8821ce) working. I now realize i misread the CE for CU in my old OS.