morrownr/8821cu-20210118 #48: Concurrent mode 5Ghz AND 2Ghz AP ?

tduong-cloud, 2022-10-04:
Hello,

First of all, many thanks for the great work. The driver was really easy to install.

I have a Raspberry Pi 4 b 8GO, on which I have installed RaspAP to replace my ISP router. I have bought a Mercusys MU6H in order to extend my wifi signal in my house.

The objective was to :
- extend 2.4 GHz network
- add 5GHz network (gaming)

So I have installed the driver, with the concurrent mode activated. I has created two new wlan interface, wlan1 and wlan2.

I have configured wlan1 to be my 5Ghz network and wlan2 the 2.4Ghz one. When I start both AP at once, I only got 2.4Ghz on both wifi (and also same channel). If I start wlan1 only, it works as expected.

So I was wondering whether I have missed something? or is it a hardware limitation? or concurrent mode is more to be able to work as AP and also client at the same time?

Many thanks for your time, if you are able to answer this.

morrownr, 2022-10-04:
Hi @tduong-cloud 

Let me see,,,

> First of all, many thanks for the great work. The driver was really easy to install.

That is great to hear. I've been working on `install-driver.sh` lately trying to make it better and I am not seeing problems so I think it is time to tweak the docs.

> I have a Raspberry Pi 4 b 8GO, on which I have installed RaspAP to replace my ISP router. I have bought a Mercusys MU6H in order to extend my wifi signal in my house.

Are you replacing your current wifi router/AP or are you extending it? Those two different things and will dramatically change my recommendation.

> So I have installed the driver, with the concurrent mode activated. I has created two new wlan interface, wlan1 and wlan2.

These usb wifi adapters only have one radio. They can do one band or the other but not both at the same time. Both interfaces in concurrent mode use the same channels so maybe something like a repeater could be done.

For a dual band RasPi4B setup, you either need two adapters or one adapter and the internal wifi. I have an extensive guide available:

https://github.com/morrownr/USB-WiFi

That is the site Main Menu. Select item 9.

That guide works well for extending wireless or adding wireless where their is only a router. If you are replacing a router, I recommend using OpenWRT but that will work better with adapters that have in-kernel driver support. What you may want to do is read Main Menu items 1 and 2. Then get back to me and tell me what direction you want to go and maybe I can help.

Nick

tduong-cloud, 2022-10-04:
Hey,

Thanks for your answer !

> Are you replacing your current wifi router/AP or are you extending it? Those two different things and will dramatically change my recommendation.

I am currently building my own home server, and doing so, I am replacing my wifi router. I have my Pi wired connected (eth0) to my current wifi router (to get internet connection). On my Pi, I have RaspAP, and OpenMediaVault (NAS) installed. 

I am planning to buy a Geekworm Gemini case for it in order to setup a dual bay 2.5 Sata NAS. The issue is that the shielding of the metal case could cause loss of 2.4 Ghz wifi. And I was afraid it could be the same for 5Ghz.

So I decided to purchase the Mercusys MU6H and plug it into one of my usb3 (can I spare this usb3 ? I believe that if I connect it on a usb2, the speed will decrease significantly). I understand from your message that I can't setup a dual band 2.4 and 5 Ghz AP at the same time with the MU6H.

Then I think the solution for me would be to setup my 2.4 Ghz (801.11n) with the MU6H and my 5Ghz (801.11ac) with the Pi internal wifi - hoping the metal case would not cause issue. This is my current setup (I have not bought the gemini case yet).

Finally, I thought the MU6H would provide me with more wifi range than my current wifi router. The Pi is in my living room and the connection is not really good in my bedroom. Is there something I can do about it ?

Many thanks again :).

morrownr, 2022-10-04:
Good day @tduong-cloud 

> So I decided to purchase the Mercusys MU6H and plug it into one of my usb3 (can I spare this usb3 ? I believe that if I connect it on a usb2, the speed will decrease significantly).

The Mercusys MU6H is a USB2 adapter so plugging it into a USB3 port will not get you any additional speed. A USB2 port will supply more than enough speed.

> I understand from your message that I can't setup a dual band 2.4 and 5 Ghz AP at the same time with the MU6H.

That is correct. The MU6H can operate in the 2.4 GHz band or in the 5 GHz band but not both at the same time. You need two wifi radios for dual band or three for tri band. Adapters have one radio as does the Pi4B internal wifi setup.

In fact, this message is coming to you through my Pi4B which is operating as an AP. I am currently testing a new adapter that uses the Mediatek mt7921au chipset. This is a tri band WiFi 6e chipset.

> The issue is that the shielding of the metal case could cause loss of 2.4 Ghz wifi. And I was afraid it could be the same for 5Ghz.

You are correct. A metal case can really hamper the range of the internal wifi but I always turn the internal wifi off and use usb wifi adapters as good wifi adapters will provide much better range than what you get from the internal wifi. 

> I thought the MU6H would provide me with more wifi range than my current wifi router. The Pi is in my living room and the connection is not really good in my bedroom. Is there something I can do about it ?

The location of the Pi can make a difference. An elevated location that is at least a couple of feet away from any walls would be best. I have several adapters as I test adapters and provide the information here on this site. I provided the link to the Main Menu earlier. The adapter that is best at range for use with a Pi AP setup is the ALFA AWUS036ACHM. The internal amp and antenna that ALFA uses on this adapter give it the best range of any dual band adapter that I have ever tested. If you go to the following list, you can scroll down to see the adapter and read about it:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

Based on what I am hearing you say, my thoughts are that you may want to think about going with a case that allows the internal wifi to work as good as it can... something like this:

https://www.amazon.com/dp/B07T2CPC2H

I use that case and have tested it with the internal 2.4 GHz wifi and it does not hinder the signal. The Pi4B internal wifi is really not bad for 2.4 GHz. Then using an ALFA AWUS036ACHM plugged into a USB2 port could provide the 5 GHz wifi and you would have about as good of range as you can get.

It sounds like you are learning how to make a setup like this work. Asking a lot of questions helps get a setup that does what you want. To me, the key is getting a usb wifi adapter that has the range and stability is key.

Nick

tduong-cloud, 2022-10-04:
Hello,

> The Mercusys MU6H is a USB2 adapter so plugging it into a USB3 port will not get you any additional speed. A USB2 port will supply more than enough speed.

Happy to hear this, I will be able to spare an usb3 port then.

> You are correct. A metal case can really hamper the range of the internal wifi but I always turn the internal wifi off and use usb wifi adapters as good wifi adapters will provide much better range than what you get from the internal wifi.

Yes, if the metal case would not let the 5ghz wifi signal out, then I will disable it. I will use the MU6H to transmit a 5GHz signal and keep my original router's wifi enable. I have read in the documentation that the metal case should not cause too much trouble for a 5Ghz signal though.

> The location of the Pi can make a difference. An elevated location that is at least a couple of feet away from any walls would be best. 

Yes, the location for my routers are good.

> I have several adapters as I test adapters and provide the information here on this site. I provided the link to the Main Menu earlier. The adapter that is best at range for use with a Pi AP setup [...]

Unfortunately, where I live, I don't have much choice, and I want to keep it a bit esthetic. That's why the look of the kind of adapter like the MU6H is good to me.

About the case, I have found this one : 
https://www.amazon.com/Acrylic-Cooling-Raspberry-Shield-Expansion/dp/B07MP232G7?ref_=ast_sto_dp

Unfortunately, there is only one slot for 2.5 HDD/SATA

I think that for now, you have been able to answer all my questions. I just need to find the right hardware, especially the case.

Many thanks again for your support.

morrownr, 2022-10-04:
You are welcome and good luck.