morrownr/8821cu-20210118 #49: error on new kernels

ktistes, 2022-10-13:
Hi, after summer I updated my system (Void linux) and dkms failed to compile the driver for the new kernel linux5.18-5.18.19_1, but installing the newest version of your driver solved. Now, with the new kernel linux5.19-5.19.14_1 same problem is back again. I attach the log file. Thank you.
[make.log](https://github.com/morrownr/8821cu-20210118/files/9776084/make.log)

morrownr, 2022-10-13:
Hi @ktistes 

Did you update this driver on your system before upgrading to kernel 5.19? If not:

$ git pull
$ sudo ./remove-driver.sh
$ sudo ./install-driver.sh

These out-of-kernel drivers are a challenge. one of these days I am going to convince you to move over to an adapter that uses in-kernel drivers. In fact, the mt7921au chipset is now available and it has BT 5.2 support. The driver is in kernel 5.19 and we expect several new adapters of the next few months. I can put you on more info if interested.

Nick

ktistes, 2022-10-14:
Driver new update solved the issue.  As for tips for a better hardware, I would greatly appreciate your contribution to my peace of mind.

morrownr, 2022-10-14:
> Driver new update solved the issue.

Good. I do my best keep the driver up to date with the latest kernel changes but I can't update what is on user harddrives. You just have to remember to refresh the driver when you are bumping up the kernel version. That isn't needed with the in-kernel drivers and I am an advocate of using the in-kernel drivers.

The main repo here is:

https://github.com/morrownr/USB-WiFi

It has a Main Menu and you can ask questions and discuss things in Issues or Discussions. Main Menu item 2 shows a long list of adapters that use in kernel drivers. A new AX3000 chipset was recently released and it already has an in-kernel driver. Come on over and join the fun.

Nick