morrownr/8821cu-20210118 #51: Build failure in 6.1-rc1

CueXXIII, 2022-10-17:
Building the current source with the upcoming 6.1-rc1 kernel fails with:
```
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/os_intfs.o
/var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/osdep_service.c: In function ‘rtw_random32’:
/var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/osdep_service.c:2918:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-function-declaration]
 2918 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                prandom_u32_max
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/osdep_service.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/os_intfs.c: In function ‘rtw_os_ndev_register’:
/var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/os_intfs.c:2151:9: error: too many arguments to function ‘netif_napi_add’
 2151 |         netif_napi_add(ndev, &adapter->napi, rtw_recv_napi_poll, RTL_NAPI_WEIGHT);
      |         ^~~~~~~~~~~~~~
In file included from /var/lib/dkms/rtl8821cu/5.12.0/build/include/osdep_service_linux.h:31,
                 from /var/lib/dkms/rtl8821cu/5.12.0/build/include/osdep_service.h:54,
                 from /var/lib/dkms/rtl8821cu/5.12.0/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/os_intfs.c:17:
./include/linux/netdevice.h:2562:1: note: declared here
 2562 | netif_napi_add(struct net_device *dev, struct napi_struct *napi,
      | ^~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/rtl8821cu/5.12.0/build/os_dep/linux/os_intfs.o] Error 1
```

I'm on commit 75f2145892149349ef57a584296e384965b36a77 - after that only changes to the documentation and the scripts were checked into git that would not affect compilation.

Attached is the make.log from dkms and my kerel config

CueXXIII, 2022-10-17:
Ok, these didn't go through on the 1st post… 
[make.log](https://github.com/morrownr/8821cu-20210118/files/9799923/make.log)
[config-6.1.0-rc1-pinguin20221017.txt](https://github.com/morrownr/8821cu-20210118/files/9799924/config-6.1.0-rc1-pinguin20221017.txt)

morrownr, 2022-10-17:
Hi @CueXXIII 

Thanks for the report. I will work on this as soon as I can.

Nick

morrownr, 2022-10-17:
Hi again @CueXXIII 

I just merged a lot of changes and additions to include the fix for your issue. It is compiling clean here. 

I really need you to test it and let me know.

It is probably easiest to do the following:

$ git pull
$ sudo ./remove-driver.sh
$ sudo ./install-driver.sh

Nick

CueXXIII, 2022-10-18:
Thanks for the really fast reply. It did compile cleanly and is running so far with no problems.