morrownr/8821cu-20210118 #52: Very low speed and unstable on 6.0.2-2-MANJARO

Electrominch, 2022-10-18:
I have a AC650 MERCUSYS (MU6H)
Wi-Fi begins work after install this driver, but with very low speed.
Screen1 - on my phone
Screen2 - pc 6.0.2-2-MANJARO
[image]
[image]

morrownr, 2022-10-22:
Hi @Electrominch 

Can I get you follow the Problem Reporting Guidlines as shown in the pinned issue? It is not clear to me what could be causing this.

Nick

Electrominch, 2022-10-22:
https://github.com/morrownr/8821cu-20210118/blob/main/8821cu.conf#L60
this is solve
sorry/thx

Electrominch, 2022-10-22:
`sudo uname -mr; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev; iw reg get                 ✔  25s  
6.0.2-2-MANJARO x86_64
The application mokutil is not installed. It may be found in the following packages:
  extra/mokutil 0.6.0-1 /usr/bin/mokutil
Do you want to Install package mokutil? (y/N)  y
Executing command: pamac install mokutil
Preparing...
Synchronizing package databases...
https://mirror.tuchola-dc.pl/manjaro/stable/core/x86_64/core.db: Unacceptable TLS certificate
http://mirror.infotronik.hu/mirrors/pub/manjaro/stable/core/x86_64/core.db: Error 404
https://mirror.tuchola-dc.pl/manjaro/stable/extra/x86_64/extra.db: Unacceptable TLS certificate
http://mirror.infotronik.hu/mirrors/pub/manjaro/stable/extra/x86_64/extra.db: Error 404
https://mirror.tuchola-dc.pl/manjaro/stable/community/x86_64/community.db: Unacceptable TLS certificate
http://mirror.infotronik.hu/mirrors/pub/manjaro/stable/community/x86_64/community.db: Error 404
https://mirror.tuchola-dc.pl/manjaro/stable/multilib/x86_64/multilib.db: Unacceptable TLS certificate
http://mirror.infotronik.hu/mirrors/pub/manjaro/stable/multilib/x86_64/multilib.db: Error 404
Resolving dependencies...
Checking inter-conflicts...

To install (1):
  mokutil  0.6.0-1    extra  25,7 kB

Total download size: 25,7 kB
Total installed size: 59,3 kB

Apply transaction ? [y/N] y
https://mirror.tuchola-dc.pl/manjaro/stable/extra/x86_64/mokutil-0.6.0-1-x86_64.pkg.tar.zst: Unacceptable TLS certificate
http://mirror.infotronik.hu/mirrors/pub/manjaro/stable/extra/x86_64/mokutil-0.6.0-1-x86_64.pkg.tar.zst: Error 404
Download of mokutil (0.6.0-1) started                                                                                                                               
Download of mokutil (0.6.0-1) finished                                                                                                                              
https://mirror.tuchola-dc.pl/manjaro/stable/extra/x86_64/mokutil-0.6.0-1-x86_64.pkg.tar.zst: Unacceptable TLS certificate                                           
http://mirror.infotronik.hu/mirrors/pub/manjaro/stable/extra/x86_64/mokutil-0.6.0-1-x86_64.pkg.tar.zst: Error 404
Checking keyring...                                                                                                                                            [1/1]
Checking integrity...                                                                                                                                          [1/1]
Loading packages files...                                                                                                                                      [1/1]
Checking file conflicts...                                                                                                                                     [1/1]
Checking available disk space...                                                                                                                               [1/1]
Installing mokutil (0.6.0-1)...                                                                                                                                [1/1]
Running post-transaction hooks...
Arming ConditionNeedsUpdate...                                                                                                                                 [1/2]
Refreshing PackageKit...                                                                                                                                       [2/2]
Transaction successfully finished.
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 011: ID 0ac8:c40a Z-Star Microelectronics Corp. A4 TECH USB2.0 PC Camera J
Bus 001 Device 008: ID 09da:9090 A4Tech Co., Ltd. XL-730K / XL-750BK / XL-755BK Mice
Bus 001 Device 006: ID 1a2c:4f7a China Resource Semico Co., Ltd USB Keyboard
Bus 001 Device 004: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 003: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 010: ID 048d:5702 Integrated Technology Express, Inc. ITE Device
Bus 001 Device 009: ID 0951:170b Kingston Technology HyperX Cloud Stinger Core Wireless + 7.1
Bus 001 Device 005: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
rtl8821cu/5.12.0, 6.0.2-2-MANJARO, x86_64: installed
rtl8821CU/5.4.1, 5.15.74-3-MANJARO, x86_64: installed
rtl8821CU/#MODULE_VERSION#: added
phy#0
        Interface wlp0s20f0u8
                ifindex 3
                wdev 0x1
                addr c0:25:2f:dc:bd:4d
                ssid KALINKA
                type managed
                channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
                txpower 17.00 dBm
global
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (6, 20), (N/A)
        (2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

`

Electrominch, 2022-10-22:
This is issue of my wi-fi router....

closed 2022-10-22 by Electrominch

morrownr, 2022-10-22:
Glad you got it worked out.

I did notice that your Country Code is not set:

> country 00: DFS-UNSET

That will cause 5 GHz to basically be not usable so your adapter will try to use 2.4 GHz, Do you know how to set the country code?

Electrominch, 2022-10-22:
Yes I do, thx. Is RU valid code?

morrownr, 2022-10-22:
> Is RU valid code?

yes