morrownr/8821cu-20210118 #54: Can rtl8811cu do deauth by aireplay in 2.4G and 5G?

Sa1Vation, 2022-10-21:
Hi morrownr:
First of all, please forgive my poor English.
This is my first time to learn something driver. I have already read most of your document, but I still haven't understand all of that.
However, I have an important thing to confirm before I figure out all of your documents. I have a `rtl8811cu` wireless card and have a simple question to ask you. Can `rtl8811cu` do deauth attack in by `aireplay` 2.4G and 5G? If `rtl8811cu` can't do this, I need to buy a new card. 😀

Sa1Vation, 2022-10-22:
[update] do deauth successfully. My card is tenda u10 which rtl8811cu inside.

closed 2022-10-22 by Sa1Vation

morrownr, 2022-10-22:
Hi @Sa1Vation 

Been busy this week.

> Can rtl8811cu do deauth attack in by aireplay 2.4G and 5G?

When last I check it would do a deauth. It seems from your second message that you got it working.

FYI: The Main Menu here is located at:

https://github.com/morrownr/USB-WiFi

You might be interested inmenu item 10 as it has to do with monitor mode.

It might be interesting if you start reading from menu item 1 and then 2 so as to add to your knowledge.

(your language skills are good) Hablamos espanol si es necesario.

Nick

Sa1Vation, 2022-10-24:
Hey Nick:
Thanks for reply. You can call me Chou if you like.
I made deauth work but a little difference with your process given in `Monitor_Mode.md`.

1. Whatever channel I changed, the `iw dev` always show me my card is on **channel 1**. I have already try to designate a channel when using `start_mon.sh` or switch a channel manually by commandline. But none of them work.
2. When I do deauth attack to 2.4g WiFi, I must add -D option to `aireplay-ng`, otherwise the Error `wlan0mon is on channel 1, but the AP uses channel ×` would occur. 

Here is my guess: It seems like the card is already change its channel. but `iw dev` always show it is on channel 1.

802.11 is quite interesting. BTW, is there a way to against deauth attack except buying a new route which support 802.11w?

morrownr, 2022-10-24:
Hi Chou,

> BTW, is there a way to against deauth attack except buying a new route which support 802.11w?

If memory serves me correct, undesired deauth is the main reason 11w was developed. However, if you have wifi router that does not support it, and you have a router supported by OpenWRT, you are in luck. What specific wifi router do you have? I'll check if it is supported.

Let me plug my 8811cu based adapter in and see if I can answer some questions...