morrownr/8821cu-20210118 #57: failing to install on Raspberry Pi 3B+

tommy1223, 2022-10-27:
Hi, I fail to install the driver on Raspberry Pi 3B+ with the following error:

sudo ./install-driver.sh
Running install-driver.sh version 20221018
5.15.61-v8+
aarch64
Installing rtw88_8821cu.conf to: /etc/modprobe.d
Installing 8821cu.conf to: /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to: /usr/src/rtl8821cu-5.12.0

Creating symlink /var/lib/dkms/rtl8821cu/5.12.0/source ->
                 /usr/src/rtl8821cu-5.12.0

DKMS: add completed.
Error! Your kernel headers for kernel 5.15.61-v8+ cannot be found.
Please install the linux-headers-5.15.61-v8+ package,
or use the --kernelsourcedir option to tell DKMS where it's located
An error occurred. dkms build error = 1
Please report this error.
Please copy all screen output and paste it into the report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

May I have some help here? Thanks

morrownr, 2022-10-28:
Hi @tommy1223 

> May I have some help here?

I will try to help.

You don't have the header files installed and they are necessary for compiling the driver. Did you run:

sudo apt install -y linux-headers-$(uname -r) build-essential bc dkms git libelf-dev

If not, run it and let me know if the installation is sucessful.

tommy1223, 2022-10-28:
> If not, run it and let me know if the installation is sucessful.

Hi @morrownr

Thank you for your reply. Actually, I did installed the packages using "sudo apt install -y raspberrypi-kernel-headers build-essential bc dkms git", as mentioned in the instructions. (I think I am using ARM64 version of raspberry pi OS, without desktop..) This command gives no error and I was able to install all the packages.

I just tried the command you provided, and the following error came out:

sudo apt install -y linux-headers-$(uname -r) build-essential bc dkms git libelf-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-headers-5.15.61-v8
E: Couldn't find any package by glob 'linux-headers-5.15.61-v8'

Then I tried to search if the linux header for my kernel exists:

sudo apt search linux-headers-$(uname -r)
Sorting... Done
Full Text Search... Done

Seems that my kernel version's header files does not exist? My kernel version is:

uname -r
5.15.61-v8+

Thanks in advance.

morrownr, 2022-10-28:
Hi again @tommy1223 

> I am using Debian Bullseye 64-bit (Raspberry Pi Lite one)

> (I think I am using ARM64 version of raspberry pi OS, without desktop..)

I am confused on what operating system you are using. It is very important to know exactly what operating system you are using. Debian Bullseye is an operating system and it works one way for installation and then the Raspberry Pi OS works another and there are things we need to change depending on whether either of those are 32 bit or 64 bit so please determine and tell me exactly what you are using or...

grab a new sd card and install a fresh copy of the `64 bit Raspberry Pi OS` and we will know exactly what to do:

Follow all installation steps and here are the specific options for steps 3 and 9:

Installation Step 3:

sudo apt install -y raspberrypi-kernel-headers build-essential bc dkms git

Installation Step 9:

./ARM64_RPI.sh

Nick

tommy1223, 2022-10-28:
Hi @morrownr

Thank you for your help. Miraculously the driver installation is successful after a clean install.

Another question raised after the installation, I am using the USB dongle as AP, but from iwconfig command my bitrate is locked to 54Mb/s. I am using hostapd for my AP, is there any recommendation for hostapd.conf or in the driver option? I would like to raise the speed for my AP. Thanks again for your help.
[image]

morrownr, 2022-10-28:
Hi again @tommy1223 

Making progress I see. Good.

>  I am using the USB dongle as AP

I use a Pi for an extra AP as well. I don't recommend using `iwconfig` as it is depreciated and unmaintained. The modern utilities that take its place are: `iw` and `ip`.

I can help with specific questions but it might help if you first take a look at my guide as it has answers to what you are asking:

How_to_Build_a_Bridged_Wireless_Access_Point

You can find it as item 9 at the Main Menu of this site:

https://github.com/morrownr/USB-WiFi

You don't have to follow the guide if you are already setup but it has a lot of the details you need to configure your setup with hostapd for maximum performance.

Nick

tommy1223, 2022-10-31:
@morrownr Thanks! Following the another guide finally achieved full speed from my AP.

Thanks so much for your help!

closed 2022-10-31 by tommy1223