morrownr/8821cu-20210118 #58: Binary /usr/lib/linux-kbuild-3.10.73-Re4son-3/scripts/sign-file not found, modules won't be signed

lucky-zwx, 2022-10-28:
`Binary /usr/lib/linux-kbuild-3.10.73-Re4son-3/scripts/sign-file not found, modules won't be signed`
`Running install-driver.sh version 20221018
3.10.73-Re4son-3.5
aarch64
Installing rtw88_8821cu.conf to: /etc/modprobe.d
Installing 8821cu.conf to: /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to: /usr/src/rtl8821cu-5.12.0
Creating symlink /var/lib/dkms/rtl8821cu/5.12.0/source -> /usr/src/rtl8821cu-5.12.0
Sign command: /usr/lib/linux-kbuild-3.10.73-Re4son-3/scripts/sign-file
Binary /usr/lib/linux-kbuild-3.10.73-Re4son-3/scripts/sign-file not found, modules won't be signed`

lucky-zwx, 2022-10-28:
[image]

morrownr, 2022-10-28:
Hi @lucky-zwx 

> sign-file not found, modules won't be signed

For systems that do not support Secure Mode, this is normal and is not a problem.

Does the driver function properly?

Nick

lucky-zwx, 2022-10-28:
> Hi @lucky-zwx
> 
> > sign-file not found, modules won't be signed
> 
> For systems that do not support Secure Mode, this is normal and is not a problem.
> 
> Does the driver function properly?
> 
> Nick

all is ok

lucky-zwx, 2022-10-28:
but
┌──(root㉿kali)-[~/8821cu-20210118]
└─# make install
install -p -m 644 8821cu.ko  /lib/modules/3.10.73-Re4son-3.5/kernel/drivers/net/wireless/
/sbin/depmod -a 3.10.73-Re4son-3.5
depmod: WARNING: could not open modules.order at /lib/modules/3.10.73-Re4son-3.5: No such file or directory
depmod: WARNING: could not open modules.builtin at /lib/modules/3.10.73-Re4son-3.5: No such file or directory
depmod: WARNING: could not open modules.builtin.modinfo at /lib/modules/3.10.73-Re4son-3.5: No such file or directory
[image]

lucky-zwx, 2022-10-28:
I use the nexus 6p mobile phone and Kali Nethunter. There is a problem with the installation!
┌──(root㉿kali)-[~/8821cu-20210118]
└─# uname -r
3.10.73-Re4son-3.5

┌──(root㉿kali)-[~/8821cu-20210118]
└─# uname -a
Linux kali 3.10.73-Re4son-3.5 #1 SMP PREEMPT Thu Apr 9 21:39:14 AEST 2020 aarch64 GNU/Linux

morrownr, 2022-10-28:
@lucky-zwx 

I am unfamiliar with using Kali Nethunter on a phone. Can you tell me something about it or point me to information?