morrownr/8821cu-20210118 #59: Build Error 10

Tom-b987, 2022-10-28:
I was doing the install on a RPI Zero 2 W, running Bullseye. Install failed, see below

pi@BarnPi:~/src/8821cu-20210118 $ ./ARM_RPI.sh
Raspberry Pi OS (32 bit) support was turned on in Makefile as planned.
pi@BarnPi:~/src/8821cu-20210118 $ sudo ./install-driver.sh
Running install-driver.sh version 20221018
5.15.61-v7+
armv7l
Installing rtw88_8821cu.conf to: /etc/modprobe.d
Installing 8821cu.conf to: /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to: /usr/src/rtl8821cu-5.12.0

Creating symlink /var/lib/dkms/rtl8821cu/5.12.0/source ->
                 /usr/src/rtl8821cu-5.12.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.15.61-v7+ KSRC=/lib/modules/5.15.61-v7+/build.....................................................................................................(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.61-v7+ (armv7l)
Consult /var/lib/dkms/rtl8821cu/5.12.0/build/make.log for more information.
An error occurred. dkms build error = 10
Please report this error.
Please copy all screen output and paste it into the report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

Any Help would be greatly appreciated.
Thanks
Tom B

Tom-b987, 2022-10-28:
Additional info:

pi@BarnPi:~/src/8821cu-20210118 $ sudo uname -mr; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev; iw reg get
5.15.61-v7+ armv7l
bash: mokutil: command not found
Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 003: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 002: ID 0a05:7211 Unknown Manufacturer hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
rtl8821cu, 5.12.0: added
phy#0
	Unnamed/non-netdev interface
		wdev 0x2
		addr e6:5f:01:50:c2:ad
		type P2P-device
		txpower 31.00 dBm
	Interface wlan0
		ifindex 2
		wdev 0x1
		addr e4:5f:01:50:c2:ad
		ssid TKAB
		type managed
		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
		txpower 31.00 dBm
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, NO-IR
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0
country 99: DFS-UNSET
	(2402 - 2482 @ 40), (6, 20), (N/A)
	(2474 - 2494 @ 20), (6, 20), (N/A)
	(5140 - 5360 @ 160), (6, 20), (N/A)
	(5460 - 5860 @ 160), (6, 20), (N/A)

Tom-b987, 2022-10-28:
Make.Log:

DKMS make.log for rtl8821cu-5.12.0 for kernel 5.15.61-v7+ (armv7l)
Fri 28 Oct 2022 07:43:59 AM EDT
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.15.61-v7+/build M=/var/lib/dkms/rtl8821cu/5.12.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.61-v7+'
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_io.o
gcc: fatal error: Killed signal terminated program cc1
compilation terminated.
make[2]: *** [scripts/Makefile.build:289: /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_debug.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1898: /var/lib/dkms/rtl8821cu/5.12.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.61-v7+'
make: *** [Makefile:2502: modules] Error 2

morrownr, 2022-10-28:
Hi @Tom-b987 

> I was doing the install on a RPI Zero 2 W running Bullseye. Install failed, 

When you tell me you are running Bullseye, that tells me you are running Debian Bullseye. I am a little skeptical so please confirm. Debian Bullseye and Raspberry Pi OS have similarities but for the purposes here, they are NOT the same. I also need to know the version or release date. 

> RPI Zero 2 W.

I need to know if your operating system is 32 bit or 64 bit.

Are you aware that you can turn off the internal wifi? I always turn off the internal wifi as it is little more than trouble waiting to happen.

Nick

Tom-b987, 2022-10-29:
pi@BarnPi:~ $ cat /proc/version
Linux version 5.15.61-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1579 SMP Fri Aug 26 11:10:59 BST 2022

This was downloaded from the official RPI site.

32 bit.

I plan to disable the built in wifi once the USB Wifi gets installed, but I needed the built-in to download the drivers and such.

Hope that helps,
Tom B

Tom-b987, 2022-10-31:
Just tried it with a fresh install from official PI site. Nothing else installed on the RPI Zero 2 W. Got the same error as before.
Any help would be appreciated.

Tom B

morrownr, 2022-10-31:
Tom,

I'm working on this driver today anyway so let me see if I can duplicate this. Confirm for me that the following is the distro you are using:

Raspberry Pi OS with desktop
https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-32-bit

I don't have a RPi Zero 2 W but this may be something that will show up with a RasPi3B or 4B as long as I have the right distro burned.

Tom-b987, 2022-10-31:
Yes, that is the distro I loaded.

morrownr, 2022-10-31:
Tom,

I'm running into an issue and could use your help.

Can I get you to go into the driver directory, run the following and post the results of `make`:

$ sudo ./remove-driver.sh
$ make clean
$ make

There seems to be an issue with the patches I applied long ago with kernel 5.15 so I need to see what you are seeing.

Tom-b987, 2022-10-31:
I didn't include the uninstall output, because it asked for a reboot and I did it without copying the output. Sorry, if you need that too, I can do it all over again.

pi@raspberrypi:~/src/8821cu-20210118 $ make clean
#make -C /lib/modules/5.15.74-v7+/build M=/home/pi/src/8821cu-20210118 clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
pi@raspberrypi:~/src/8821cu-20210118 $ make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.15.74-v7+/build M=/home/pi/src/8821cu-20210118  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.74-v7+'
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_cmd.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_security.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_debug.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_io.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_ioctl_query.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_ioctl_set.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_ieee80211.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_mlme.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_mlme_ext.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_mi.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_wlan_util.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_vht.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_pwrctrl.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_rf.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_chplan.o
  CC [M]  /home/pi/src/8821cu-20210118/core/monitor/rtw_radiotap.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_recv.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_sta_mgt.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_ap.o
  CC [M]  /home/pi/src/8821cu-20210118/core/wds/rtw_wds.o
  CC [M]  /home/pi/src/8821cu-20210118/core/mesh/rtw_mesh.o
  CC [M]  /home/pi/src/8821cu-20210118/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/pi/src/8821cu-20210118/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_xmit.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_p2p.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_rson.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_tdls.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_br_ext.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_iol.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_sreset.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_btcoex_wifionly.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_btcoex.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_beamforming.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_odm.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_rm.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_rm_fsm.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_ft.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_wnm.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_mbo.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_rm_util.o
  CC [M]  /home/pi/src/8821cu-20210118/core/efuse/rtw_efuse.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_roch.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/aes-internal.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/aes-internal-enc.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/aes-gcm.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/aes-ccm.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/aes-omac1.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/ccmp.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/gcmp.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/aes-siv.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/aes-ctr.o
[... 106 lines cut ...]
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/halrf/halrf_debug.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/halrf/halphyrf_ce.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/halrf/halrf_powertracking_ce.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/halrf/halrf_powertracking.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/halrf/halrf_psd.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/rtl8821c/halhwimg8821c_bb.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/rtl8821c/halhwimg8821c_mac.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/rtl8821c/phydm_hal_api8821c.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/rtl8821c/phydm_regconfig8821c.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/rtl8821c/phydm_rtl8821c.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/halrf/rtl8821c/halhwimg8821c_rf.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/halrf/rtl8821c/halrf_8821c.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/halrf/rtl8821c/halrf_iqk_8821c.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/btc/halbtc8821cwifionly.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/btc/halbtc8821c1ant.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/btc/halbtc8821c2ant.o
  CC [M]  /home/pi/src/8821cu-20210118/platform/platform_ops.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_mp.o
  LD [M]  /home/pi/src/8821cu-20210118/8821cu.o
  MODPOST /home/pi/src/8821cu-20210118/Module.symvers
  CC [M]  /home/pi/src/8821cu-20210118/8821cu.mod.o
  LD [M]  /home/pi/src/8821cu-20210118/8821cu.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.15.74-v7+'
pi@raspberrypi:~/src/8821cu-20210118 $

morrownr, 2022-11-01:
Well, that is a clean compile. Run:

$ sudo make install
$ make clean
$ sudo reboot

...and see if the adapter works.

From where I am sitting, it is not clear what happened when dkms was running the show.

Tom-b987, 2022-11-01:
You can make this as solved, at least for the manual build process. You may still want to figure out why the build with dkms was throwing the error. But my install is up and running.

Thanks so much, really appreciate the time and effort to get this fixed.

Tom B

morrownr, 2022-11-01:
Well, you are welcome but all we did is find a workaround. I do my best to make sure I keep compatibility with the old manual installation just in case and this seems to be one of those cases. I'll keep this mind as I continue to work on the driver.

Nick