morrownr/8821cu-20210118 #61: Kali 5.19.0 is not booting after shutdown and start again.

KNIGHTFALL-CS, 2022-10-29:
Issue:

Wifi driver works fine with rtl8811cu,
after installation i rebooted my system everything works fine until i shutdown my pc and start again,
the problem is after selecting kali from grub kali is not booting up isntead there is a black screen with blinking curser but when i remove the usb adapter it boots up. Everytime i need to do this when booting.

OS & device info:

5.19.0-kali2-amd64 x86_64
SecureBoot disabled
Platform is in Setup Mode
Bus 002 Device 006: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 30fa:2350 INSTANT USB Keyboard
Bus 001 Device 003: ID 18f8:0fc0 [Maxxter] USB GAMING MOUSE 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
rtl8821cu/5.12.0, 5.19.0-kali2-amd64, x86_64: installed
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 9c:88:2b:00:33:8c
                ssid HUAWEI Y9s
                type managed
                channel 6 (2437 MHz), width: 40 MHz, center1: 2427 MHz
                txpower 19.00 dBm

morrownr, 2022-10-29:
Hi @KNIGHTFALL-CS 

Interesting. Sometimes s**t happens. I've seen a lot of different things.

Have you tried using different usb ports?

I usually keep an installation of Kali for testing reasons but the last release would not install so I'm a little blind on Kali right now. Kali is not the most stable distro. Is your Kali installation updated?

It would be nice to see some system log output. Do you know how to turn logging on with this driver? However, if you can't get it to boot, getting logs can be a PITA.

Nick

KNIGHTFALL-CS, 2022-10-29:
> 

Hello Nick, Thanks for your reply.

Yes i tried different ports but didn't work and my kali is in latest update.

I've tried Debug mod to login and this ia what the last line shows:
[x] reset high-speed USB device number 3 using echci-pci
It could not be a hardware issue bcz the device works totally fine with windows.

I think i should change my linux distro bcz, as you said kali became more and more unstable these days.

morrownr, 2022-10-29:
Checking out a different distro might be the best option. You can always do a multi-boot and you do not have to use Kali to do security analysis. I test the various apps on Xubuntu and they work fine, you just have install the ones you want.

Right now I am not much help since Kali won't even install. This driver is working well on the other distros that I test all the way to and including kernel 6.1.

KNIGHTFALL-CS, 2022-10-30:
sure i'll check that soon.

knightfall-cs, 2022-11-30:
Re-checked;

knightfall-cs, 2022-11-30:
Only problem is the adapters driver inbuild rom. 
If adapter is plugged in when pc is booting its causing an error with ehci driver, Other than this driver is working totally fine with Kali Linux and Parrot OS.

knightfall-cs, 2022-11-30:
I Wonder how windows be able to

knightfall-cs, 2022-11-30:
I am thinking how windows be able to automatically reset these type of devices, it could be very nice if linux contains the same settings.

morrownr, 2022-11-30:
Hey @knightfall-cs 

I was finally able to install Kali again. I had to use a weekly. It is running kernel 6.0.

However, I have an idea. Myself and a few others are currently working on a new version of this driver as new source was made available. It has several fixes. Would you be interested in testing? The repo is currently private for testing so I would have to add you and you would need to sign in to see it. Interested? We can use another tester.

knightfall-cs, 2022-12-01:
> Hey @knightfall-cs
>
> I was finally able to install Kali again. I had to use a weekly. It is running kernel 6.0.
>
> However, I have an idea. Myself and a few others are currently working on a new [...]

Of course, It was really interesting!

knightfall-cs, 2022-12-01:
Hello @morrownr
Of Course i am really interested!

morrownr, 2022-12-01:
Done. You should get a message.

knightfall-cs, 2022-12-01:
> I am thinking how windows be able to automatically reset these type of devices, it could be very nice if linux contains the same settings.

Do you have any idea about this?

morrownr, 2022-12-01:
> Do you have any idea about this?

No. Another mystery to solve. May not be Linux. Could be the driver.

closed 2022-12-08 by knightfall-cs

morrownr, 2022-12-08:
What did you find out?

knightfall-cs, 2022-12-08:
Nothing new but, what  I found is nothing wrong with our driver even after uninstalling the driver same problem occurs.
Wifi dongle I was using was a USB device with inbuilt ROM for (windows) driver software to automatically install driver if not installed the problem is when the device is connected when booting inbuilt Rom causes problem with ehci-pci driver unfortunately Linux can't be able to automatically reset the device so that NIC in wifi dongle can be revoked by our driver.

I found some articles about this issue, but dint tried them yet.

http://billauer.co.il/blog/2013/02/usb-reset-ehci-uhci-linux/
https://unix.stackexchange.com/questions/704341/how-to-reset-usb-controllers

knightfall-cs, 2022-12-08:
> What did you find out?

Nothing new but, what I found is nothing wrong with our driver even after uninstalling the driver same problem occurs.
Wifi dongle I was using was a USB device with inbuilt ROM for (windows) driver software to automatically install driver if not installed the problem is when the device is connected when booting inbuilt Rom causes problem with ehci-pci driver unfortunately Linux can't be able to automatically reset the device so that NIC in wifi dongle can be revoked by our driver.

I found some articles about this issue, but dint tried them yet.

http://billauer.co.il/blog/2013/02/usb-reset-ehci-uhci-linux/
https://unix.stackexchange.com/questions/704341/how-to-reset-usb-controllers

knightfall-cs, 2022-12-10:
> Oh, you are talking about the adapter being a multi-state adapter. I did not understand what you were describing.
> 
> Go to the site Main Menu:
> 
> https://github.com/morrownr/USB-WiFi
> 
> And select menu item 7, How to Modeswitch.
> 
> Let me know if you have any trouble and if you could suggest a good place to put information about this in the README for the new driver, that would be great.
> 
> Nick

Is there anyway to run modeswitch when booting up.

morrownr, 2022-12-11:
> Is there anyway to run modeswitch when booting up?

Every modern distros I have checked runs it on boot via systemd.

knightfall-cs, 2022-12-13:
@morrownr 
Sorry for late reply!

[image]

This is the Vender and Product ID of the device from windows.
So, How can I configure modeswitch for linux?

morrownr, 2022-12-13:
> It's 0bda:c811 in Linux, the same.

That is the wifi id. It would be different if usb_modeswitch was not working,

Does you system bios and/or Windows have Fastboot active? I have heard of users having problems somewhat similar to this if Fast boot is active. The fix is to turn it off in the bios and/or Windows.

With that said, the disclaimer is that I know very little more than nothing about Windows as I have never been a Windows user.

knightfall-cs, 2022-12-14:
Fast boot is turned off already and the only problem I'm facing with some adapters in linux pc is not booting when adapters are connected to the usb port but when they plugged in after the bootup they are working totally fine.
I don't know what kind of issue is this maybe it's all about ehci-pci driver as I mentioned before.
> https://github.com/morrownr/8821cu-20210118/issues/61#issuecomment-1295944742