morrownr/8821cu-20210118 #62: Testing Debian 11 (i686)?

Jibun-no-Kage, 2022-10-30:
\# uname -mr; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev; iw reg get
5.10.0-19-686-pae i686
EFI variables are not supported on this system
Bus 003 Device 003: ID 09cc:2106 Workbit Corp. 4HDD
Bus 003 Device 002: ID 09cc:2106 Workbit Corp. 4HDD
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 002: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
rtl8821cu, 5.12.0, 5.10.0-19-686-pae, i686: installed
phy#0
        Interface wlx00e032816834
                ifindex 5
                wdev 0x1
                addr 00:e0:32:81:68:34
                type managed
                txpower -100.00 dBm
global
country 00: DFS-UNSET
        (755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

This is on an older HP Compaq Evo SFF desktop.

ifconfig $(ip link | grep wlx | cut -d ':' -f2 | tr -d ' ') up

# ifconfig $(ip link | grep wlx | cut -d ':' -f2 | tr -d ' ')
wlx00e032816834: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::2e0:32ff:fe81:6834  prefixlen 64  scopeid 0x20<link>
        ether 00:e0:32:81:68:34  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 8 overruns 0  carrier 0  collisions 0

Had to force association...
iwconfig $(ip link | grep wlx | cut -d ':' -f2 | tr -d ' ')
wlx00e032816834  unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

wpa_supplicant -B -D wext -i $(ip link | grep wlx | cut -d ':' -f2 | tr -d ' ') -c /etc/wpa_supplicant/wpa_supplicant.conf
Successfully initialized wpa_supplicant
ioctl[SIOCSIWAP]: Operation not permitted

Which generated the following error...
[  275.211165] ------------[ cut here ]------------
[  275.211266] WARNING: CPU: 0 PID: 130 at net/wireless/sme.c:757 __cfg80211_connect_result+0x398/0x400 [cfg80211]
[  275.211269] Modules linked in: 8021q garp stp mrp llc ppdev evdev serio_raw pcspkr 8821cu(OE) i915 snd_intel8x0 snd_ac97_codec ac97_bus cfg80211 snd_pcm snd_timer video iTCO_wdt snd sg intel_pmc_bxt soundcore drm_kms_helper iTCO_vendor_support watchdog cec rfkill parport_pc parport i2c_algo_bit rng_core button nfsd auth_rpcgss drm nfs_acl bonding lockd grace sunrpc fuse configfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c crc32c_generic raid0 multipath linear raid1 md_mod dm_mod uas usb_storage sd_mod sr_mod t10_pi crc_t10dif cdrom crct10dif_generic crct10dif_common ata_generic ata_piix psmouse i2c_i801 libata i2c_smbus scsi_mod uhci_hcd ehci_pci ehci_hcd usbcore lpc_ich e100 e1000 mii usb_common floppy
[  275.211402] CPU: 0 PID: 130 Comm: kworker/u4:4 Tainted: G           OE     5.10.0-19-686-pae #1 Debian 5.10.149-2
[  275.211404] Hardware name: Compaq Evo D510 SFF/07E4h, BIOS 686O2 v3.18 05/28/2003
[  275.211425] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[  275.211446] EIP: __cfg80211_connect_result+0x398/0x400 [cfg80211]
[  275.211451] Code: 84 81 fd ff ff 3e ff 40 ec e9 78 fd ff ff 8d 74 26 00 0f 0b 8b 56 08 e8 36 64 fd ff e9 2f fe ff ff 90 0f 0b 8d b6 00 00 00 00 <0f> 0b e9 1f fe ff ff 90 e8 9b 59 ec da e9 14 fe ff ff 8d b6 00 00
[  275.211454] EAX: 00000000 EBX: f5556400 ECX: c2ebdc44 EDX: 00000000
[  275.211457] ESI: c2ebdc0c EDI: 00000000 EBP: f5fc9f08 ESP: f5fc9ee4
[  275.211460] DS: 007b ES: 007b FS: 00d8 GS: 00e0 SS: 0068 EFLAGS: 00010246
[  275.211462] CR0: 80050033 CR2: 0048d8c0 CR3: 02eba000 CR4: 000006f0
[  275.211465] Call Trace:
[  275.211491]  cfg80211_process_wdev_events+0x12d/0x1a0 [cfg80211]
[  275.211510]  cfg80211_process_rdev_events+0x37/0x80 [cfg80211]
[  275.211526]  cfg80211_event_work+0x1b/0x30 [cfg80211]
[  275.211535]  process_one_work+0x16c/0x2e0
[  275.211540]  worker_thread+0x15e/0x3e0
[  275.211545]  kthread+0xf6/0x110
[  275.211549]  ? process_one_work+0x2e0/0x2e0
[  275.211552]  ? kthread_associate_blkcg+0xb0/0xb0
[  275.211558]  ret_from_fork+0x1c/0x28
[  275.211562] ---[ end trace 1afe89774f54087f ]---

Odd thing is if I call dhcpcd, I get an IP address.
wlx00e032816834: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.84  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::2e0:32ff:fe81:6834  prefixlen 64  scopeid 0x20<link>
        ether 00:e0:32:81:68:34  txqueuelen 1000  (Ethernet)
        RX packets 575  bytes 111921 (109.2 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 19  bytes 2238 (2.1 KiB)
        TX errors 0  dropped 11 overruns 0  carrier 0  collisions 0

But clearly the driver is not stable?  Or the network stack, repeated pings to the assigned IP address, randomly work.

Thought you might find it interesting.

morrownr, 2022-10-30:
I have an old 32 bit laptop that I try to keep going. I have good luck with:

https://www.raspberrypi.com/software/raspberry-pi-desktop/

Jibun-no-Kage, 2022-10-31:
Yeah, I have been pretty lucky with this old HP Compaq Evo.  Does not really need WiFi, but since I was going to test, figured I would try it.  I tried 2 or 3 other older drivers I found on github they all resulted in the same 'cfg80211' comparable issue.  I am quite familiar with Pi OS, I will give that a try, just to see the results.

closed 2022-10-31 by Jibun-no-Kage