morrownr/8821cu-20210118 #63: Quirk?  Had to reboot just before running ./install_driver.sh on fresh install of Pi OS 11?

Jibun-no-Kage, 2022-10-31:
Quirk?  Had to reboot just before running ./install_driver.sh on fresh install of Pi OS 11?

Running as root, did all steps for install on 32 bit Pi OS 11 (most recent image as of today), Set concurrent mode off, Set 32 bit architecture.

Executed ./install_driver.sh, reported kernel headers not found.  But explicitly did same as per documented steps.
Ran ./remove_driver.sh
Reboot
Ran ./install_driver.sh... worked this time after reboot, odd.

# dkms status
rtl8821cu, 5.12.0, 5.15.74-v7l+, armv7l: installed

==========================

Here is the error report...

./install-driver.sh
Running install-driver.sh version 20221018
5.15.61-v7l+
armv7l
Installing rtw88_8821cu.conf to: /etc/modprobe.d
Installing 8821cu.conf to: /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to: /usr/src/rtl8821cu-5.12.0

Creating symlink /var/lib/dkms/rtl8821cu/5.12.0/source ->
                 /usr/src/rtl8821cu-5.12.0

DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.15.61-v7l+ cannot be found at
/lib/modules/5.15.61-v7l+/build or /lib/modules/5.15.61-v7l+/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
An error occurred. dkms build error = 1
Please report this error.
Please copy all screen output and paste it into the report.
Run the following before reattempting installation.

morrownr, 2022-10-31:
Hi @Jibun-no-Kage 

I've been seeing this a lot lately. I think the most effective work around is:

$ sudo apt purge raspberrypi-kernel-headers
$ sudo reboot
$ sudo apt install -y raspberrypi-kernel-headers build-essential bc dkms git

I have not ran into the problem here so I am just going off of what I have read. Your mileage may vary.

Jibun-no-Kage, 2022-11-01:
Yeah, happened on 32 bit and 64 bit ARM.  The reboot before install-driver.sh seems to work consistently.  Have tested with the current commit/clone.  Now trying it on i686/amd64 just curious if it happens on these as well.

morrownr, 2022-11-01:
Let me know if you get this sorted out. I can't see where it is a problem with the driver unless I'm missing something. It seems to be a problem specific to the RasPiOS.

Jibun-no-Kage, 2022-11-02:
I will do a clean test again tomorrow or next day at the latest.  Last testing was on clean systems but what to see if I can consistently create the issue.

You the old adage...

Once is an anomaly, Twice is a pattern, and Thrice is a problem!