morrownr/8821cu-20210118 #64: Module installed but cannot find any networks and shows as unavailable in gnome settings

BurritoJohn, 2022-11-02:
As said in the title, the module stopped working after upgrading to Ubuntu 22.10. Updating the driver led to a mountable module, but my network card shows as unavailable in the settings and seems to search forever for networks without finding anything.

Using Ubuntu 22.10 with baseline kernel, here is what I get with `sudo uname -mr; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev; iw reg get `: 

```
5.19.0-23-generic x86_64
SecureBoot disabled
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 003 Device 002: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
rtl8821cu/5.12.0, 5.19.0-23-generic, x86_64: installed
phy#0
	Interface wlx1cbfceef4c82
		ifindex 4
		wdev 0x1
		addr 1c:bf:ce:ef:4c:82
		type managed
		txpower -100.00 dBm
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

```

Using `lsmod`, I can see : 
```
8821cu               2351104  0
cfg80211             1040384  1 8821cu
```

Thanks !

morrownr, 2022-11-02:
Hi @BurritoJohn 

We may not get this the first but let's give it a go.

> country 00: DFS-UNSET

This is not good. Raspberry Pi OS is the only distro I know of that makes the user set the country code as part of an initial configuration so hats off to them. This shuts down 5 GHz. Try:

$ sudo iw reg set US
$ iw reg get

The first line sets the country code and you should change US to whatever your country code is. The second line lets you know if it worked.

> upgrading to Ubuntu 22.10

Just to see if people were reading the README, I have some fine print that says you should remove this driver before upgrading the version of the distro. Things don't usually work out if this is not done. Try this from the driver directory:

$ git pull
$ sudo ./remove-driver.sh
if you were going to upgrade the distro, do it here but since you have already done it, not necessay
$ sudo install-driver.sh
after setting the appropriate driver options and saving, reboot

Let me know how it goes.

BurritoJohn, 2022-11-03:
Hi, many thanks for the help !
Your proposition did not solve my issue but led to another solution. Now my country code is correct and wireless connection working.
It appears that the issue is with `iwd` missing after the upgrade. For those experiencing the same issue, try this : 
`sudo apt install iwd; sudo systemctl enable iwd.service; systemctl start iwd.service`

morrownr, 2022-11-03:
> It appears that the issue is with iwd missing after the upgrade. For those experiencing the same issue, try this :
sudo apt install iwd; sudo systemctl enable iwd.service; systemctl start iwd.service

I wonder how that happened. I knew Buntu was going from wpa_supplicant to iwd but would have not thought to look at this. Have you consider filing a bug with Ubunru?