morrownr/8821cu-20210118 #65: Program doesn't work for pi zero 2w

Ripper346, 2022-11-06:
Hi, I am certainly doing something wrong, I have a pi zero 2 w running raspberry pi os lite 32bit. The linux kernel is at version 5.15.74-v7. 
All I have installed before the driver is docker, a part of that it is of default.

I followed the instructions from the readme, it completed them without any warning/error. I am missing how to enable the interface because after the reboot if I run `ip addr` I get
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether  brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.15/24 brd 192.168.1.255 scope global dynamic noprefixroute wlan0
       valid_lft 86373sec preferred_lft 75573sec
    inet6 fe80::ecda:eec:c1ea:f978/64 scope link
       valid_lft forever preferred_lft forever
3: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default
    link/ether  brd ff:ff:ff:ff:ff:ff
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever
4: br-0be208caf6c5: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default
    link/ether  brd ff:ff:ff:ff:ff:ff
    inet 172.18.0.1/16 brd 172.18.255.255 scope global br-0be208caf6c5
       valid_lft forever preferred_lft forever
    inet6 fe80::42:fcff:fe8a:5dfc/64 scope link tentative
       valid_lft forever preferred_lft forever
5: veth5780101@vethbeda945: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default
    link/ether  brd ff:ff:ff:ff:ff:ff
    inet6 fe80::dd14:2bf3:110a:4cd3/64 scope link tentative
       valid_lft forever preferred_lft forever
6: vethbeda945@veth5780101: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-0be208caf6c5 state UP group default
    link/ether  brd ff:ff:ff:ff:ff:ff
    inet6 fe80::b0d1:c488:d8bf:77cd/64 scope link tentative
       valid_lft forever preferred_lft forever
    inet6 fe80::5447:adff:fee1:7910/64 scope link tentative
       valid_lft forever preferred_lft forever
```
Doing `lsusb` this is the result (I have two usb ports out of the keyboard and I connected the BrosTrend AC5L there since I don't have any other direct usb port available):
```
Bus 001 Device 004: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 003: ID 413c:2010 Dell Computer Corp. Keyboard
Bus 001 Device 002: ID 413c:1003 Dell Computer Corp. Keyboard Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
I tried even to disable the integrated wifi card from the `/boot/config.txt` file but all it is done is to remove the wlan0 card and nothing else. Can anyone help me? Thanks.

morrownr, 2022-11-06:
Hi @Ripper346 

So, RasPiOS, 32 bit, kernel 5.15 on Pi Zero 2 w.

I checked the ID and this is the right driver for your usb adapter. Let's try this:

Go into the driver directory and run the following commands:

$ make clean
$ make

Paste the results.

Ripper346, 2022-11-07:
Thank you for the fast reply. The output is
```
#make -C /lib/modules/5.15.74-v7+/build M=/home/pi/src/8821cu-20210118 clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.15.74-v7+/build M=/home/pi/src/8821cu-20210118  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.74-v7+'
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_cmd.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_security.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_debug.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_io.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_ioctl_query.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_ioctl_set.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_ieee80211.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_mlme.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_mlme_ext.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_mi.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_wlan_util.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_vht.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_pwrctrl.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_rf.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_chplan.o
  CC [M]  /home/pi/src/8821cu-20210118/core/monitor/rtw_radiotap.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_recv.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_sta_mgt.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_ap.o
  CC [M]  /home/pi/src/8821cu-20210118/core/wds/rtw_wds.o
  CC [M]  /home/pi/src/8821cu-20210118/core/mesh/rtw_mesh.o
  CC [M]  /home/pi/src/8821cu-20210118/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/pi/src/8821cu-20210118/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_xmit.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_p2p.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_rson.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_tdls.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_br_ext.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_iol.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_sreset.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_btcoex_wifionly.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_btcoex.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_beamforming.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_odm.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_rm.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_rm_fsm.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_ft.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_wnm.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_mbo.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_rm_util.o
  CC [M]  /home/pi/src/8821cu-20210118/core/efuse/rtw_efuse.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_roch.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/aes-internal.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/aes-internal-enc.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/aes-gcm.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/aes-ccm.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/aes-omac1.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/ccmp.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/gcmp.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/aes-siv.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/aes-ctr.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/sha256-internal.o
  CC [M]  /home/pi/src/8821cu-20210118/core/crypto/sha256.o
[... 105 lines cut ...]
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/halrf/halphyrf_ce.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/halrf/halrf_powertracking_ce.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/halrf/halrf_powertracking.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/halrf/halrf_psd.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/rtl8821c/halhwimg8821c_bb.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/rtl8821c/halhwimg8821c_mac.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/rtl8821c/phydm_hal_api8821c.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/rtl8821c/phydm_regconfig8821c.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/rtl8821c/phydm_rtl8821c.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/halrf/rtl8821c/halhwimg8821c_rf.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/halrf/rtl8821c/halrf_8821c.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/phydm/halrf/rtl8821c/halrf_iqk_8821c.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/btc/halbtc8821cwifionly.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/btc/halbtc8821c1ant.o
  CC [M]  /home/pi/src/8821cu-20210118/hal/btc/halbtc8821c2ant.o
  CC [M]  /home/pi/src/8821cu-20210118/platform/platform_ops.o
  CC [M]  /home/pi/src/8821cu-20210118/core/rtw_mp.o
  LD [M]  /home/pi/src/8821cu-20210118/8821cu.o
  MODPOST /home/pi/src/8821cu-20210118/Module.symvers
  CC [M]  /home/pi/src/8821cu-20210118/8821cu.mod.o
  LD [M]  /home/pi/src/8821cu-20210118/8821cu.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.15.74-v7+'
```
It seems to me that it is all correct from the compilation. This evening (for me in 8 hour), I will try also to install the driver to a brand new installation of the OS in another sd card to exclude something wrong made by docker.

morrownr, 2022-11-07:
That compile is about as clean as it gets. 

I've noticed that this driver/chipset combination seems to suffer from some issues that other drivers do not. Sometimes changing or adding certain driver options can help.

Ripper346, 2022-11-09:
I was finally able to run the installation with a clean OS. The result was:
```
Running install-driver.sh version 20221031
Kernel=5.15.76-v7+
Architecture=armv7l
Installing rtw88_8821cu.conf to: /etc/modprobe.d
Installing 8821cu.conf to: /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to: /usr/src/rtl8821cu-5.12.0

Creating symlink /var/lib/dkms/rtl8821cu/5.12.0/source ->
                 /usr/src/rtl8821cu-5.12.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.15.76-v7+ KSRC=/lib/modules/5.15.76-v7+/build.............................................................................................................
cleaning build area...

DKMS: build completed.

8821cu.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.15.76-v7+/kernel/drivers/net/wireless/

depmod........

DKMS: install completed.
The driver was installed successfully.
Do you want to edit the driver options file now? [y/N] n
Do you want to reboot now? (recommended) [y/N] y
```
and it doesn't appear between the network interfaces.

@morrownr what options are you thinking to? Couldn't be that it just needs an activation of the usb dongle? Now I want to try it also on a pi zero w to see if it changes something

Ripper346, 2022-11-09:
Today I have tried to install the driver to a clean OS and nothing it installed it fine but then it didn't display the interface. Then I tried something else, I disconnected the keyboard and connected directly the interface: surprise surprise the network interface appeared. I swap the sd with the previous with docker with the plug directly connected and again it worked. Last try I checked was to use an orthodox usb hub (2.0) and it also worked. Somehow the two ports in the keyboard work because I can see the devices but in fact they don't. Shame on me, thank you for your support @morrownr, I should have checked it in the first place

closed 2022-11-09 by Ripper346

morrownr, 2022-11-10:
Glad you got it going!