morrownr/8821cu-20210118 #66: BananaPi M4 not compiling

eboye, 2022-11-17:
sudo ./install-driver.sh

gives me this error:

```
Running install-driver.sh version 20221101
Kernel=4.9.119-BPI-M4-Kernel
Architecture=aarch64
Installing rtw88_8821cu.conf to: /etc/modprobe.d
Installing 8821cu.conf to: /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to: /usr/src/rtl8821cu-5.12.0

Creating symlink /var/lib/dkms/rtl8821cu/5.12.0/source ->
                 /usr/src/rtl8821cu-5.12.0

DKMS: add completed.
The driver was added successfully.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=4.9.119-BPI-M4-Kernel KSRC=/lib/modules/4.9.119-BPI-M4-Kernel/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.9.119-BPI-M4-Kernel (aarch64)
Consult /var/lib/dkms/rtl8821cu/5.12.0/build/make.log for more information.
An error occurred. dkms build error = 10
```

cat /var/lib/dkms/rtl8821cu/5.12.0/build/make.log

```
DKMS make.log for rtl8821cu-5.12.0 for kernel 4.9.119-BPI-M4-Kernel (aarch64)
Thu 17 Nov 2022 02:09:27 AM CET
make ARCH=aarch64 CROSS_COMPILE= -C /lib/modules/4.9.119-BPI-M4-Kernel/build M=/var/lib/dkms/rtl8821cu/5.12.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.9.119-BPI-M4-Kernel'
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_debug.o
cc1: sorry, unimplemented: -mfloat-abi=hard and VFP
make[2]: *** [scripts/Makefile.build:334: /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_cmd.o] Error 1
make[2]: *** Waiting for unfinished jobs....
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_io.o
cc1: sorry, unimplemented: -mfloat-abi=hard and VFP
make[2]: *** [scripts/Makefile.build:334: /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_security.o] Error 1
cc1: sorry, unimplemented: -mfloat-abi=hard and VFP
make[2]: *** [scripts/Makefile.build:334: /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_debug.o] Error 1
cc1: sorry, unimplemented: -mfloat-abi=hard and VFP
make[2]: *** [scripts/Makefile.build:334: /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_io.o] Error 1
make[1]: *** [Makefile:1613: _module_/var/lib/dkms/rtl8821cu/5.12.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.9.119-BPI-M4-Kernel'
```

sudo uname -mr; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev; iw reg get

```
4.9.119-BPI-M4-Kernel aarch64
EFI variables are not supported on this system
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 003: ID 046d:c31f Logitech, Inc. Comfort Keyboard K290
Bus 001 Device 002: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: bluetooth: Bluetooth
	Soft blocked: yes
	Hard blocked: no
rtl8821cu, 5.12.0: added
global
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, NO-IR
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, NO-IR
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, NO-IR
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, NO-IR
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, NO-IR
	(5735 - 5835 @ 80), (N/A, 20), (N/A), NO-IR
```

The problem I'm facing is that original kernel was not allowing me to use 2GB of RAM, but only 1GB. So I compiled the kernel using this source: https://github.com/BPI-SINOVOIP/BPI-M4-bsp

But then WiFi and Bluetooth are not working.

Can anybody help?

morrownr, 2022-11-17:
Hi @eboye 

> Can anybody help?

I can try to help but I do not have a BananaPi. I do have two RasPis and regularly test this and the other drivers on them. While I can suggest, it will be up to you to do the work. With that said, I only test and support kernels 4.19 and later. Especially with ARM, there have been a lot of changes over the years so this could be easy or very hard to figure out.

> cc1: sorry, unimplemented: -mfloat-abi=hard and VFP

The above is they key problem. Recommend you do a search to see what pops up and get back to me.

morrownr, 2022-11-17:
I happened to do a little looking at the above issue while working on something else. That is an arm specific issue due to the fact arm originally did not have what we used to call a math coprocessor. It had to be emulated in software. The compiler you are using with that very old distro does not have that capability. Your best option is likely to find a much more modern distro.

There are adapters that are still for sale that in-kernel (plug and play) drivers that work on kernels that old but it would be single band, WiFi 4 only.

https://github.com/morrownr/USB-WiFi

Look at Main Menu item 2. Scroll down to the WiFi 4 adapters.

eboye, 2022-11-18:
Thanks for looking up, you helped for sure,

I do have external adapter, but I was hopping on using the internal one. I'll see if I can compile the kernel with those modules but still keeping the 2GB flag.

Also I'll try to build newer kernel. I haven't done it in years, I do hope I'll be successful :P

The main problem is that the board is not supported since 2019 and is stuck on old, very old 4.x kernel.

morrownr, 2022-11-18:
I have a lot of what many people might consider old hardware. My main dev box here is 10+ years old but I think it is quite snappy. It has an Intel i7 processor and 12 gb or RAM plus fairly fast SSD. I have other hardware that is 15+ old years and works just fine. I even have an old IBM PS/VP from the early nineties that runs fine. I don't do any work with it these days but it turns on and is fun to play with at times. Linus got my attention the other day when he said that it is about time to remove 486 support.

I'd offer more help but I don't know anything about BananaPi hardware so you may need to do some digging and make this a longer term project. Good luck.

eboye, 2022-11-18:
Thanks,

Similar story here. I refuse to throw away "old" hardware. I just make it do something else. I really hope that I'll be able to compile newer kernel on it.

I might try Arch on it as aur has your firmware packages there. But first I need to figure out the kernel compilation stuff.