morrownr/8821cu-20210118 #67: Doesn't compile (unreachable instruction)

Pickelgold, 2022-11-17:
Okay, first of all an information dump:

Output of `sudo uname -mr; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev; iw reg get`:

`6.0.8-arch1-1 x86_64
mokutil: error while loading shared libraries: libcrypto.so.3: cannot open shared object file: No such file or directory
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 0bda:c820 Realtek Semiconductor Corp. 802.11ac NIC
Bus 003 Device 004: ID 2808:9338 Focal-systems.Corp FT9201Fingerprint.̚
Bus 003 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0edc:5830 HRY USB Camera
Bus 001 Device 005: ID 05ac:12a8 Apple, Inc. iPhone 5/5C/5S/6/SE
Bus 001 Device 003: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 002: ID 058f:6387 Alcor Micro Corp. Flash Drive
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
rtl8821cu/5.12.0: added
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)`

Contents of `make.log` (the interesting part):

`  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/core/rtw_mp.o
  LD [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/8821cu.o
/var/lib/dkms/rtl8821cu/5.12.0/build/8821cu.o: warning: objtool: mp_xmit_packet_thread+0x9e: unreachable instruction
  MODPOST /var/lib/dkms/rtl8821cu/5.12.0/build/Module.symvers
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/8821cu.mod.o
  LD [M]  /var/lib/dkms/rtl8821cu/5.12.0/build/8821cu.ko
  BTF [M] /var/lib/dkms/rtl8821cu/5.12.0/build/8821cu.ko
Failed to parse base BTF 'vmlinux': -22
make[3]: *** [scripts/Makefile.modfinal:60: /var/lib/dkms/rtl8821cu/5.12.0/build/8821cu.ko] Error 1
make[3]: *** Deleting file '/var/lib/dkms/rtl8821cu/5.12.0/build/8821cu.ko'
make[2]: *** [scripts/Makefile.modpost:134: __modpost] Error 2
make[1]: *** [Makefile:1766: modules] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.0.8-arch1-1/build'
make: *** [Makefile:2503: modules] Error 2
`

Full log: 
[make.log](https://github.com/morrownr/8821cu-20210118/files/10027256/make.log)

OS: EndeavourOS
Host: [Gateway GWNR71517](https://gatewayusa.com/GWNR71517.html)
Wifi (according to Gateway): Realtek AB b/g/n /RTL8821CU
Current I have my iPhone usb tethered to provide internet.

It's interesting that wifi doesn't even show up in rfkill or ifconfig, so I'm thinking that this may or may not even be a problem with this driver. Wifi works fine in windows however.

If any other information is needed I will be happy to provide.

morrownr, 2022-11-17:
I reformated your original msg so that we can read it. This markup or markdown stuff can present a challenge. You can look and see what I did by editing.

> ID 0bda:c820 Realtek Semiconductor Corp. 802.11ac NIC

This driver should work... in theory.

> Failed to parse base BTF 'vmlinux': -22

That is probably a hint as to what the problem is but it is a new one to me.

Tell me what kernel version you are running:

$ uname -r

Tell me what you ran for installation Step 3 since your OS is not listed?

Is your OS based on another distro such as Debian, Ubuntu, Arch, Fedora?

Pickelgold, 2022-11-17:
Kernel is 6.0.8-arch1-1

OS is EndeavourOS, which is basically Arch but with a lot of things preinstalled. All the relevant packages are installed and up-to-date.

morrownr, 2022-11-17:
Okay so amd64? Right?

Is this what you ran for installation Step 3:

$ sudo pacman -S --noconfirm linux-headers dkms git bc

The driver appeared to compile clean and kicked out at the last moment. Something seems to be slightly off in your dev environment. That is why I am asking what you installed with Step 3.

Pickelgold, 2022-11-17:
Yes, amd64, and yes, that is the exact command I ran. In fact, I just reinstalled those packages as well as the kernel to make sure. I will try compiling on a fresh vanilla arch installation and seeing what happens.

morrownr, 2022-11-17:
>  I will try compiling on a fresh vanilla arch installation and seeing what happens.

If it works on a fresh Arch install, then it is time to bring the issue to the attention of the EndeavourOS devs. It is very very common for distros, especially the ones with limited devs, to mess the dev environment up.

Pickelgold, 2022-11-18:
Good news, the driver works perfectly on my Arch install, which is great since I've always planned on using Arch anyways. I'll bring up the issue with the EndeavourOS community but I doubt much will come of it.

closed 2022-11-18 by Pickelgold

morrownr, 2022-11-18:
Good to hear. There is another issue but if you are on Arch now, it may not matter:

> country 00: DFS-UNSET

Your country code is not set. That can cause problems with 5 Ghz.

Ex.

$ iw reg get

$ iw reg set US