morrownr/8821cu-20210118 #69: Works Like a Charm on PopOS

greenm01, 2022-11-19:
Thanks a ton for making this driver available. My cheap little realtek USB wifi adapter works like a charm.

I'd buy you a coffee if I could.

Rock on!

morrownr, 2022-11-21:
Hi @greenm01 

> I'd buy you a coffee if I could.

Better than that. Myself and another guy are working on a new version of this driver based on updated source code. Right now the repo is private. Would you like to help test?

At some point in the next month or two, it should be made public if all goes well.

Nick

greenm01, 2022-11-22:
Yes, would be happy to help!

morrownr, 2022-11-25:
@greenm01 

Sorry for the delay. The holidays have kept me busy.

I will add you to the private repo. Right now it is only myself and one other person. We have been working. We are using the new driver code as a testbed for new features in the installationa and removal scripts as well as changes in the README so post an issue saying that you are there when you get to the new repo and we will take it from there.

Nick

tamojitray, 2022-11-28:
@morrownr 
>Would you like to help test?

I can also test if you need more tester