morrownr/8821cu-20210118 #70: Fail to build on Kali kernel 6.0.0

lemonade7296, 2022-11-21:
The output information:
Running install-driver.sh version 20221101
Kernel=6.0.0-kali3-amd64
Architecture=x86_64
Installing rtw88_8821cu.conf to: /etc/modprobe.d
Installing 8821cu.conf to: /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to: /usr/src/rtl8821cu-5.12.0
Creating symlink /var/lib/dkms/rtl8821cu/5.12.0/source -> /usr/src/rtl8821cu-5.12.0
The driver was added successfully.
Sign command: /usr/lib/linux-kbuild-6.0/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Error! Your kernel headers for kernel 6.0.0-kali3-amd64 cannot be found at /lib/modules/6.0.0-kali3-amd64/build or /lib/modules/6.0.0-kali3-amd64/source.
Please install the linux-headers-6.0.0-kali3-amd64 package or use the --kernelsourcedir option to tell DKMS where it's located.
An error occurred. dkms build error = 1
Please report this error.
Please copy all screen output and paste it into the report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

I don't know how to fix it. Any suggestions? Thanks a lot.

morrownr, 2022-11-21:
Hi @lemonade7296 

> Error! Your kernel headers for kernel 6.0.0-kali3-amd64 cannot be found

Your kernel headers are not installed.

> Please install the linux-headers-6.0.0-kali3-amd64 package

Try:

$ sudo apt install linux-headers-6.0.0-kali3-amd64

Side note: Which option did you use for installation Step 3?

Back to subject: Once the header files are installed, do a cleanup and try to install again:

$ sudo ./remove-driver.sh
$ sudo ./install-driver,sh

lemonade7296, 2022-11-22:
Ohhhhh!!!! Thank you @morrownr
I finally succeed to install the driver. The comment you leave is really helpful.
According to your advise, I do:
$ sudo apt-get dist-upgrade
$ sudo apt-get install linux-headers-6.0.0-kali3-amd64

For installation Step 3, I used the option for Debian, Kali, and Raspberry Pi Desktop (x86):
$ sudo apt install -y linux-headers-$(uname -r) build-essential bc dkms git libelf-dev

However, I missed the warning info about failure installazation of linux-header.
Therefore, I upgrade my kali this time.

Thanks again.

closed 2022-11-22 by lemonade7296