morrownr/8821cu-20210118 #71: fails on latest 5.15.76-v7+ (raspian)

ashah7, 2022-11-23:
Fails on the latest Nov release 
tried buidling from src as well same result

Linux <serial> 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l GNU/Linux
```

pi@<serial>:~/8821cu-20210118 $ sudo ./install-driver.sh
Running install-driver.sh version 20221101
Kernel=5.15.76-v7+
Architecture=armv7l
Installing rtw88_8821cu.conf to: /etc/modprobe.d
Installing 8821cu.conf to: /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to: /usr/src/rtl8821cu-5.12.0

Creating symlink /var/lib/dkms/rtl8821cu/5.12.0/source ->
                 /usr/src/rtl8821cu-5.12.0

DKMS: add completed.
The driver was added successfully.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.15.76-v7+ KSRC=/lib/modules/5.15.76-v7+/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.76-v7+ (armv7l)
Consult /var/lib/dkms/rtl8821cu/5.12.0/build/make.log for more information.
An error occurred. dkms build error = 10
Please report this error.
Please copy all screen output and paste it into the report.
Run the following before reattempting installation.

$ sudo ./remove-driver.sh
pi@<serial>:~/8821cu-20210118 $ sudo ./remove-driver.sh
Running remove-driver.sh version 20221101
Kernel=5.15.76-v7+
Architecture=armv7l

------------------------------
Deleting module version: 5.12.0
completely from the DKMS tree.
------------------------------
Done.
rtl8821cu/5.12.0 has been removed
If older versions of rtl8821cu are installed, run:
$ sudo dkms remove rtl8821cu/<version> --all
dkms shows the following drivers are still installed:
Removing rtw88_8821cu.conf from /etc/modprobe.d
Removing 8821cu.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl8821cu-5.12.0
The driver was removed successfully.
You may now delete the driver directory if desired.
Do you want to reboot now? (recommended) [y/N]
```

ashah7, 2022-11-29:
Hi
Managed to fix it using the suggestion here about changing the Makefile 
https://github.com/diederikdehaas/rtl8812AU/issues/103

Abid

morrownr, 2022-11-29:
> Managed to fix it using the suggestion here about changing the Makefile

Okay, that tells me you did not follow the instructions in installation step 8.

FYI: I am working on a new version of this driver driver in the background right now. It will eliminate step 8 as autodetection is being added. Hopefully this issue will be avoided in the future.

morrownr, 2022-11-29:
Hey @ashah7 

Would you like to help test the new version?

ashah976, 2022-11-29:
Sure I only have the raspberry pi 3 to test on at the moment if okay

RyanScott96, 2022-11-29:
> Okay, that tells me you did not follow the instructions in installation step 8.

This is definitely what it was. I DID follow step 8, but mistakenly assumed I had a 64 bit OS. Re-running with the correct ARM_PI config worked for me.

P.S. thanks for your work on this! I greatly appreciate being able to use an antenna on my RPI!

morrownr, 2022-11-30:
@RyanScott96 

> mistakenly assumed I had a 64 bit OS.

You are far from the only one that has had this problem. I think I have a solution in the new driver. Do you want to be tester?

RyanScott96, 2022-11-30:
I won't have time to do anything tonight, but sure. Just let me know what you need me to do

ashah7, 2022-12-02:
Sure let me know what you want testing

morrownr, 2022-12-02:
@ashah7 

The repo with new driver is showing a pending invitation for you. You have to check your email and accept the invitation. I can redo it if you need me to do so.

The repo is private for testing. Once you are in, there is a place to post a check in message and you can look around to get an idea what is going on. If you have questions, ask in issues inside the private repo.

Nick

ashah7, 2022-12-03:
> @ashah7
>
> The repo with new driver is showing a pending invitation for you. You have to check your email and accept the invitation. I can redo it if you need me to do so.
>
> The repo is private for [...]

Please send again to this username (I think I was logged in last time from my phone and it must have gone to the other account)