morrownr/8821cu-20210118 #72: Concurrent mode

tamojitray, 2022-11-28:
I was curious and tried to install the driver with `CONFIG_CONCURRENT_MODE = y`.  Surprisingly driver installed perfectly and I got two interfaces wlan0 and wlan1. 

So first I tried to connect to a wifi network via wlan0 and create a hotspot on wlan1. Both interfaces were set to the same channel on 2,4Ghz and it worked flawlessly. In `iw dev` wlan0 was at **managed** and wlan1 was at **AP** at the same time.

So next I tried to put the wlan0 in monitor mode and create a hotspot on wlan1. And here the problem begins. Once any of the interfaces is in monitor mode the other interface will not create AP mode sometimes it creates AP mode but devices won't connect.

If concurrent mode is supported by this driver or it may be available in the future.

morrownr, 2022-11-28:
Hi @tamojitray 

> So next I tried to put the wlan0 in monitor mode

The Realtek concurrent mode does not support monitor mode.

You would need an adapter that uses in-kernel drivers to bring up multiple interfaces on the same adapter and one of them be monitor mode:

https://github.com/morrownr/USB-WiFi

That is the Main Menu to this site. Menu item 1 gives basic info and menu item 2 is a large list if adapters that use in-kernel drivers.

Nick

tamojitray, 2022-11-28:
Can we expect monitor mode in concurrent mode in the future or Realtek will never support that. 

Other than the Realtek adapter nothing is available in my country and if some are present those are very expensive.

In **88x2bu-20210702** repo I see you have mentioned **Supported interface modes: Concurrent** does this also mean no monitor mode on concurrent mode.

morrownr, 2022-11-28:
> Can we expect monitor mode in concurrent mode in the future or Realtek will never support that.

Unknown. Realtek does not communicate with community developers.

> Other than the Realtek adapter nothing is available in my country and if some are present those are very expensive.

May I ask what is your country?

> In 88x2bu-20210702 repo I see you have mentioned Supported interface modes: Concurrent does this also mean no monitor mode on concurrent mode.

None of the Realtek adapters have concurrent mode that supports monitor mode. I am currently in the process of removing concurrent mode from the README in all of the Realtek drivers here at the site as it has become a real support headache for me. I will leave the document in the docs folder but will no longer mention it in the installation. The problem is that Realtek's concurrent mode doesn't do what people want it to do and fixing that is beyond capability as we don't have documentation from Realtek.

closed 2022-12-15 by morrownr