morrownr/8821cu-20210118 #75: support TL-WDN5200

0neday, 2022-12-04:
I have ac650, Tp-link, TL-WDN5200 usb wireless adapter

morrownr, 2022-12-04:
Hi @0neday 

I think you are asking if the subject adapter is supported by this driver? If so, please open and read the file called:

supported-device-IDs

0neday, 2022-12-05:
Thank you,
I check it 

[image]

`0BDA:C811` 

this usb adapter have a CD driver for install driver

morrownr, 2022-12-05:
Ummm... I meant this file:

https://github.com/morrownr/8821cu-20210118/blob/main/supported-device-IDs

> this usb adapter have a CD driver for install driver

That means it is a multi-state adapter and requires usb_modeswitch. If you go to the site Main Menu:

https://github.com/morrownr/USB-WiFi

And select menu item 7 for more information.

Let me know how it goes.

morrownr, 2023-02-14:
Hi @King-of-Infinite-Space 

Ah, the joys of having a multi-state adapter. Not!

The new version this driver is up now so how about we continue after you remove this driver and install the new one. Look at the top of the README and you will find a link that will get you over to the new driver. Repost there.