morrownr/8821cu-20210118 #77: UBSAN: array-index-out-of-bounds in halrf.c

jeff-hiner, 2022-12-06:
I've got a PopOS system that hasn't been able to suspend with the Wifi dongle plugged in, both on the current 6.08 kernel and on earlier 5.x ones. The chipset otherwise seems to function correctly-- it connects to an AP and traffic works. But suspend causes the system to hang and requires a power cycle to fix.

Given the dmesg output I suspect that it might be related to an array out of bounds deref bug.

```
[   62.049215] usb 1-2: new high-speed USB device number 10 using xhci_hcd
[   62.197930] usb 1-2: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[   62.197948] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   62.197953] usb 1-2: Product: 802.11ac NIC
[   62.197958] usb 1-2: Manufacturer: Realtek
[   62.197962] usb 1-2: SerialNumber: 123456
[   62.246736] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   62.246982] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   62.452600] usbcore: registered new interface driver rtl8821cu
[   62.460134] rtl8821cu 1-2:1.0 wlx0ccf896fc3d4: renamed from wlan0
[   63.159843] IPv6: ADDRCONF(NETDEV_CHANGE): wlx0ccf896fc3d4: link becomes ready
[   67.973711] ================================================================================
[   67.973723] UBSAN: array-index-out-of-bounds in /var/lib/dkms/rtl8821cu/5.12.0/build/hal/phydm/halrf/halrf.c:445:17
[   67.973732] index 1 is out of range for type 'bool [1][2]'
[   67.973738] CPU: 10 PID: 3997 Comm: RTW_CMD_THREAD Tainted: P           OE      6.0.6-76060006-generic #202210290932~1669062050~22.04~d94609a
[   67.973746] Hardware name: Gigabyte Technology Co., Ltd. H610M S2H DDR4/H610M S2H DDR4, BIOS FC 12/17/2021
[   67.973750] Call Trace:
[   67.973754]  <TASK>
[   67.973760]  show_stack+0x52/0x69
[   67.973773]  dump_stack_lvl+0x49/0x6d
[   67.973787]  dump_stack+0x10/0x18
[   67.973791]  ubsan_epilogue+0x9/0x43
[   67.973796]  __ubsan_handle_out_of_bounds.cold+0x44/0x49
[   67.973802]  ? hal_mpt_CCKTxPowerAdjust+0xee0/0x102e [8821cu]
[   67.974050]  halrf_iqk_dbg+0x99/0x95c [8821cu]
[   67.974230]  phy_iq_calibrate_8821c+0x124/0x19d [8821cu]
[   67.974371]  halrf_segment_iqk_trigger+0x213/0x25a [8821cu]
[   67.974530]  rtw_phydm_iqk_trigger+0x4e/0x5e [8821cu]
[   67.974749]  rtl8821c_switch_chnl_and_set_bw+0x14f/0x34d [8821cu]
[   67.974956]  rtl8821c_handle_sw_chnl_and_set_bw+0x1ce/0x1d3 [8821cu]
[   67.975145]  rtl8821c_set_channel_bw+0x2d/0x45 [8821cu]
[   67.975327]  rtw_hal_set_chnl_bw+0xca/0x15d [8821cu]
[   67.975554]  set_channel_bwmode+0x288/0x2de [8821cu]
[   67.975715]  ? sethwreg+0x187/0x18c [8821cu]
[   67.975929]  join_cmd_hdl+0x2eb/0x379 [8821cu]
[   67.976076]  ? rtw_chk_start_clnt_join+0x1ba/0x1ba [8821cu]
[   67.976209]  rtw_cmd_thread+0x4b9/0x6b4 [8821cu]
[   67.976292]  ? rtw_stop_cmd_thread+0x51/0x51 [8821cu]
[   67.976373]  kthread+0xeb/0x120
[   67.976384]  ? kthread_complete_and_exit+0x20/0x20
[   67.976393]  ret_from_fork+0x1f/0x30
[   67.976404]  </TASK>
[   67.976407] ================================================================================
[   67.976410] ================================================================================
[   67.976413] UBSAN: array-index-out-of-bounds in /var/lib/dkms/rtl8821cu/5.12.0/build/hal/phydm/halrf/halrf.c:445:17
[   67.976418] index 1 is out of range for type 'bool [1][2]'
[   67.976421] CPU: 10 PID: 3997 Comm: RTW_CMD_THREAD Tainted: P           OE      6.0.6-76060006-generic #202210290932~1669062050~22.04~d94609a
[   67.976427] Hardware name: Gigabyte Technology Co., Ltd. H610M S2H DDR4/H610M S2H DDR4, BIOS FC 12/17/2021
[   67.976429] Call Trace:
[   67.976430]  <TASK>
[   67.976432]  show_stack+0x52/0x69
[   67.976438]  dump_stack_lvl+0x49/0x6d
[   67.976444]  dump_stack+0x10/0x18
[   67.976448]  ubsan_epilogue+0x9/0x43
[   67.976453]  __ubsan_handle_out_of_bounds.cold+0x44/0x49
[   67.976458]  ? hal_mpt_CCKTxPowerAdjust+0xee0/0x102e [8821cu]
[   67.976674]  halrf_iqk_dbg+0xaa/0x95c [8821cu]
[   67.976850]  phy_iq_calibrate_8821c+0x124/0x19d [8821cu]
[   67.976996]  halrf_segment_iqk_trigger+0x213/0x25a [8821cu]
[   67.977160]  rtw_phydm_iqk_trigger+0x4e/0x5e [8821cu]
[   67.977379]  rtl8821c_switch_chnl_and_set_bw+0x14f/0x34d [8821cu]
[   67.977586]  rtl8821c_handle_sw_chnl_and_set_bw+0x1ce/0x1d3 [8821cu]
[   67.977778]  rtl8821c_set_channel_bw+0x2d/0x45 [8821cu]
[   67.977956]  rtw_hal_set_chnl_bw+0xca/0x15d [8821cu]
[   67.978186]  set_channel_bwmode+0x288/0x2de [8821cu]
[   67.978348]  ? sethwreg+0x187/0x18c [8821cu]
[   67.978562]  join_cmd_hdl+0x2eb/0x379 [8821cu]
[   67.978715]  ? rtw_chk_start_clnt_join+0x1ba/0x1ba [8821cu]
[   67.978858]  rtw_cmd_thread+0x4b9/0x6b4 [8821cu]
[   67.978969]  ? rtw_stop_cmd_thread+0x51/0x51 [8821cu]
[   67.979079]  kthread+0xeb/0x120
[   67.979087]  ? kthread_complete_and_exit+0x20/0x20
[   67.979096]  ret_from_fork+0x1f/0x30
[   67.979106]  </TASK>
[   67.979124] ================================================================================

~$ sudo uname -mr
6.0.6-76060006-generic x86_64
~$ mokutil --sb-state
SecureBoot disabled
Platform is in Setup Mode
~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 007: ID 2109:8817 VIA Labs, Inc. PD3.0 USB-C Device     
Bus 001 Device 009: ID 0d8c:0014 C-Media Electronics, Inc. Audio Adapter (Unitek Y-247A)
Bus 001 Device 008: ID 05ac:024f Apple, Inc. Aluminium Keyboard (ANSI)
Bus 001 Device 006: ID 2109:4817 VIA Labs, Inc. USB2.0 Hub             
Bus 001 Device 005: ID 062a:5918 MosArt Semiconductor Corp. 2.4G Keyboard Mouse
Bus 001 Device 003: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub             
Bus 001 Device 002: ID 0b05:190e ASUSTek Computer, Inc. ASUS USB-BT500
Bus 001 Device 010: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
~$ rfkill list all
0: hci0: Bluetooth
	Soft blocked: yes
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
~$ dkms status
nvidia/515.65.01, 6.0.2-76060002-generic, x86_64: installed
nvidia/515.65.01, 6.0.3-76060003-generic, x86_64: installed
nvidia/515.65.01, 6.0.6-76060006-generic, x86_64: installed
rtl8821cu/5.12.0, 6.0.6-76060006-generic, x86_64: installed
system76/1.0.14~1643391291~22.04~78ede46, 6.0.2-76060002-generic, x86_64: installed
system76/1.0.14~1643391291~22.04~78ede46, 6.0.3-76060003-generic, x86_64: installed
system76/1.0.14~1643391291~22.04~78ede46, 6.0.6-76060006-generic, x86_64: installed
system76_acpi/1.0.2~1659568831~22.04~a88dcce, 6.0.2-76060002-generic, x86_64: installed (original_module exists)
system76_acpi/1.0.2~1659568831~22.04~a88dcce, 6.0.3-76060003-generic, x86_64: installed (original_module exists)
system76_acpi/1.0.2~1659568831~22.04~a88dcce, 6.0.6-76060006-generic, x86_64: installed (original_module exists)
system76-io/1.0.2~1655490480~22.04~0217576, 6.0.2-76060002-generic, x86_64: installed
system76-io/1.0.2~1655490480~22.04~0217576, 6.0.3-76060003-generic, x86_64: installed
system76-io/1.0.2~1655490480~22.04~0217576, 6.0.6-76060006-generic, x86_64: installed
~$ iw dev
phy#0
	Interface wlx0ccf896fc3d4
		ifindex 3
		wdev 0x1
		addr <redacted>
		ssid <redacted>
		type managed
		channel 149 (5745 MHz), width: 80 MHz, center1: 5775 MHz
		txpower 17.00 dBm
~$ iw reg get
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)
```

jeff-hiner, 2022-12-07:
Okay! The bad news: there are several OOB references that size arrays off NUM and then proceed to index via loops up to 2 (e.g. `iqk_fail`). But the good news: changing the hardcoded `NUM` from 1 to 2 allows the indexes to happen correctly (even when they're unused later) and fixes the suspend issue!

https://github.com/morrownr/8821cu-20210118/pull/78

I'll PR this change into the other repo as well.

morrownr, 2022-12-07:
Jeff,

That is good news. Please post the original details as you did above in the new driver repo along with the PR so we can let the other testers know to test this, I know I will be testing it to see if any adverse issues arise. That for taking the initiative to work on this.

Nick

PR #78 (2022-12-07): Avoid UB from out-of-bounds indexing
Fixes #77
closed 2022-12-07 by morrownr